hoylabs/OpenDTU-OnBattery #1263: OpenDTU Live data (Power from Inverter) frozen when DC Power from Inverter was cut off by external Battery Protect

Kermit1974, 2024-09-17:
### What happened?

Inverter Power on Live Data Page freezes when Inverter-DC-power was removed.

### To Reproduce Bug

Unfortunately every time when my Hoymiiles HMS-800 Inverter was "hard disconnected" by external Battery Protect Switch on DC-Side (if Battery Power falls down 25,4 Volts) the value of current Power is frozen and shows the same value until the Inverter will be switched on again.
Seems like open DTU doesn´t recognized that related inverter is not active/available.
Is there any possibility to fix this ?? 

### Expected Behavior

set actual Inverter Power to "0" if unit is not connected/available (time out)

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

2024.06.03

### What firmware variant (PIO Environment) are you using?

generic_esp32

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2024-09-17:
> every time when my Hoymiiles HMS-800 Inverter was "hard disconnected" by external Battery Protect Switch on DC-Side (if Battery Power falls down 25,4 Volts)

That is to be avoided. Check your (DPL) settings. Why does this happen?

Try the "Zero runtime data" setting (Inverter setting):
[image]

I am closing this issue because (1) I assume that "Zero runtime data" will solve your issue and (2) your battery-connected inverter should never be disconnected from the battery during normal operation (even in autumn or winter, as the solar yield is expected to cover the 3~4W the inverter consumes in standby).

closed 2024-09-17 by schlimmchen

Kermit1974, 2024-09-17:
Hello schlimmchen, thanks for your fast response in this issue !
Sorry for opening this case. I thought that the function "Zero runtime data" is only to clear the **sum** of the watts, not to reset also actual power value. 
I will activate this function and try! 
P.S. The hardwired deactivation on DC-Side of the Inverter is to prevent to empty the battery if the voltage drops below critical level at night. I know that now it is possible to include the battery voltage level into calculation of the DTU, but this part of installation was from my former installation with the open dtu (before open dtu **on battery**)