hoylabs/OpenDTU-OnBattery #1273: [Request] Allow using U0TXD/U0RXD as GPIO

ranma, 2024-09-22:
### Is your feature request related to a problem? Please describe.

On my ESP32S3 board (LilyGo T-Display S3) I'm a bit short on free GPIOs, so I haven't hooked up the Victron TX pin yet.

Since the console is available over USB during development, I'd like to use the U0TXD/U0RXD GPIOS (GPIO43/GPIO44) for the Victron connection. However currently these are assigned to the default serial output and don't seem to be usable as regular GPIOs.

This might require calling `gpio_pad_select_gpio()` and/or disabling the UART.

### Describe the solution you'd like

When the pin config includes the UART pins, disable the UART automatically and switch the pins into GPIO mode.

### Describe alternatives you've considered

Set build-time defines to disable the UART.

schlimmchen, 2024-09-22:
Understandable.

> However currently these are assigned to the default serial output and don't seem to be usable as regular GPIOs.

Not sure what you mean. The bootloader spits out some messages on UART0 by default. Once the firmware loads, there should be no output on GPIO43/44.

You can disable the output of bootloader messages using a strap pin. Please look into the ESP32-S3 datasheet. There is a strap pin for that.

I have had no issue using GPIO44/43 on my Fusion board for other stuff (accepting that the bootloader will print some garbage on startup). Why do you think that something must be done in the firmware? I am not even sure that it is possible to do something in the firmware to keep the bootloader from talking.

Tpaydt, 2024-09-22:
I am currently using the GPIO43 + 44 for the W5500 module with my ESP32S3 board. Haven't had any problems so far.

ranma, 2024-09-23:
> You can disable the output of bootloader messages using a strap pin. Please look into the ESP32-S3 datasheet. There is a strap pin for that.

Hmm, need to check that. It may be that the strap is not on an accessable GPIO on my board.

> Why do you think that something must be done in the firmware? I am not even sure that it is possible to do something in the firmware to keep the bootloader from talking.

The bootloader talking is fine (I plan on taking that into account, using the pin for something that doesn't care about some noise at the wrong baud rate), but the firmware is also talking at runtime over U0RXD/U0TXD, any printfs from the espressif firmware still appear there. And, as mentioned, they don't actually seem to work as GPIO at the moment and are fixed as the UART output pins. AFAIU calling gpio_pad_select_gpio() should allow turning them back into GPIO at runtime.
"gpio_pad_select_gpio function is equivalent to PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[pin], PIN_FUNC_GPIO)."

ranma, 2024-09-23:
> Hmm, need to check that. It may be that the strap is not on an accessable GPIO on my board.

Yeah, looks like the strap is GPIO46 and that is hardwired to the onboard TFT display data line on the LilyGo T-Display S3 (which I had lying around unused).

schlimmchen, 2024-09-23:
> Yeah, looks like the strap is GPIO46 and that is hardwired to the onboard TFT display data line on the LilyGo T-Display S3 (which I used because I had it lying around unused).

So? That is most certainly an output on the ESP32 and an input on the display. In reset, the line if floating. Add a weak pull-up or pull-down and you can set the strap option.

> any printfs from the espressif firmware still appear there (on my board).

I really don't follow... What output is that? AFAIK, as far as you use the pin for something else, it will be muxed accordingly and it will hence not be used to print messages. In particular, there is no hardware UART that is even configured to use these pins.

You are using the `generic_esp32s3_usb` Variant, are you?

ranma, 2024-09-24:
> You are using the generic_esp32s3_usb Variant, are you?

Yes.

 > So? That is most certainly an output on the ESP32 and an input on the display. In reset, the line if floating. Add a weak pull-up or pull-down and you can set the strap option.
 
 In theory, yes :) That pin isn't very accessible on the board though. :)
 
 > I really don't follow... What output is that?

GPIO43/44 (U0TXD/U0RXD).

e.g. over USB I see:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4208ba74
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (406) esp_core_dump_flash: No core dump partition found!
E (406) esp_core_dump_flash: No core dump partition found!
[   483][E][main.cpp:45] setup(): [main] Entered OpenDTU setup()

Starting OpenDTU
[...]
[Pytes] Initialize interface...
[Pytes] Interface rx = 16, tx = 21
[Pytes] Twai driver installed
[Pytes] Twai driver started
[  1272][E][main.cpp:203] setup(): [main] test printf: pins should be reconfigured now. Reached end of setup()

Switch to WiFi mode

WiFi got ip: 192.168.8.242
Network connected
E (8337) wifi:CCMP replay detected: A1=30:30:f9:34:7a:f0 A2=04:f0:21:23:27:2a PN=4, RSC=5 seq=64624
Websocket: [/livedata][1] connect
```

At the same time, on U0TXD (GPIO43) I see:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4208ba74
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (406) esp_core_dump_flash: No core dump partition found!
E (406) esp_core_dump_flash: No core dump partition found!
[   483][E][main.cpp:45] setup(): [main] Entered OpenDTU setup()
[  1272][E][main.cpp:203] setup(): [main] test printf: pins should be reconfigured now. Reached end of setup()
E (8337) wifi:CCMP replay detected: A1=30:30:f9:34:7a:f0 A2=04:f0:21:23:27:2a PN=4, RSC=5 seq=64624
```
 
You can see that ESP framework messages using the ESP_LOG functions still go to the hardware serial output on the GPIO.

> AFAIK, as far as you use the pin for something else, it will be muxed accordingly

That said, apparantly I can't reproduce the issue anymore, testing again now the GPIO assignment of GPIO43/44 is working fine, maybe I had bad pin config last time...

closed 2024-09-24 by ranma