hoylabs/OpenDTU-OnBattery #1295: [Request] Add support for multiple inverters for power limiter

remigauthier, 2024-09-28:
### Is your feature request related to a problem? Please describe.

It is not currently possible to limite power for multiple inverters

### Describe the solution you'd like

Two possibilities I can think off :
- limit all inverters at the same percentage (using max power for instance to calculate if the total acheive the limit)
- shut down inverter one after the other till the limit is reached

remigauthier, 2024-09-28:
duplicate of https://github.com/helgeerbe/OpenDTU-OnBattery/issues/230

spcqike, 2024-09-29:
See https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1216

closed 2024-09-29 by schlimmchen