hoylabs/OpenDTU-OnBattery #1310: [Request] Generic Device Control via HTTP requests i.e. Shelly Plugs by configureable trigger conditions

Andrix82, 2024-10-04:
### Is your feature request related to a problem? Please describe.

Controlling devices like shelly plugs manually while monitoring openDTUonBattery data is getting exhausting :)

### Describe the solution you'd like

OpenDTUonBattery has many informations/intelligence directly avaiblable (Hoymiles, Smart Meter, MPPT, Huawei, Battery )   ( and now plenty ressources for new features.
I feel a very usefull extension would be the possibility to trigger generic http requets based on configureable trigger conditions to control all devices which support http requests like i.e. Shelly Plugs ( http://192.168.X.X/relay/0?turn=on/off )

Feature could look like that:

Define several http requests by IP adress ( like http://192.168.X.X/relay/0?turn=on )

Define for each of them trigger conditions:
Combination of at least 2 maybe 3 trigger conditions with && (AND)   OR ||(OR)  logic

Trigger Conditions can be:
Smart Meter Value
MPPT actual power
Hoymiles actual power
Huawei actual power
Battery Level ( both SoC or Voltage ) 

 => larger or smaller threshold

and additional a debounce paramter for each (combined) trigger

With such a feature very powerful smart home control could be done for all devices supporting common http requests!

Someone interested or like to take up ? 

I could also start or support but would need following help/infos:

-Reference to an Issue where a new feature "Tab" ( Level like DPL, Huawei... ) has been added to have code changes (mainly UI and scheduler) as reference
-Infos about the scheduling details (how often are tasks on those level executed ) to implement a debouncing counter.
-Triggering http request I would take over from Smart Meter http implementation...  ?!

Snoopy-HSS, 2024-10-21:
look at PullRequest [1330](https://github.com/hoylabs/OpenDTU-OnBattery/pull/1330)

Andrix82, 2024-10-21:
Wow, i am happy and surprised that this basic idea was considered and already something developed since no reaction yet :o

I am eager to see/test the changes in the next release :)

After first quick check of changes one question maybe in advance, is it possible then to configure exactly one shelly plug (for explicit use case generic charger) or also several plugs for i.e. load shedding usecases ?

Many thanks for the motivation and efforts! :)

schlimmchen, 2024-10-23:
@Snoopy-HSS's implementation is only about enabling and disabling a shelly to start or to stop charging a battery, based on limited parameters.

> With such a feature very powerful smart home control could be done for all devices supporting common http requests!

There it is: You are asking for a HomeAssistent in OpenDTU-OnBattery. Or a NodeRED in OpenDTU-OnBattery. The use cases "start charging a battery if the BMS says it requires immediate charging to prevent damage" or "start charging when power meter says we are exporting energy" are quite limited and target to charging the battery. It makes sense that OpenDTU-OnBattery takes on this task. A generic "do something when some conditions are met, but not those other conditions, or this additional condition" is quite a complex feature. Also, I don't see why OpenDTU-OnBattery should be responsible for a whole bunch of automations.

I won't veto such a feature if it is well-crafted and someone implements it, but I doubt that someone is willing to put in the time for that. It will take *a lot* of time, believe me. I can only speak for myself, and I will not touch this.

Snoopy-HSS, 2024-10-24:
Same from me.OnBattery is a controldevice around the hoymiles inverters and the Battery.To do homeautomation there are al lot more Tools and Projects with alle the features…For me, openHAB ist the number 1. it has also an openDTU Integration.

Andrix82, 2025-02-10:
Also when I still would see this a cool generic feature I understand your arguments and of course respect decision to not implement into openDTU.

I will try to implement the intelligence and logic in the Shelly Plugs itself. (Gen2+ Javascripts)

I would use the openDTU webapi:   "IP"/api/livedata/status     ... should give all the information needed! :)

Can someone explain or give more infos about this statement:

[image]

Is this really a risk ? , why should this usefull "standardized" JSON interface for other local network participants be removed or changed ?

Are there really plans to change/remove it ?

Just want to avoid to spend efforts for a script which becomes useless with the future developement of openDTUonBattery....

Thanks!

schlimmchen, 2025-02-10:
Well, this disclaimer is there for one reason: We don't want to force ourselves to keep compatibility. Example: The JSON returned for the Huawei status will change significantly in the future (I am currently working on that). There will be more data, but the existing data will also be restructured -- for good reasons. If someone relies on the current JSON structure, scripts/apps will break. And we need users to accept that.

That being said: You don't have to fear that some HTTP API endpoint is going away. As far as I can tell we never removed any feature from OpenDTU-OnBattery, yet. However, endpoints might move, and the returned data might be restructured, and maybe even the values change their meaning slightly. Another example: Efficiency was previously transported as "0.96" for 96%, but will be "96.12" in the future for the Huawei (live) status. Be prepared for these kind of changes when you use the Web API to retrieve data.

Consider subscribing to respective topics on the MQTT broker instead, if possible (support needed in Shelly firmware, topics must exist).

Andrix82, 2025-02-10:
Thanks for your quick response!

Now its clear and understandable and see no relevant risk or blocker to implement such an openDTU API based shelly script :)