hoylabs/OpenDTU-OnBattery #1316: [Request] target_power_consuption als MQTT-Topic für Huawei bereitstellen

dragricola, 2024-10-11:
### Is your feature request related to a problem? Please describe.

(English version see below)
Im Sommer bei Sonnenschein ist es meist kein Problem die Batterie mit PV voll zu laden, sodass man beim Laden der Batterie eher auch Energie ins öffentliche Netz einspeisen kann um zu vermeiden, dass beim Einschalten von Verbrauchern kurzzeitig Energie aus dem Versorgungsnetz bezogen werden muss. Im Winter und bei trübem Wetter dagegen kann die Batterie mit der PV oft nicht voll geladen werden, sodass man auch tagsüber möglichst vermeiden will, dass bei Lastabwürfen kurzzeitig Energie aus der Batterie ins Versorgungsnetz abgegeben wird oder man möchte bei Vorhandensein eines variablen Stromtarifs auch die Batterie laden, wenn der Strom billig ist und entladen bei hohen Strompreisen. Daher ist es wünschenswert, wenn die angestrebte Netzleistung und damit indirekt auch die Batterieleistung über MQTT von einer übergeordneten Homeautomation wie z.B. ioBroker, HA usw. gesteuert werden kann. Das Entladen der Batterie lässt sich indirekt über den Topic powerlimiter/cmd/target_power_consumption per MQTT einstellen. Dieser wirkt jedoch nur auf die Wechselrichterleistung und somit kann man nur die Einspeisung aus der Batterie ins Versorgungsnetz beeinflussen indem der Wechselrichter mehr Leistung liefert als die PV erzeugt. Zum Laden aus dem Versorgungsnetz müsste man auch die angestrebte Netzleistung des Huawei AC-Chargers per MQTT einstellen können. Diesen Wert kann man derzeit in den AC Ladegerät Einstellungen über Limits nur fest vorgeben, gemäß Doku existiert hierfür jedoch kein MQTT-Topic um diesen Wert verändern zu können.  
Mit der vorgeschlagenen Änderung wird es möglich, durch eine übergeordnete Steuerung in der Homeautomatisierung die Batterieladung in Abhängigkeit von der prognostizierten Solareinstrahlung zu steuern.

In summer it is often no problem to fully charge the battery with solar power. So it is better to supply a part of the solar energy into the public grid to avoid energy consumption from public grid when consumers are switched on. In winter and cloudy days the battery often can't be fully charged only from the PV panels so it is better to avoid feeding back energy from the battery into the public grid at load rejection. So it is desirable to control target power consuption by a homeautomation system like iobroker, HA, etc. over MQTT. For charging the battery with energy from the public grid the target power consumption of the huawei ac-charger additionally schould be configurable over MQTT similar to the Topic powerlimiter/cmd/target_power_consumption which only affects the inverter. 
The proposal would allow a superordinate home automation system to control the charging and discharging of battery conditional on the prognosed solar production. 

### Describe the solution you'd like

Bitte einen beschreibbaren Topic huawei/cmd/target_power_consumption eingerichten um auch das Batterieladen unabhängig von der Solarproduktion variabel vorgeben zu können.

Please establish a writable topic huawei/cmd/target_power_consumption to allow a variable charging of the battery independent from PV production.

drahdiwaberl, 2024-10-25:
oh? dann ist es ja eigentlich gut dass ich meinen huawei direkt über esphome an einem extra esp32 hängen hab - war eigentlich nur Notlösung weil ich das unterstützte can-bus-modul nicht zum laufen bekam.. 

obwohl.. was ist _huawei/cmd/limit_online_current_ ?
#https://opendtu-onbattery.net/firmware/mqtt_topics/#huawei-ac-charger-topics

dragricola, 2024-11-10:
> weil ich das unterstützte can-bus-modul nicht zum laufen bekam 

Bei mir läuft das can-bus modul ohne Probleme. Folgendes ist zu beachten: 
1. Bei den Modulen MCP2515-TJA1050 mit 7 Pins muss die Stromversorgung aufgetrennt und separat 3,3V und 5V Versorgung zugeführt werden. Siehe https://github.com/KlausLi/Esp-HuaweiR4850-Controller/blob/main/Bastelplan3192.png.  Auf Youtube gibt es dazu auch ein Video. Bei den Modulen mit 8 Pins ist das nicht nötig. 
2. Der Jumper für den CAN-Bus-Abschluss sollte ebenfalls gesetzt sein.

> obwohl.. was ist huawei/cmd/limit_online_current ?

ja, damit kann man den Huawei auch separat steuern. Das müsste dann gezielt separat vom ioBroker o. ä. gesteuert werden. 
Ich möchte aber nur über MQTT vorgeben, ob der Huawei im Modus 3 (mqtt.0.openDTUonBattery.huawei.cmd.mode) bei einer PV-Prognose mit viel Sonnenschein überschüssige Energie aus der PV eher großzügig ins Netz entlässt und somit die Batterie nur langsamer als möglich lädt oder bei geringer PV-Prognose knausrig allen Überschuss in die Batterie speichert oder sogar noch  mit zusätzlichen Strom aus dem Netz lädt. Dazu müsste die Konfiguration angestrebter Netzbezug unter AC Ladegerät Einstellungen / Limits über MQTT veränderbar gemacht werden.