hoylabs/OpenDTU-OnBattery #1325: DPL Mode change over http

Cartman586, 2024-10-17:
### Describe the solution you'd like

I would like to send the dpl mode (1,2,3) via an http command 

### Additional context

I would like to send the dpl mode (1,2,3) via an http command

Cartman586, 2024-10-17:
> ### Is your feature request related to a problem? Please describe.
> _No response_
>
> ### Describe the solution you'd like
> I would like to send the dpl mode (1,2,3) via an http command
>
> ### Describe [...]

To do this see pic (inverter on/of via dpl)
[image]

AndreasBoehm, 2024-12-07:
Please take a look here why we won't provide a stable public HTTP for this and also the alternative that we offer: https://github.com/hoylabs/OpenDTU-OnBattery/issues/252#issuecomment-2051888923

You can check which API calls are made by the WebUI and replicate them within your setup but pleas be warned that this API is not stable and might change/break at any point.

closed 2024-12-07 by AndreasBoehm