hoylabs/OpenDTU-OnBattery #1327: connection refused when using non default port as HTTP+JSON power meter

1-am-r00t, 2024-10-17:
### What happened?

When using HTTP+JSON power meter and specifying "http://192.168.10.205:1880/smartmeter" as URL, "Value 1: HTTP Error: connection refused" is being returned on the GUI at the bottom. This is a web endpoint on my Node Red instance.

### To Reproduce Bug

Use custom port in Power meter settings page, e.g. http://192.168.10.205:1880/smartmeter

### Expected Behavior

Connect to the given URL and display the correct data.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

83ed72e

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Anything else?

Almost looks like a regression bug from this issue that apparently should've been fixed: https://github.com/hoylabs/OpenDTU-OnBattery/pull/612

DTU and Node Red are in the same subnet (192.168.10.0/24) so it cannot be a firewall blocking the requests. 

Data format is correct when opening the page on any device in the same subnet:
`{"total_power":"262"}`

This used to work some months ago, unfortunately I don't recall anymore the version I last used HTTP+JSON for Power meter. I was using MQTT in the meantime but need to change back to HTTP+JSON soon.

It's not working on my new ESP32-S3 which is using the latest release (83ed72e / 2024.09.11) from Github, but neither on my "old" openDTU on ESP32 that used to run 5ab6fe9 (2024.06.03).

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

1-am-r00t, 2024-10-17:
> The code is in the HttpGetter here:
> 
> https://github.com/hoylabs/OpenDTU-OnBattery/blob/844d92008bfe2af9ff13c12d5c0e08fd8bc22c1b/src/HttpGetter.cpp#L65

Yeah, I was checking the commits from #612 . This part of the code does not seem to have been touched in the meantime.

> Could it be you need a special Content Header or a POST request ?

No, nothing at all. In fact, when opening the URL from any of my devices in LAN, it's working just fine. I can't even see the packets arriving on my Raspberry Pi's eth0 interface:
`
pi@raspberrypi:~ $ sudo tcpdump -i eth0 host 192.168.10.53 and port 1880 -vvv
tcpdump: listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes

^C
0 packets captured
3 packets received by filter
0 packets dropped by kernel
`
So I recon it must have something to do with openDTU.

When saving the config even if the test failed + turning on verbose logging, the console log unfortunately also does not provide more information than is already known:
`22:44:30.190 > [PowerMeterHttpJson] Value 1: HTTP Error: connection refused`

1-am-r00t, 2024-10-17:
After some more packet capturing (adding a filter to remove the MQTT traffic to the same host, which by the way is working fine), I found out the following:
```
22:55:53.358676 IP (tos 0x0, ttl 64, id 63544, offset 0, flags [none], proto TCP (6), length 44)
    192.168.10.53.52473 > 192.168.10.205.0: Flags [S], cksum 0x3120 (correct), seq 1700628468, win 5760, options [mss 1436], length 0
E..,.8..@..@..
5..
.....e]......`...1 ........
22:55:53.358837 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 40)
    192.168.10.205.0 > 192.168.10.53.52473: Flags [R.], cksum 0x5f31 (correct), seq 0, ack 1700628469, win 0, length 0
E..(..@.@..}..
...
5........e]..P..._1..
22:56:08.102498 IP (tos 0x0, ttl 64, id 63964, offset 0, flags [none], proto TCP (6), length 44)
    192.168.10.53.52474 > 192.168.10.205.0: Flags [S], cksum 0x9ce0 (correct), seq 1302017013, win 5760, options [mss 1436], length 0
E..,....@.....
5..
.....M.3.....`.............
22:56:08.102564 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 40)
    192.168.10.205.0 > 192.168.10.53.52474: Flags [R.], cksum 0xcaf1 (correct), seq 0, ack 1302017014, win 0, length 0
E..(..@.@..}..
...
5........M.3.P.......
```
As you can see, the DTU is trying to initiate a connection with destination port 0 - which of course will not work and therefor my Raspberry answers to this TCP SYN with a TCP RST. Anybody has a wild guess where?

stefan123t, 2024-10-17:
You may want to add the _port to this log message:

https://github.com/hoylabs/OpenDTU-OnBattery/blame/844d92008bfe2af9ff13c12d5c0e08fd8bc22c1b/src/HttpGetter.cpp#L111

schlimmchen, 2024-10-18:
@1-am-r00t I acknowledge this issue and thank you for reporting it.

In #1077 we, i.e, I, did not test this:
[image]

I guess the issue is with this snippet:
https://github.com/hoylabs/OpenDTU-OnBattery/blob/844d92008bfe2af9ff13c12d5c0e08fd8bc22c1b/src/HttpGetter.cpp#L62-L66

However, I don't understand right away what could be the issue there, that snippet seems legit.

I will come back to this next week.

stefan123t, 2024-10-19:
@1-am-r00t can you double check and post the exact URL you use in your config ? Maybe use a config export to get the json string that is stored on OpenDTU ?

stefan123t, 2024-10-19:
@schlimmchen I am wondering if `index = 0` is actually a valid match here, this would imply no domain/ip before the port, right ?

1-am-r00t, 2024-10-20:
@stefan123t I was checking the config export of both my old ESP32 and the new ESP32-S3. Both configs are equal and have the correct string for the URL saved: `"http_phases":[{"enabled":true,"url":"http://192.168.10.205:1880/smartmeter","auth_type":0,"username":"","password":"","header_key":"","header_value":"","timeout":1000,"json_path":"total_power","unit":0,"sign_inverted":false}`

@schlimmchen thanks! Unfortunately I wasn't paying attention on the project for some time because, you know, life just happens while you go - otherwise I would have participated as a candidate in the tests of the new power meter class. But yeah, here we are :D I don't quite understand aswell why it would not work like this. I may take the spare ESP32-S3 I have laying around and try to add some debugging as @stefan123t proposed, just to have some insight on the content of the variables like `_host`, `_port` and `index`.

The only difference I can see between the original fix from #612 and the current implementation is the following:
https://github.com/hoylabs/OpenDTU-OnBattery/blob/75e3d03ea4025ac77140d4c63a7909c1ff6302a2/src/HttpPowerMeter.cpp#L264-L266
https://github.com/hoylabs/OpenDTU-OnBattery/blob/844d92008bfe2af9ff13c12d5c0e08fd8bc22c1b/src/HttpGetter.cpp#L64-L65
In the original fix a new variable `the_host` got implemented and an additional `remove` on the original `_host` variable has been used to remove the hostname + colon while in the new power meter only `substring` is used twice. Not sure if that's the cause, just trying to narrow down the potential causes.

stefan123t, 2024-10-20:
@1-am-r00t the old version made a copy of `host` and stored the part before colon in `the_host`. It then removed the hostname’s characters + 1 for the colon `:` from the `host` which left you with the `port`.

See my previous comment for an explanation for a possible fix without using an extra variable.

stefan123t, 2024-10-20:
@schlimmchen you may flip / switch line 65 and 64 to fix the issue IMO.

schlimmchen, 2024-10-20:
Ah, yes! I can see it now. Swapping the lines should indeed do it. Thanks! 💪

1-am-r00t, 2024-10-20:
I didn't see your latest edit of the previous answer @stefan123t, sorry, but yes, you are right. Since `_host` is being shortened already in the first line, `_port` will never be anything else if there is a custom port inside the URL.

@schlimmchen are you going to fix it on your own or do you prefer to have a PR for it?

stefan123t, 2024-10-20:
@1-am-r00t I tend to update / correct mistakes in my posts.

schlimmchen, 2024-10-20:
Is https://github.com/1-am-r00t/OpenDTU-OnBattery/commit/a8d29a1a62167342b7c819367acf6d3f665cbcf0 tested and does indeed fixes the issue? Sure, I would love to merge your PR.

1-am-r00t, 2024-10-20:
> Is [1-am-r00t@a8d29a1](https://github.com/1-am-r00t/OpenDTU-OnBattery/commit/a8d29a1a62167342b7c819367acf6d3f665cbcf0) tested and does indeed fixes the issue? Sure, I would love to merge your PR.

Affirm, tested and verified (Y)

[image]
[image]
[image]

PR #1333 (2024-10-20): fix http power meter custom port regression bug
Due to power meter refactoring (#1077) a regression bug appeared, rendering the original bug (#612) useless.
Thx to @stefan123t for proposing a quick & easy fix by just swapping line 64 w/ line 65.
Fixes #1327
closed 2024-10-20 by schlimmchen