hoylabs/OpenDTU-OnBattery #1329: You were all wrong with this, in my case the solution was quite simple!

Solar320, 2024-10-19:
https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1249

"Exception: it seems 'yarn' is not installed/available on your system:"

Since I am only want to change some things in the code for myself I don't want to learn all about every detail in VSC or all the different programming languages. I said this many times but all I hear is, " you do not do this before" or you have to compile something else.  

But no one can explain an simple installation process to use yarn and which dependencies it has, also the yarn installer appearently do not test if node.js is also installed?!?

At this time I do not know many about what yarn or node.js  and this is still the same :) 

The problem is that this  "yarn not installed warning" does not appear because "yarn" was not installed in my system but because node.js was not installed! 

Why no one can tell me this? Because you don't think about this possibility and I do not know that I also need node.js 

I now Installed again Yarn "yarn-1.22.22.msi" then I Installed Node-v20.18.0-x64.msi
Without any other changes I now can compile my own version of Open DTU on Battery with VSC again.

https://github.com/yarnpkg/yarn/releases
https://nodejs.org/en

----------------------------------------------------------------------------------
Old Thread:
I have the same problem, I want to compile my own version and do not even know what to H yarn is !?!? 
( I only compile and I am no programmer, I only change something into the Plattform IO Override eg. the use of [env:opendtufusionv2]
or -DBOARD_HAS_PSRAM=1 and so on. ) ................................
---------------------------------------------------------------
The Answer was something like this :

"spcqike
Sep 13, 2024

Das Problem:
Viele User (wie du) wollen die Firmware compilieren ohne vorher die Webapp zu bauen. Das ist im Wiki beschrieben.

Ohne aktueller und gültiger Webapp kann es sein, dass man entweder eine Firmware bekommt, die nicht korrekt läuft, oder auch gar keine.

Daher das pre-script, welches die webapp automatisch compiliert, wenn nötig. Dafür braucht es zwingend die Abhängigkeiten (die es auch ohne Skript braucht, nur eben muss man es sonst manuell ausführen)

wenn du keine eigenen Anpassungen vornehmen musst oder willst, brauchst du eigentlich keine Firmware selbst kompilieren. Wenn du es doch willst, musst du die webapp bauen."

schlimmchen, 2024-10-19:
> You were all wrong with this

🚨 Watch your language.

> But no one can explain an simple installation process

[RTFM](https://opendtu-onbattery.net/firmware/compile_webapp/)!

> I now Installed again Yarn "yarn-1.22.22.msi"

You are not supposed to do that. Again: [RTFM](https://opendtu-onbattery.net/firmware/compile_webapp/)

stefan123t, 2024-10-19:
If I understand the issue right (fine print between the CAPITALS) the user wants to build only the firmware without changes to the vue WebApp.
Is it possible that the pre-script skips building the WebApp in case a dependency like node.js / uarn is missing ?
This would allow to build the firmware with some custom overrides built in.

schlimmchen, 2024-10-20:
> Is it possible that the pre-script skips building the WebApp in case a dependency like node.js / uarn is missing ?

Stefan... [RTFM](https://opendtu-onbattery.net/firmware/compile_webapp/):

[image]

The web app artifact is not versioned in OpenDTU-OnBattery. You cannot build OpenDTU-OnBattery without building the web app.

stefan123t, 2024-10-20:
@schlimmchen I think the pre-script to build it in case all the pre-requisites are in place are way better than the upstream manual triggered build! 

While I agree that is sufficiently documented that you need node.js matching the Frontend you want to build and corepack to download the yarn package manager in the manual on building the WebApp, it could use some highlighting to show the differences with the upstream document.

* https://www.opendtu.solar/firmware/compile_webapp/
* https://opendtu-onbattery.net/firmware/compile_webapp/#general-information

Maybe it should even be featured more prominently in the Readme.md that these are the prerequisites for building the project.

As this is in contrast to the upstream OpenDTU master where the webapp is versioned in Github I try to understand the valid reasons for this:

* Can the github actions script not build the WebApp using node.js and yarn when building the  whole project ?
* Do we need a subproject sobwe can specify a github action for building it automatically?
* How does Thomas handle this upstream, does he always build the WebApp locally and update the archive with his merge to master ?

schlimmchen, 2024-10-20:
> @schlimmchen I think the pre-script to build it in case all the pre-requisites are in place are way better than the upstream manual triggered build!

Nice, that we agree on this.

> Maybe it should even be featured more prominently in the Readme.md that these are the prerequisites for building the project.

# TODO

That sounds like a good suggestion.

> * Can the github actions script not build the WebApp using node.js and yarn when building the  whole project ?

Well, it does this, of course, to build firmware artifacts.

> * Do we need a subproject so we can specify a github action for building it automatically?

Hm?

> * How does Thomas handle this upstream, does he always build the WebApp locally and update the archive with his merge to master ?

Yes, webapp artifacts are committed manually by Thomas.

Notice that the workflow is fundamentally different: Thomas works more or less "in secret", then commits appear on his master branch and he adds the webapp binaries and creates a release.

We actually work on the development branch in public, and it makes no sense whatsoever to version a webapp binary for each commit we merge.

Also, I really think that people should use pre-built binaries, not build ones themselves. There should be no need to do so. What are they doing that? Do we need to release more often? Don't they understand the concept of pin_mapping.json? And if they do compile, I want them to read the docs.

stefan123t, 2024-10-22:
@Solar320 here is a good example how this can and should be communicated in an Open Source project: #1255 
Please read it and consider next time you object to some documentation deficiency that this is Free Software. Thanks.

schlimmchen, 2024-10-29:
Completed through 33b7697b371185c79de00613a4beee92f4933318. See [README](https://github.com/hoylabs/OpenDTU-OnBattery?tab=readme-ov-file#important-differences).

closed 2024-10-29 by schlimmchen