hoylabs/OpenDTU-OnBattery #1331: Victron MPPT Status Lastausgang

hoschiking, 2024-10-20:
### What happened?

ich habe schon mal davon berichtet, ohne Erfolg, daher ein Issue.
Wenn ein VICTRON MPPT an die OpenDTUoB angeschlossen, wird auch der Status des LASTAUSGANGS angezeigt.
Fälschlicher Weise wird dieser jedoch als LADEAUSGANG angezeigt und das ist falsch.
Auch wenn es eine Kleinigkeit ist, sollte es korrigiert werden, da es dadurch einen falschen Status suggeriert.
Siehe Screenshot. Wenn der Laderegler sich im Modus "Bulk" befindet, kann der Ladeausgang unmöglich OFF sein.
Dies betrifft alle MPPT, die einen Lastausgang haben, also bis 20A soweit ich weiß.
[image]

### To Reproduce Bug

Lastausgang in der Victron App ein oder ausschalten, schon sieht man es. 

### Expected Behavior

Umbennenung von " Status Ladeausgang" zu "Status Lastausgang"

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.09.11

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
nicht notwendig.
```

### Anything else?

Es betrifft alle mir bekannten Versionen, die ich bis jetzt benutzt habe seit Ende 2023

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

ferdlgogosch, 2024-10-20:
Ist mir auch schon aufgefallen, dass in der deutschen Übersetzung das falsche Wort steht. 👍

SW-Niko, 2024-10-21:
Hallo @hoschiking ,
da hast du recht, die deutsche Übersetzung ist etwas irreführend.

Victron bezeichnet den Ausgang in seinen Unterlagen als "Load output" und die englischen Texte verwenden die gleiche Bezeichnung.

[image]

Ich kann die deutsche Übersetzung demnächst mal auf "Lastausgang" umstellen.
Vielleicht gleich zusammen mit #947 ... dann wäre das auch erledigt.

hoschiking, 2024-10-21:
Das klingt gut, auch die Ausgabe des Stromes am Lastausgang gefällt mir :-)
Btw: load heißt wörtlich übersetzt Last, charging ist Laden, daher passt nach der Anpassung auch die wörtliche Übersetzung

schlimmchen, 2024-10-22:
Erledigt per 3aa850a6a6513f19b0cb5ca2a74404e0f8d50463.

closed 2024-10-22 by schlimmchen