hoylabs/OpenDTU-OnBattery #1348: DPL regelt HMS700 nicht auf angestrebten Netzbezug

Frohbi83, 2024-10-25:
### What happened?

Wechselrichter ist angelegt, auslesbar und liefert Werte
Stromzähler ist angelegt und liefert Werte 
DPL ist aktiviert und die gewünschten Schwellwerte sind eingestellt 
Der HMS700 speist jedoch mit voller Leistung ins Hausnetz ein, obwohl der Stromzähler schon stark negative Werte anzeigt
Manuell läst sich der Wechselrichter auf  gewünschte Leistungswerte begrenzen

### To Reproduce Bug

Wechselrichter starten
DPL aktivieren

### Expected Behavior

DPL regelt den Wechselrichter auf angestrebten Netzbezug 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

22.10.2024

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell
RX Period End
23:04:31.039 > Last missing
23:04:31.055 > Request retransmit: 3
23:04:31.063 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 83 86 
23:04:31.074 > Interrupt received
23:04:31.093 > Frame kaputt
23:04:31.182 > RX Period End
23:04:31.200 > Last missing
23:04:31.217 > Request retransmit: 3
23:04:31.247 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 83 86 
23:04:31.257 > Interrupt received
23:04:31.265 > Frame kaputt
23:04:31.271 > Websocket: [/livedata][10] disconnect
23:04:31.292 > RX Period End
23:04:31.306 > Last missing
23:04:31.322 > Request retransmit: 3
23:04:31.334 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 83 86 
23:04:31.347 > Interrupt received
23:04:31.359 > Frame kaputt
23:04:31.450 > RX Period End
23:04:31.466 > Last missing
23:04:31.484 > Request retransmit: 3
23:04:31.496 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 83 86 
23:04:31.511 > Interrupt received
23:04:31.530 > Frame kaputt
23:04:31.542 > RX Period End
23:04:31.556 > Last missing
23:04:31.571 > Retransmit timeout
23:04:33.234 > [DPL::announceStatus] waiting for inverter device information to be available
23:04:34.567 > [PowerMeterHttpJson] New total: -479.00
23:04:35.323 > Fetch inverter: 1144837xxxxx
23:04:35.412 > TX RealTimeRunData 868.00 MHz --> 15 83 76 75 13 80 11 36 24 80 0B 00 67 1C 07 E2 00 00 00 00 00 00 00 00 6F DE A1 
23:04:35.427 > Interrupt received
23:04:35.440 > RX 868.00 MHz --> 95 83 76 75 13 80 11 36 24 01 00 01 00 FE 00 FE 04 FB 05 13 0C 87 0C C1 00 00 2A | -82 dBm
23:04:35.530 > Interrupt received
23:04:35.549 > Frame kaputt
23:04:36.000 > RX Period End
23:04:36.205 > Last missing
23:04:36.229 > Request retransmit: 2
23:04:36.240 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 82 87 
23:04:36.257 > Interrupt received
23:04:36.270 > Frame kaputt
23:04:36.276 > RX Period End
23:04:36.287 > Last missing
23:04:36.301 > Request retransmit: 2
23:04:36.314 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 82 87 
23:04:36.332 > Interrupt received
23:04:36.347 > Frame kaputt
23:04:36.360 > RX Period End
23:04:36.374 > Last missing
23:04:36.391 > Request retransmit: 2
23:04:36.402 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 82 87 
23:04:36.413 > Interrupt received
23:04:36.422 > Frame kaputt
23:04:36.437 > RX Period End
23:04:36.454 > Last missing
23:04:36.471 > Request retransmit: 2
23:04:36.485 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 82 87 
23:04:36.501 > RX Period End
23:04:36.513 > Last missing
23:04:36.529 > Request retransmit: 2
23:04:36.540 > TX RequestFrame 868.00 MHz --> 15 83 76 75 13 80 11 36 24 82 87 
23:04:36.555 > RX Period End
23:04:36.570 > Last missing
23:04:36.588 > Retransmit timeout
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

spcqike, 2024-10-26:
Kann alles und nichts sein. 

Bitte lies dir das Wiki durch

https://github.com/hoylabs/OpenDTU-OnBattery/wiki/How-to-Search-for-Solutions%2C-Ask-Questions%2C-Report-Problems

anschliessend kannst du mit passendem log den Fehler genauer beschreiben.

Frohbi83, 2024-10-26:
Hallo, vielen Dank für deine Rückmeldung. Ich habe jetzt die ausführliche Protokollierung nur für den DPL und den Stromzähler aktiviert und folgenden Log erstellt:
https://pastebin.com/hwCJmS8Z
Dort ist zu sehen, dass der Stromzähler stark negative Werte liefert, der DPL aber keine Befehle an den Wechselrichter sendet.

Frohbi83, 2024-10-26:
Anbei noch einige Screenshots der DPL-Einstellungen und Wechselrichter-Infos
[image]
[image]
[image]
[image]
[image]

spcqike, 2024-10-26:
Dein Problem ist, dass der wechselrichter anhand seiner Hardware Teilenummer nicht korrekt erkannt wird. 

Damit gibt es keine maximale leistung (0W) und der DPL fängt gar nicht erst an. 

Bitte melde deine Hardware Nummer, wie im Screen gemeldet, im upstream Projekt.

spcqike, 2024-10-26:
@schlimmchen 
Spricht etwas dagegen, diese Überprüfung zu entfernen? 
klar, wir brauchen die maximale Leistung des wechselrichters zur Berechnung des prozentualen Limits. Aber könnte man das Limit nicht als absolutes Limit übertragen, wenn der Fall wie hier eintritt? 

Gibt es beim prozentualen überhaupt ein Vorteil gegenüber dem absoluten? 

Wenn es beim absoluten Limit keine nachteile gibt (Ansprechzeit / Regelzeit?), fällt mir nur 1 Szenario ein, wo der fehlende wechselrichter max limit wert Probleme macht: beim überskalieren, wenn das selbst definierte obere Leistungslimit kleiner ist, als das Maximum des wechselrichters. 
Extrembeispiel: ein auf 400W limitierter HMS-1600 mit überskalierung. Wenn alles korrekt läuft, kann man jetzt auch auf einem Eingang die 400W erzeugen. Wenn das WR Limit (1600) wegfällt, müsste man sich auf die 400 des oberen Limits beschränken und damit geht die Skalierung eben nicht mehr korrekt. Doch sonst? 🤷‍♂️

Steuern lassen sich wechselrichter so ja manuell. Nur eben nicht vom DPL bisher.

Frohbi83, 2024-10-26:
> Dein Problem ist, dass der Wechselrichter anhand seiner Hardware-Teilenummer nicht korrekt erkannt wird.
> 
> Damit gibt es keine maximale Leistung (0W) und der DPL fängt gar nicht erst an.
> 
> Bitte melden Sie Ihre Hardware-Nummer, wie im Bildschirm gemeldet, im Upstream-Projekt.

Hallo, die Hardwarenummer meines HMS700 wird laut tbnobody zum nächsten Release eingefügt. Danach sollte alles funktionieren. Ich warte also das nächste Update ab und berichte kurz ob alles funktioniert. Vielen Dank.

schlimmchen, 2024-10-28:
> Spricht etwas dagegen, diese Überprüfung zu entfernen?

Mindestens spricht dagegen, dass ich keinen wichtigen Grund sehe, darauf zu verzichten und dafür so viel Code umzustellen und Regressionen zu riskeiren. Der Wert fehlt nur bei unbekannten WR bzw. für eine kurze Weile nach dem Booten, ansonsten ist der verfügbar.

> Ich warte das nächste Update ab und berichte kurz ob alles funktioniert.

Hm... Ich werde dir mal schon vorher eine Firmware verlinken, die du ausprobieren kannst, wenn es soweit ist.

AndreasBoehm, 2024-11-22:
@Frohbi83 mit dem aktuellsten Release sollte dein HMS-700 nun korrekt funktionieren.
Das Release findest du hier: https://github.com/hoylabs/OpenDTU-OnBattery/releases/tag/2024.11.20

AndreasBoehm, 2024-12-07:
Hallo @Frohbi83, kannst du kurz Bescheid geben ob nun alles klappt?

Frohbi83, 2024-12-09:
Hallo, ich habe heute das Update durchgeführt und es funktioniert alles wunderbar. Vielen Dank nochmal.

closed 2024-12-09 by schlimmchen