hoylabs/OpenDTU-OnBattery #1350: OTA Update auf 2024.10.22 - HTTP ERROR Bad Request - ESP32-S3

Dani1802, 2024-10-26:
### What happened?

Beim Update auf 2024.10.22 kommt es immer zum Fehler "HTTP Error bad Request"

2024.09.11 lies sich damals problemlos per OTA flashen (und kann auch jetzt noch per OTA "drübergeflashed" werden)

Neustart und stromlos machen habe ich bereits ohne Erfolg versucht.

Ich verwende "opendtu-onbattery-generic_esp32s3.bin"

[image]

[image]

### To Reproduce Bug

OTA Update von 2024.09.11 auf 2024.10.22 mit ESP32-S3

### Expected Behavior

OTA Update auf 21024.10.22 funktioniert

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.09.11

### What firmware variant (PIO Environment)?

generic_esp32s3

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2024-10-26:
[image]
[image]
# Link klicken & Anweisungen lesen: [Upgrade Instructions](https://opendtu-onbattery.net/firmware/howto/upgrade_8mb/)

Dani1802, 2024-10-26:
Danke für die schnelle Antwort, das hab ich gelesen und da steht "Coming from firmware version 2024.06.03 or older", aber ich bin ja schon auf 2024.09.11.
Ich hatte damals beim Wechsel auf den ESP32-S3 die 2024.08.18 neu geflashed und das OTA Update auf die 2024.09.11 hatte auch geklappt.

schlimmchen, 2024-10-26:
Jap. Ich hab beim Abendessen schon reflektiert, dass meine Reaktion (wieder) bisschen zu stumpf/streng war. Bitte entschuldige. Ich hätte mich jetzt sowieso nochmal gemeldet.

Dass du den Sprung auf 2024.09.11 geschafft hast (siehe [Doku](https://opendtu-onbattery.net/firmware/howto/upgrade_8mb/#exception-for-esp32-s3) warum das geklappt hat) befreit dich nicht davon, das Partitionslayout aktualisieren zu müssen.

Kannst du mir helfen, den Hinweis in den Release Notes dahingehend zu verbessern, dass du nicht in diese Falle gerannt wärst? Vielleicht muss man im zweiten Absatz schreiben "If you already upgraded to version 2024.08.18 or newer **using the `.factory.bin` image over a wired connection to the ESP32**, you should be good to perform [...]"?

Der erste Absatz hat dann immer noch ein Problem...

Vielleicht muss man weg von den Versionsnummern und sich auf "altes" und "neues" Partitionslayout beschränken? Das muss man dann aber erklären und dann wir der Hinweis ein Artikel... Oder es muss so wage sein, dass man nicht kapiert worum es geht und dann muss man den Link klicken und die Doku lesen?

Also ich bin hier um Hilfe nicht verlegen.

spcqike, 2024-10-26:
Vielleicht „in order to use OTA, make sure to have 3.840kb sketch partition size. If not flashed allready, you need to wire-flash a .factory.bin from 2024.08.18 or above initially” ?

 Oder sowas.

Dani1802, 2024-10-26:
@schlimmchen Alles gut, kein Thema ;-)

Ich hatte den generellen Partitionswechsel dann wohl missverstanden, ich war nämlich der Meinung wenn ich auf den ESP32-S3 wechsle und 2024.08.18 per USB flashe, dann bin ich safe für die künftigen Releases.
Entweder ist dem nicht so, oder ich hab zwischendurch irgendwas falsch gemacht.
Ich werd die Tage dann das aktuelle Release neu per Kabel flashen und schauen wie groß der Sketch-Speicher dann ist.

Die Idee, auf die Größe der Sketch-Partition zu referenzieren halte ich auch für sinnvoll.
Ob die "neue" Sketch Partition dann 3.840kb groß ist werde ich sehen.

Ich würde den Vorschlag von @spcqike noch erweitern:
„in Order to use OTA make sure to have at least 8Mb flash size and the sketch-partition size greater than 1.920 kB,…..”

Dani1802, 2024-10-26:
Ich glaube langsam mir ist damals beim Wechsel auf den 32-S3 ein Fehler unterlaufen und ich hatte die 2024.06.03 erwischt.. :-/

Habe nun die 2024.10.22 per USB geflashed und siehe da
[image]

closed 2024-10-26 by Dani1802

stefan123t, 2024-10-27:
@schlimmchen die Doku zu verbessern ist immer ein guter Ansatz:
für OTA braucht man ab Version x 
* min 8MB Flash
* das neue Partitionslayout
* folgende Images
für die USB Version braucht man ab Version x
* min 4MB Flash
* ein USB Kabel
* folgende Images

Ausserdem fände ich es gut auf der Info > System Seite auch ein paar Flags anzuzeigen:
* neues Partitionslayout
* min 4/8MB Flash
* OTA Flashable
Die Information könnte man dann auch auf der Settings > Firmware Update Seite heranziehen und dem Nutzer einen Wink mit dem Zaunpfahl geben dass evtl kein OTA möglich ist oder ein bestimmtes Image per USB notwendig ist damit der ESP überhaupt noch supported ist.

schlimmchen, 2024-10-27:
> Die Information könnte man dann auch auf der Settings > Firmware Update Seite heranziehen und dem Nutzer einen Wink mit dem Zaunpfahl geben dass evtl kein OTA möglich ist oder ein bestimmtes Image per USB notwendig ist damit der ESP überhaupt noch supported ist.

Wir können keine veröffentlichten Firmwares verändern... Das wäre ja dafür nötig. Wenn kein OTA möglich ist, weil der Benutzer `generic_esp32_4mb_no_ota` verwendet, dann wird das auf der Firmware-Update-Seite auch glasklar.