hoylabs/OpenDTU-OnBattery #1354: [Request] Request to Display Individual Phase Power Values (L1, L2, L3) in Addition to Total Power

robbieffm, 2024-10-28:
### Is your feature request related to a problem? Please describe.

[image]
add a feature that displays the individual phase power values (L1, L2, L3) alongside the total power in the "Leistung / Netz xxx W" section. This would provide users with more detailed information, especially when their meter settings support retrieving values for individual phases.

### Describe the solution you'd like

display the individual phase power values (L1, L2, L3) alongside the total power

robbieffm, 2024-10-28:
https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1200

stefan123t, 2024-10-28:
In the Hoymiles Source Code there were the following commands:

https://github.com/lumapu/ahoy/wiki/Protocol#welche-device-info-req_arw_dat_all-0x15-sub-kommandos-gibt-es

RealTimeRunData_A_Phase (0x0D) ???
RealTimeRunData_B_Phase (0x0E) ???
RealTimeRunData_C_Phase (0x0F) ???

schlimmchen, 2024-10-28:
@stefan123t This is about PowerMeter values. Why are you citing Hoymiles source code?

stefan123t, 2024-10-28:
@schlimmchen I may be mistaken but do we use the three above commands for three-phase inverters like the HMT series ?
I do not know whether OP has specifically meant Display of Power Meter values or values from three-phase Inverters. It was unclear to me. But maybe that should have been clear from the screenshot ?

schlimmchen, 2024-10-28:
> maybe that should have been clear from the screenshot ?

I think so.