hoylabs/OpenDTU-OnBattery #1364: LittleFS not re-built when files in `data`dir change

schlimmchen, 2024-10-30:
### What happened?

When only changing files in the `data` dir, the newly created firmware image is the same as in the build before.

### To Reproduce Bug

Change a file in `data` but only that file, then rebuild firmware.

### Expected Behavior

LittleFS image is re-built and integrated into new firmware image.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU-OnBattery?

1234

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2024-11-06:
Could you please doublecheck if it works as expected if you replace this line:
https://github.com/hoylabs/OpenDTU-OnBattery/blob/827a272a19249f24771d00bc93bc9cb8fb1aa0ae/pio-scripts/create_factory_bin.py#L133

by 
```python
env.AddPostAction("buildprog", esp32_create_combined_bin)
```

schlimmchen, 2024-11-11:
Sorry for not getting back to you sooner. I was AFK over the weekend, merely skimmed some of the notifications, and I did not see this one yet. I came back here to note: "Probably fixed by ecb5e9cc3233bcd04b17134f2d88058e533ccaae". Seeing your request I tested it, but unfortunately, changes to files in the `data/` directory (in my test I changed the Wi-Fi SSID in `config.json`) did not create a new... Well, nothing new entirely:

```
[...]
|-- Update @ 2.0.0
|-- VeDirectFrameHandler
|-- ResetReason
|-- MqttSubscribeParser
|-- Ticker @ 2.0.0
|-- DNSServer @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- SPI @ 2.0.0
|-- TimeoutHelper @ 0.0.1
|-- SdmEnergyMeter
|-- SMLParser
Building in release mode
Retrieving maximum program size .pio/build/generic_esp32s3_usb/firmware.elf
Checking size .pio/build/generic_esp32s3_usb/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  26.5% (used 86828 bytes from 327680 bytes)
Flash: [======    ]  56.8% (used 2233409 bytes from 3932160 bytes)
============================================================================================ [SUCCESS] Took 7.23 seconds ============================================================================================

Environment          Status    Duration
-------------------  --------  ------------
generic_esp32s3_usb  SUCCESS   00:00:07.226
============================================================================================ 1 succeeded in 00:00:07.226 ============================================================================================
```

Given the approach you suggest, I asked ChatGPT about it. The code it produced was unusable, as is often the case, but there was a pointer, as is also often the case: `AlwaysBuild`. Using that I found some [documenation](https://docs.platformio.org/en/latest/platforms/creating_platform.html) that uses it.

This works as expected (`esp32_create_combined_bin` is always executed):
```python
from SCons.Script import AlwaysBuild
post_action = env.AddPostAction("buildprog", esp32_create_combined_bin)
AlwaysBuild(post_action)
```

tbnobody, 2024-11-11:
You are right. Will add it with the next release.

schlimmchen, 2025-01-14:
Implemented upstream, will land here shortly: https://github.com/tbnobody/OpenDTU/commit/eac2e2fb39c6fcac238cc5cd26b106fde27b515f

schlimmchen, 2025-02-01:
Merged from upstream => completed.