hoylabs/OpenDTU-OnBattery #1365: [Request] Hide inverter values in the live view if no inverter is present && Totals display of several DTUs

huey84, 2024-10-31:
### Describe the solution you'd like

As the radio conditions in my house are very difficult, I have an OpenDTU-OnBattery, which is only used to read out a smart meter (SDM630). The data is then distributed via MQTT. Since no inverter is configured, it would be nice if the associated display could be hidden in the live view.

AND:

It would also be helpful if the “Gesamtertrag ...” could be called up in the live view via an MQTT broker. What for? I have several OpenDTU-OnBattery in use, which send their data to a broker. The sum of the services could then be displayed on a single DTU. This way you could easily get a cheap total display.

schlimmchen, 2024-10-31:
> Since no inverter is configured, it would be nice if the associated display could be hidden in the live view.

Reasonable. We can do that.

> It would also be helpful if the “Gesamtertrag ...” could be called up in the live view via an MQTT broker.

Nope. That's an insanely complex feature with settings and stuff for a very few users.

> This way you could easily get a cheap total display.

There are side-projects just for that. Maybe one of them fits your needs?

huey84, 2024-10-31:
> > Since no inverter is configured, it would be nice if the associated display could be hidden in the live view.
> 
> Reasonable. We can do that.
Great! I'll wait patiently...
> 
> > It would also be helpful if the “Gesamtertrag ...” could be called up in the live view via an MQTT broker.
> 
> Nope. That's an insanely complex feature with settings and stuff for a very few users.
Yes, understandable. I'm sure not many users have this issue
> 
> > This way you could easily get a cheap total display.
> 
> There are side-projects just for that. Maybe one of them fits your needs?
That's a good tip. I'll have a look for it

Thanks a lot!

schlimmchen, 2024-11-13:
We should also consider

[image]

and hide the totals of no radio is configured/working.

spcqike, 2024-11-14:
i wonder, why you not simply use tasmota for your usecase.

you could use one tasmota device to read out your smartmeter, publish the needed readings via mqtt, like a lot of users do. (and you could get a lot more readings out of your smartmeter, compared to opendtu)

> It would also be helpful if the “Gesamtertrag ...” could be called up in the live view via an MQTT broker. [...] The sum of the services could then be displayed on a single DTU. 

this can be done with tasmota, too. you can use one tasmota device with display, let it listen to all needed mqtt topics and create your very own visuilization.

idk but thats what i would do. 

to remove inverter from live view and hide radio information, if no radio chip is configured, makes sense, somehow. 
but .... idk, it feels wrong. at least for me :) opendtu is (was?) designed to communicate with inverters. so a radio chip and inverter is an essential part. just using opendtu as "smartmeter gateway" and "overall display", when there are several other projects, that can do this and more even better, feels like tearing out your rear seats because you want to transport a bigger load, when you have a pickup right in your garage.

anyway, @schlimmchen these changes (hide in live view and hide radios) would be part of the upstream, wouldn't they?

stefan123t, 2024-11-14:
@huey84 just wanted changes to the life view. I would not want to change the system info / config pages.

@schlimmchen I would not hide the RF config flags in case it is not configured. These were added so the average user can easily detect mis-configurations.

schlimmchen, 2024-11-14:
> and more even better

Excuse me?

> would be part of the upstream, wouldn't they?

No, of course not. OpenDTU is essentially worthless without a radio interface. OpenDTU-OnBattery can actually do interesting things even when not talking to inverters.

> I would not hide the RF config flags in case it is not configured.

Who said that? What I meant was that we should not only remove the totals from the live view when no inverter is configured, but also until a working radio is configured.

spcqike, 2024-11-14:
> Excuse me?

no offense.
 I meant that there are other projects like Tasmota, which would be better, if you just want a device to read your smart meter. Maybe I’m wrong with that, but can openDTU read out and publish all obis or whatever messages from all possible smart meters? Like voltages, current, power factors,… per phase? Can you, if you want or need to, create scripts to publish your very own readings? I think you know Tasmota quite well and you know what it can do. 
maybe I’m wrong and openDTU can also read such messages, but I thought it can’t. 
and, if he just wants to have a central overall display, where he can show all his devices stats, Tasmota is also great. As it supports a variety of displays and you can create your own dashboards with menus and such. 

So, I don’t know if one must put time and effort into integrating such functions into openDTU, too. As, at least for me, openDTU is a great device to communicate with and control inverters.

huey84, 2024-11-14:
Hello everyone, 

if I had known what I would trigger with my question, I would not have asked it. Everything works perfectly and the changes are not necessary. It was just an idea!

Why not Tasmota? Lack of time... I had an ESP with openDTU OnBattery left over from summer, hooked it up to my smart meter and it's been working perfectly ever since. It was that simple... But thanks for the tip. That gives me a little project for Christmas ;o)

stefan123t, 2024-11-14:
> > I would not hide the RF config flags in case it is not configured.
> 
> Who said that? What I meant was that we should not only remove the totals from the live view when no inverter is configured, but also until a working radio is configured.

@schlimmchen I got the wrong assumption reading it on the small print mobile device :nerd_face: 
You really did not mention that, it was just me being scared under this impression :scream_cat:

schlimmchen, 2024-12-01:
> if I had known what I would trigger with my question, I would not have asked it.

I did not catch this earlier, so let me add to your concern that I think the debate is very civilized, no need for you to feel your request would be out of place.

[image]

[image]

Implemented in cf59d77b.

closed 2024-12-01 by schlimmchen