hoylabs/OpenDTU-OnBattery #1370: [Request] Add bulk/float mode for AC (Huawei) charger

gitisgreat2023, 2024-11-04:
### Is your feature request related to a problem? Please describe.

Balancing typically occurs at say 3.45V per cell, or 55.2V for 16S. After say an hour it is desirable to drop the voltage lower, say 3.35V per cell (53.6V for 16S) to still charge to battery further but not stress the battery with a high voltage.
For the Huawei charger however currently only a single charging voltage ("Charge Voltage Limit") can be configured.

### Describe the solution you'd like

Set a bulk voltage  (and duration after that has been reached, so a timer) and a float voltage
So two additional variables would be introduced for the Huawei charger.

gitisgreat2023, 2025-01-30:
@schlimmchen You already are touching the Huawei code at the moment. Would this be the time to pick it up to make the Huawei a "real solar charger"?
You have a Huawei and JK BMS, right? This float mode is basically the only thing which is missing compared to a Victron Solar charger.

Load voltage 55.2V (3.45V, start of balancer) -> already present
Float voltage 53.6V (3.35V), voltage which said after a configurable timer (say typically 10 minutes) has been reached since first reaching the load voltage. Optionally a third parameter, charge current, could also be offered. In case the battery isn't balanced and draws still quite some current at the load voltage.

But basically two new parameters would be needed: float voltage (<= load voltage) and a timer in minutes.

I'm happy to implement and test it first in Node Red. Just like the set/get compensation for the inverter power, if you remember.

schlimmchen, 2025-02-02:
I get what you mean. However, I don't see why this is an important feature. Hence I'll set the "contribution needed" label.