hoylabs/OpenDTU-OnBattery #1377: [Request] Status Huawei PowerPin über mqtt verfügbar machen

tompsis, 2024-11-07:
### Is your feature request related to a problem? Please describe.

Wenn der Status des PowerPin  vom Huawei Netzteil über mqtt verfügbar wäre, könnte darüber ein shelly oder ähnliches gesteuert werden. 
Bei einigen / meinem Huawei sind die SlotDetect-Anschlüsse ohne Funktion. 

### Describe the solution you'd like

Status Huawei PowerPin über mqtt verfügbar machen

dragricola, 2024-11-12:
> Bei einigen / meinem Huawei sind die SlotDetect-Anschlüsse ohne Funktion.

Das kann ich bestätigen. Mein Huawei ignoriert ebenfalls den Slot-Detect-Input. Ich habe mir beholfen, indem ich über das am Power-Pin gesteuerte Relais einen Sonoff DualR3 steuere und damit den Huawei AC-seitig ein- und abschalte. Der DualR3 misst auch die Leistung und kann zusätzlich über WiFi gesteuert werden. Über den anderen Kanal des DualR3 werden übrigens die Inverter ein- und ausgeschaltet.

tompsis, 2025-02-03:
Ich habe ein Huawei R4850G2 und habe das Problem gelöst, indem ich mit dem Power-Pin und einen Opto-Koppler über einen Shelly das Netzteil komplett ein- und ausschalte. Das funktioniert sehr zuverlässig und hat den Vorteil, dass ich im Shelly zur Sicherheit eine Leistungsbegrenzung einstellen kann und notfalls auch manuell das Netzteil remote ausschalten kann.   

Die Möglichkeit, über CAN das Netzteil zu schalten, wäre natürlich eine sehr schöne Alternative und würde Verdrahtungsaufwand sparen.

dragricola, 2025-02-04:
@schlimmchen Ich habe ebenfalls einen Huawei R4850G2 und schalte ähnlich wie @tompsis die 220V-Versorgung über einen Sonoff DUALR3 2-Gang WiFi switch ein und aus. Der Switch wird durch ein Relais gesteuert, das wiederum mit dem power-Signal aus openDTUoB gesteuert wird. Die Vorteile dieser Lösung sehe ich genau so wie tompsis. Der zweite Kanal des DUALR3 schaltet übrigens nach Bedarf die Hoymiles-Wechselrichter ein und aus.

Das Ein- und Ausschalten über CAN-Bus wäre sicher gut. Das Schalten sollte aber auch über ein MQTT-Topic auszulösen sein, damit die gleiche Steuerbarkeit manuell oder durch eine übergeordnete Hausautomatisierung wie mit einem Sonoff- oder Shelly-Switch möglich wird. In meiner Anlage würde ich vermutlich den Sonoff-DUALR3 trotzdem drin lassen.

schlimmchen, 2025-02-04:
> Das Ein- und Ausschalten über CAN-Bus wäre sicher gut. Das Schalten sollte aber auch über ein MQTT-Topic auszulösen sein

Das habe ich als erstes implementiert, weil es die einfachste Möglichkeit ist, solche Befehle während der Entwicklung bei Bedarf abzusetzen. Also das wird es auf jeden Fall geben.