hoylabs/OpenDTU-OnBattery #1378: Huawei: "Limit setzen, CAN Bus nicht verbunden" funktioniert nicht

Andrix82, 2024-11-09:
### What happened?

Wenn die CAN Verbindung zum Huawei R4850G2 verloren geht oder die openDTU keine Steuerbefehle sendet, ladet das Huawei Ladegerät immer mit Maximalleistung 3kW.
Der Fehlerzustand ist somit leider kein sicherer/unbedenklicher Zustand.    -   System kann man so kaum bedenkenlos zurücklassen und einfach mal auf Urlaub fahren :D

### To Reproduce Bug

Nun mit 2 unterschiedlichen sonst einwandfrei funktionierenden Huawei R4850G2 getestet:
-"Limit setzen, CAN Bus nicht verbunden"  mit 53V und 1A setzen
-CAN Kabel abstecken ODER Ladegerät Feature in openDTU deaktivieren
-AC Verbrauch des Huawei observieren -> geht nach einigen Sekunden auf 3kW Ladeleistung
nach Wiederherstellung der CAN Verbindung Ladeleistung unmittelbar wieder geregelt

### Expected Behavior

Nach setzen des Limits wird dieses auch eingehalten

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.10.22

### What firmware variant (PIO Environment)?

generic_esp32s3

### Relevant log/trace output

```Shell
Während dem Drücken des Buttons für das permanente Limit sehe ich in der Konsole ( paralleles Browser Fenster ) auch bei erweiterter Protokollierung keine Hinweise dass ein permanentes Limit gesendet wurde.
```

### Anything else?

Hat zu 9x% in der Vorgängerversion auch nicht funktioniert

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

gitisgreat2023, 2024-11-09:
Mir ist nicht klar wie man überhaupt überprüfen kann ob das Limit ankommt. Im Interface kann man die permanente Limits senden ("write")  aber es gibt keinen ("read").

Solarteur, 2024-11-11:
Die Punkte

_- "Limit setzen, CAN Bus nicht verbunden" mit 53V und 1A setzen
-Ladegerät Feature in openDTU deaktivieren
-AC Verbrauch des Huawei observieren -> geht nach einigen Sekunden auf 3kW Ladeleistung
nach Wiederherstellung der CAN Verbindung Ladeleistung unmittelbar wieder geregelt_

kann ich genau so bestätigen.

knopers1, 2024-11-12:
Bei der fehlenden CAN Bus -Verbindung sollte der Power GPIO das Huawei abschalten! Das müsste zu erst implementiert sein.

Solarteur, 2024-11-12:
Den Vorschlag von @knopers1 finde ich mit Blick auf die Sicherheit des gesamten Systems sehr gut, noch besser als (ein funktionierendes) "Limit setzen, CAN Bus nicht verbunden".

gitisgreat2023, 2024-11-12:
> Bei der fehlenden CAN Bus -Verbindung sollte der Power GPIO das Huawei abschalten! Das müsste zu erst implementiert sein, damit überhaupt das Huawei halbwegs sicher betrieben werden kann.

Was genau meinst Du? slot detect oder was anderes?

knopers1, 2024-11-13:
Ja, Slot Detect wird doch auf GnD über das Relai gezogen , damit das Netzteil anläuft. Das Relai selbst zieht es an, wenn Power GPIO angesteuert wurde. Richtig?

Andrix82, 2024-11-13:
Über den PowerPin der OpenDTU wäre ein anderes Thema, würde z.b. in meinem Fall auch gar nichts bringen weil ich den Power Pin garnicht in Verwendung habe mit Absicht. 
Ich schalte nur AC seitig ein zu Tageszeiten.
Vom "Safety-Case" wäre es über den PowerPin auch nur eine halbe Lösung, wenn die openDTU sich dann aufhängt und den Pin beliebig ausgibt, der Pin abfällt, Relaisschaltungsprobleme was auch immer kann der Huawei wieder 3kW ausgeben.
Wirklich sicher wäre nur wenn der Huawei wann auch immer er keine zyklischen gültigen CAN Frames zur Steuerung mehr bekommt selbst ein einen Safe state geht und die Ladeleistung stark begrenzt.

gitisgreat2023, 2024-11-13:
> Ja, Slot Detect wird doch auf GnD über das Relai gezogen , damit das Netzteil anläuft. Das Relai selbst zieht es an, wenn Power GPIO angesteuert wurde. Richtig?

Ja, aber Power GPIO (oder slot detect) wird ja über die PCB geregelt. D.h. wenn der abraucht/sich aufhängt, wird der Huawei nicht geregelt.

Genau für den Fall sind die offline settings von der Huawei sinnvoll... oder?

schlimmchen, 2024-11-13:
> Vom "Safety-Case" wäre es über den verwendeten PowerPin aber auch nur eine halbe Lösung, wenn die openDTU sich dann aufhängt und den Pin beliebig ausgibt, der Pin abfällt, Relaisschaltungsprobleme was auch immer kann/würde der Huawei wieder 3kW ausgeben.

Korrekt.

> Wirklich sicher wäre nur wenn der Huawei wann auch immer er keine zyklischen gültigen CAN Frames zur Steuerung mehr bekommt selbst ein einen Safe state geht und die Ladeleistung stark begrenzt.

Korrekt.

> Das unterstützt der Huawei auch über die Konfiguration eines permanenten Limits, die openDTU will es offensichtlich können sonst gäbe es den Button nicht, aber macht es wie schon mehrere bestätigt haben wohl einfach nicht richtig.

Aha, hier ist also etwas von Seite OpenDTU-OnBattery zu tun? Erklärt mir das mal jemand? Sind das die Offline-Settings, die hier angesprochen wurden? Die werden nicht richtig "eingebrannt" in das Netzteil?

Zusätzlich könnte man dennoch auch das Slot Detect Dingens abschalten, wenn man "lange" nichts vom Netzteil gehört hat.

Andrix82, 2024-11-13:
> Aha, hier ist also etwas von Seite OpenDTU-OnBattery zu tun? Erklärt mir das mal jemand? Sind das die Offline-Settings, die hier angesprochen wurden? Die werden nicht richtig "eingebrannt" in das [...]

Genau die offline Settings welche die openDTU siehe screenshot wohl einstellen können will, funktionieren einfach nicht 
[image]
-> Dies wäre daher wohl eher ein BugFix

Den Power Pin kann man natürlich zusätzlich auch nach CAN-Empfangstimeout huawei->openDTU abschalten
-> new feature

Solarteur, 2024-11-14:
Aus Beitrag  @schlimmchen 

> Aha, hier ist also etwas von Seite OpenDTU-OnBattery zu tun? Erklärt mir das mal jemand? Sind das die Offline-Settings, die hier angesprochen wurden? Die werden nicht richtig "eingebrannt" in das Netzteil?

Ja, ein das Netzteil "einbrennen" so wie hier beschrieben:

https://www.beyondlogic.org/review-huawei-r4850g2-power-supply-53-5vdc-3kw/

Abschnitte "Modes of Operation" und folgende.
.

gitisgreat2023, 2024-11-14:
> Aus Beitrag @schlimmchen
>
>
> Ja, ein das Netzteil "einbrennen" so wie hier beschrieben:
>
> https://www.beyondlogic.org/review-huawei-r4850g2-power-supply-53-5vdc-3kw/
>
> Abschnitte "Modes of Operation" [...]

In dem Text: “ To date, I’m unaware how to read the above parameters back. If you know more, please leave us a comment below.”

schlimmchen, 2024-11-14:
> Genau die offline Settings welche die openDTU siehe screenshot wohl einstellen können will, funktionieren einfach nicht
-> Dies wäre daher wohl eher ein BugFix

Verstanden. Da kann ich nicht helfen. Bock hätte ich, ich schrecke auch nicht zurück die Hardware zu kaufen, aber die Zeit fehlt. Kann das jemand anders (nicht Andreas) versuchen umzusetzen bzw. zu reparieren?

> Den Power Pin kann man natürlich zusätzlich auch nach CAN-Empfangstimeout huawei->openDTU abschalten
-> new feature

Verstanden. @Andrix82 Kannst du bitte ein passendes Issue (Feature Request) erstellen?

gitisgreat2023, 2024-11-14:
@MalteS kennst Du dieses Thema? Hast Du es mal ausprobiert, das setzen vom offline limit?

MalteSchm, 2024-11-14:
Danke fürs taggen
Ich bin da ehrlich gesagt nicht sicher. Aber das ist auch eine Weile her.
Ich glaube das ich damals angenommen habe das das offline Limit die Einschaltbegrenzug festlegt. Das es die Werte festlegt die verwendet werden wenn der CAN bus ausfällt habe ich glaube erst später verstanden. 
Wenn ich danach schauen soll: ich bin gerade sehr eingespannt. Vermutlich geht das erst im Weihnachtsurlaub

gitisgreat2023, 2025-01-15:
@MalteSchm 
Gestern lief der Huawei mal wieder auf 3kW weil die DTU mal aus war... das erinnerte mich dran dass mein offline limit anscheinend auf 3kW effektiv steht... :-(

Daher: es wäre super wenn du das Thema aufnehmen würdest. Ich kenne mich mit dem Thema nicht aus, aber habe die Doku und code mal angeschaut um zu sehen ob was offensichtlich nicht passst. Dass ist leider nicht der Fall.... :-(

Ich habe mir die Doku mal angeschaut:
CAN R4850G2 Protocol:
[image]

[Hier](https://github.com/hoylabs/OpenDTU-OnBattery/blob/development/include/gridcharger/huawei/HardwareInterface.h) steht es wie erwartet: 
[image]

[image]

Dann man die [GUI](https://github.com/hoylabs/OpenDTU-OnBattery/blob/development/src/WebApi_Huawei.cpp) oberflächlich angeschaut. Fällt mir auch nicht direkt was auf.
[image]
Weiss natürlich nicht ob das event "offline" gut verarbeitet wird.

Noch zwei Verständnisfragen: der XLS sagt multiplier 1020 für voltage und 30 für current, aber der code nutzt 1024 und 20.
1) [Hier]([OpenDTU-OnBattery](https://github.com/hoylabs/OpenDTU-OnBattery/tree/development)/[src](https://github.com/hoylabs/OpenDTU-OnBattery/tree/development/src)/[gridcharger](https://github.com/hoylabs/OpenDTU-OnBattery/tree/development/src/gridcharger)/[huawei](https://github.com/hoylabs/OpenDTU-OnBattery/tree/development/src/gridcharger/huawei)
/HardwareInterface.cpp)  müsste 1024 (Zeile 164) nicht 1020 sein? (Zeile 28 und 29 von der XLS Liste)

2)  [hier](https://github.com/hoylabs/OpenDTU-OnBattery/blob/development/include/gridcharger/huawei/HardwareInterface.h) steht static unsigned constexpr _maxCurrentMultiplier = 20;, wo die Doku XLS Zeile 31,32 30 sagt?!

MalteSchm, 2025-01-15:
Die Skalierungsfaktoren habe ich damals von https://www.beyondlogic.org/review-huawei-r4850g2-power-supply-53-5vdc-3kw/ übernommen. Das hatte gepasst. Kann aber für die Spannung falsch sein (da ist der Unterschied klein)
Strom passt sicher. Der Faktor 20 vs. 30 ist zu groß und ich sehe im BMS was in die Batterie rein geht.

Online limits müssten sich eigentlich setzen lassen. Das ging zumindest bei mir jetzt immer. Reagiert das Netzteil denn wenn du das setzt?

Zum Offline Limit: Da ist mein Verständnis das das erst greift wenn
* Du Netzteil und DTU gestartet hast
* Die Offline limits gesetzt hast
* Dann die DTU ausschaltest (CAN Bus trennst)
* Dann das Netzteil einen CAN Bus Fehler erkennt (dauert etwas) 

War das bei dir in diesem Fall so der Fall? Das Limit wird nicht im Netzteil gespeichert und greift daher auch nicht wenn du nur das Netzteil anschaltest.

Im Moment etwas zu machen ist gerade extrem langsam. Ich habe einen neuen Rechner bekommen und es immerhin jetzt geschafft zumindest den Code auszuchecken. Bauen geht aber noch nicht. Dann ist mir gerade auch nicht klar wie ich testen kann weil mein ESP noch eine Version mit dem kleinen Speicher ist.

schlimmchen, 2025-01-15:
> Der Faktor 20 vs. 30 ist zu groß und ich sehe im BMS was in die Batterie rein geht.

Fürs R4850 passt das ganz sicher, das konnte ich auch schon nachmessen.

> Das ging zumindest bei mir jetzt immer.

Online Limits setzen geht definitiv.

Offline Limits... Ich kanns leider nicht testen bisher, siehe unten.

Jedenfalls bin ich mir einigermaßen sicher, dass es einen Fehler gab:
https://github.com/hoylabs/OpenDTU-OnBattery/blob/81b1e6e158ff90c2b9e7c9c5b7617571a0714cfc/src/Huawei_can.cpp#L179-L186

Der `parameterType` mit Wert `HUAWEI_OFFLINE_CURRENT` wurde nie gesetzt, weil in dem if-statement ein `<=` hätte stehen müssen, statt ein `<`.

Wenn es jetzt einen Fehler gibt, der verhindert, dass offline limits gesetzt werden, dann ist es ein anderer.

> Dann ist mir gerade auch nicht klar wie ich testen kann weil mein ESP noch eine Version mit dem kleinen Speicher ist.

Mit PIO Umgebung `generic_esp32_4mb_no_ota` und dem ESP per USB-Kabel am Rechner. Ich würde aber auch nicht zögern dich auszustatten mit neuer Hardware, wenn du denn Bock drauf hast.

Hast du denn schon bemerkt, dass ich den Code erst kürzlich stark verändert habe?

Darf ich zwei Fragen an dich richten?
1. Ist es korrekt, dass das Netzteil bei kleinen Leistungen (~200W) nicht lange läuft? Meins schaltet sich stets nach einer kurzen Weile ab. Ich versuche Tests auf dem Schreibtisch zu machen mit kleinen Lasten, das scheint aber schlicht nicht zu klappen.
2. Du hast nicht zufällig eine Idee, wie ich ne gescheite Last ans Netzteil klemmen kann, ohne gleich eine Batterie aufzubauen? Beim Testen und probieren passt es mir eigentlich nicht, dass ich das an meinem Produktivsystem tun soll...

schlimmchen, 2025-01-16:
> Eigentlich denke ich das wir die offline Werte nicht mehr extern setzten sollten sondern den offline current hart auf 0 setzten.

Ja, das ist ein sehr guter Gedanke. Für den Fall, dass sich Nutzer beschweren, kann mir gut die offline Einstellungen als Einstellungen des AC Chargers in der Web UI vorstellen. Jedenfalls ist dann das Modal im Live View nur noch dazu da, die online Limits zu setzen, was sehr gut (aufgeräumt) wäre.

> Das Thema mit dem ESP war vor allem das ich einen mit externer Antenne verwende.

Wenn ich dir kein Fusion mit CAN/Iso Shield andrehen kann, dann sucht du vermutlich [sowas](https://www.conrad.de/de/p/espressif-esp32-devkitc-vie-entwicklungsboard-esp32-devkitc-vie-2490159.html). Soll ich dir eins schicken und du machst, dass ich was (alte Modellbauakkus) habe, um das Netzteil auch auf dem Schreibtisch zumindest für ein paar Minuten testen zu können? E-Mailadresse findest du im Git log.

gitisgreat2023, 2025-01-16:
| Eigentlich denke ich das wir die offline Werte nicht mehr extern setzten sollten sondern den offline current hart auf 0 setzten.
Finde ich ein super Vorschlag, V offline 48V und I offline 0A. Das wäre sauber. Vielleicht am besten mit einem Status in der GUI dass es tatsächlich geklappt hat (zB bei mir müsste online setzen gehen mittels GUI geht aber nicht).

DPL und auch MQTT kann die online limits bei mir ohne Problem setzen.
Die zwei Fehler die ich bekomme, bei online und offline das Limit setzen, passieren wenn ich es händisch per GUI machen möchte:
[image]

Kann jemand bestätigen dass mittels diesen GUI sich die online limits setzen lassen? Gerne auch mit log aus der Konsole...

@schlimmchen ich habe versucht zurückzuverfolgen wo der Fehler `[HuaweiCanClass::loop] Data set error` ausgespuckt wird, wurde aber nicht fündig.
Ich sehe Huawei_can.cpp gar nicht https://github.com/hoylabs/OpenDTU-OnBattery/tree/development/src , arbeitest du gerade an dem Code?

Bgzl Regelverhalten: bei mir regelt er sauber ab 50W.

Ich kann gerne auch builds testen, allerdings habe ich noch einen 4MB Version also muss jedes Mal mühsam es flashen... leider gibt es noch keinen Board welche einen Huawei und HMT und JK BMS (eventuell Pylontech CAN emulation) steuern kann. Oder kann das Fusion diese dreier Kombi mitterweile?

Andrix82, 2025-01-16:
!!!STOP Leute STOP!!!

Was auch immer ihr im Detail an der R4850G2 Implementierung geändert habt mit dem letzten Release, es funktioniert nun! :D

Mein oben beschriebener Testfall funktioniert nun und aus meiner Sicht als "Ticketersteller" kann dies geschlossen werden.

Folgendes habe ich eben getestet:

Randbedingung: Batteriespannung ca. 52V, keine Ladeanforderung seitens openDTU( AC Leistungsaufnahme R4850G2 ~5W )
Offlinelimits setzen mit 51V, 1A
AC Ladegerät in openDTU deaktivieren
=> gemessene AC Leistungsaufnahme des R4850G2 bleibt über längeren Zeitraum(mehrere Minuten) bei den Standard 5W

AC Ladergeät in openDTU aktivieren
=> verbindet wieder problemlos und kann problemlos Werte lesen, AC Leistungsaufnahme bleibt ~5W

Offlinelimit Setzen mit 54V, 1A
AC Ladegerät in openDTU deaktivieren
=> gemessene AC Leistungsaufnahme des R4850G2 geht nach ~ 1 Minute auf 50W
AC Ladegerät in openDTU aktivieren
=> verbindet wieder problemlos und kann problemlos Werte lesen, AC Leistungsaufnahme fällt wieder auf ~5W

Danke!! 👍

closed 2025-01-16 by schlimmchen

schlimmchen, 2025-01-17:
Info: Man darf den Lüfter **NICHT** abklemmen.

Das Netzteil startet auch ohne, aber dann passiert, was ich beschrieben habe: Es schaltet sich nach kurzer Zeit wieder ab.

Also mein Problem ist damit teilweise gelöst. Es ist nur so, dass der Lüfter unerträglich laut ist, jedenfalls in meinem Exemplar. Vielleicht kann ich den noch "retten", zumindest für eine Weile...

gitisgreat2023, 2025-01-18:
Bei mir heult er auch auf, aber kurz drauf geht die Drehzahl runter und passt sich zu Last an. Warte einfach 1-2 Minuten, dann müsste es besser werden.

Ich habe noch mal getestet, er funktioniert sogar bis etwa 30 W.
Mit dem Lüfter aktiv, schafft er jetzt stabil die 50W zu liefern?

> Info: Man darf den Lüfter **NICHT** abklemmen.
>
> Das Netzteil startet auch ohne, aber dann passiert, was ich beschrieben habe: Es schaltet sich nach kurzer Zeit wieder ab.
>
> Also mein Problem ist [...]

schlimmchen, 2025-01-19:
> Mit dem Lüfter aktiv, schafft er jetzt stabil die 50W zu liefern?

So ist es.

Ich habe einen neuen Lüfter bestellt. Der Stecker ist nicht kompatibel, aber das kann man ja leicht anpassen. Der jetzige ist jedenfalls mechanisch defekt, so laut sollte er dann trotz allem nicht sein. Es ist nicht auszuhalten. Mit etwas Öl und Kratzern in der Messing-Buchse (um diese zu verengen) habe ich ihn für ein paar Versuche nochmal auf ein erträgliches Geräuschlevel gebracht.

gitisgreat2023, 2025-01-20:
Ich habe leider ein 4MB ESP, also muss es händisch flashen bevor ich den letzten release testen kann. Dann melde mich mal ob ich deien Ergebnisse reproduzieren kann... 

> !!!STOP Leute STOP!!!
>
> Was auch immer ihr im Detail an der R4850G2 Implementierung geändert habt mit dem letzten Release, es funktioniert nun! :D
>
> Mein oben beschriebener Testfall funktioniert nun [...]

gitisgreat2023, 2025-01-22:
Also 2025.01.10 ist drauf... aber beim offline limit setzen gibt er im log einen Fehler:

`21:08:52.539 > [Huawei::HwIfc] Failed to set parameter`

Eine Sekunde später lieset er erfolgreich den Status aus

`21:08:53.740 > [Huawei::HwIfc] [208153.781] InputTemperature: 23.00°C
21:08:53.798 > [Huawei::HwIfc] [208153.781] OutputTemperature: 13.00°C
21:08:53.859 > [Huawei::HwIfc] [208153.781] InputVoltage: 227.66V
21:08:53.922 > [Huawei::HwIfc] [208153.781] OutputCurrentMax: 0.00A
21:08:53.970 > [Huawei::HwIfc] [208153.766] InputFrequency: 50.10Hz
21:08:54.089 > [Huawei::HwIfc] [208153.781] OutputCurrent: 0.00A
21:08:54.137 > [Huawei::HwIfc] [208153.766] InputPower: 0.00W
21:08:54.199 > [Huawei::HwIfc] [208153.766] InputCurrent: 0.00A
21:08:54.262 > [Huawei::HwIfc] [208153.766] OutputPower: 0.00W
21:08:54.395 > [Huawei::HwIfc] [208153.766] Efficiency: 0.00%
21:08:54.445 > [Huawei::HwIfc] [208153.781] OutputVoltage: 52.69V
`

@Andrix82 kannst du deinen Log posten wenn du den offline limit setzt?

schlimmchen, 2025-01-22:
> kannst du deinen Log posten wenn du den offline limit setzt?

Da steht dann nichts -- es sei denn er hat den gleichen Fehler "Failed to set parameter". Ich hab den nicht...

Du spricht aber (bisher) mit einem R4850, kein R4875? Oder R4830? Wenn der maximale Ausgangsstrom falsch skaliert wird, könnte das ja die Fehlermeldung erklären.

gitisgreat2023, 2025-01-22:
> Da steht dann nichts -- es sei denn er hat den gleichen Fehler "Failed to set parameter". Ich hab den nicht...
>
> Du spricht aber (bisher) mit einem R4850, kein R4875? Oder R4830? Wenn der maximale [...]

R4850G2.
Weiter wird er perfekt geregelt, Strom Spannung AC und DC passen, Effizienz, alles gemessen und überprüft.