hoylabs/OpenDTU-OnBattery #1386: [Request] make DPL adjustment possible with MQTT

ThomasCr, 2024-11-12:
### Describe the solution you'd like

It would be nice to make the DPL settings adjustable from Mqtt and so also by Home Assistant. So we can create automations for eg set a lower `Minimum Power Limit` or a other `Maximum Power Limit` when you want it.

spcqike, 2024-11-12:
is this really needed? you allready can set it with http requests, check out the [web-api](https://opendtu-onbattery.net/firmware/web_api/)

and keep in mind, that changing the DPL settings will write to flash, as these are persistent settings. so its not helpful to change it several times per day.

anyway, could you describe a usecase, where you need to change the settings every now and then?

ThomasCr, 2024-11-12:
As an example, If weather Forcast tells me, we don't get so much sun, I would change the max/min Power Limit, so that the battery can last longer.

But it is good to know, that changes are written to the flash. Maybe implement them as temporary changes on top the permanent ones?

To use the web api we need again to set up a other access method, that's not so nice.

spcqike, 2024-11-13:
well, i think i kind of understand why you would like to change the max setting, but i dont get why you want to change min, too.

on the other hand, seen from energy point of view, it doesn't matter anyway. if you produce like 200W for 4h or 100W for 8h, its still 800Wh. 

what do you think would be the benefit of having a ..... say "second and soft max limit" (which could be retrieved by mqtt and is non persistant, so not in eeprom)?

as a workaround, you could "fake" your powermeter reading. if you send this via mqtt and have an automation system like node red or HA (which you need for your forcast related calculation, i guess), you can change the reading to your wishes.

spcqike, 2024-11-14:
so we allready can change upper limit and target consumption? i wasn't aware of it, or i forgot :D 

if we use this, is it persistent?

schlimmchen, 2024-11-14:
The missing part is HASS autodiscovery, but yeah, those topics have been around and in use for quite some time now.

> if we use this, is it persistent?

Yes. I hate that it is, but yes.

AndreasBoehm, 2024-12-07:
I think this is a duplicate of https://github.com/hoylabs/OpenDTU-OnBattery/issues/1178

Can we close this if you agree that its a duplicate @schlimmchen ?

closed 2024-12-07 by schlimmchen