hoylabs/OpenDTU-OnBattery #1387: Eingabe für Full-Solar-Passtrough

hoschiking, 2024-11-12:
### What happened?

Ich nutze OpenDTUoB der VE.Direct Schnittstellen, somit ist Solar-Passtrough nicht nutzbar.
Ich muß jedoch bei der Konfiguration des DPL jedes mal sinnvolle Werte für Full-Solar-Passtrough angeben, damit gemachte Eingaben übernommen werden. Beim nächsten Aufruf stehen die Werte wieder auf Standard (100/100), da diese nicht übernommen werden, da Full-Passtrough nicht nutzbar.

### To Reproduce Bug

Den DPL ohne Aktivierung der VE.Direct Schnittstelle parametrieren, schon ist der Bug sichtbar

### Expected Behavior

Die Abfrage der Werte für Full-Solar-Passtrough ausblenden und nicht abfragen, wenn diese Funktion nicht genutzt werden kann.
[image]

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.10.22

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

gitisgreat2023, 2024-11-13:
Ich war gerade dabei ein bug report zu erstellen... zum gleichen Thema.

Allerdings: bei mir gibt's die 3 Felder bei esp32s3_usb variante, aber nicht bei meinem esp32_4mb_no_ota.

Bei beide ist VE.Direct nicht aktiv.

hoschiking, 2024-11-13:
Ich habe keine Ahnung wovon das abhängig ist, aber genau diese Parameter sollten doch nicht davon abhängig sein, wie groß der flash und dem zufolge welches Image genutzt wird. Vielleicht ist bei dir in diesem Fall ein anderer Parameter anders, soc ignorieren oder ähnliches....
[image]
[image]

schlimmchen, 2024-11-13:
## Workaround

* VE.Direct einschalten
* In den DPL Einstellungen diese beiden Schwellwerte auf 66 stellen
* In den DPL Einstellungen Solar-Passthrough ausschalten
* DPL Einstellungen speichern
* VE.Direct ausschalten

Anschließend sind diese Werte auf 66 eingestellt und das Formular kann abgeschickt werden, ohne jedes Mal 66 in diese Felder eintragen zu müssen. Weil Solar-Passthrough jetzt auch "offiziell" aus ist, werden diese Schwellwerte, wie ihr es erwartet, auch nicht angezeigt.

## Analyse

Es ist mir sofort gelungen, das nachzustellen. Ihr habt völlig recht, da ist ein Fehler.

https://github.com/hoylabs/OpenDTU-OnBattery/blob/7e19f196551150e38064a8aa0d1497181aa8ccf8/webapp/src/views/PowerLimiterAdminView.vue#L515-L517

Die Einstellung "Solar Passthrough" ist im Default eingeschaltet. Das an sich ist schon eine fragliche Sache, aber das ist jetzt so bei vielen Benutzern, insbesondere bei euch.

Die Bedingung in der WebApp muss auch berücksichtigen, ob VE.Direct überhaupt eingeschaltet ist.

Andernfalls funktioniert auch der Workaround nicht, diese beiden Werte auf 66 zu stellen, um Full Solar Passthrough zu deaktivieren, denn diese Werte werden nicht übernommen beim Speichern, weil VE.Direct aus ist. Also sind sie wieder 100 (was in babb24a behoben wurde) und beim nächsten Mal habt ihr wieder Spaß. Das ist natürlich erst recht doof.

Das zu reparieren wird aber einen Konflikt verursachen mit #1216, und außerdem habe ich das mit #1216 schon repariert. Das muss mir beim Umbau der UI schon von selbst gekommen sein, dass diese Bedingung mit hinein muss.

https://github.com/hoylabs/OpenDTU-OnBattery/blob/28302fe7f072eb1c455c53fbfb4afe0646eceda0/webapp/src/views/PowerLimiterAdminView.vue#L485-L490

Daher erlaubt bitte, dass das erst im Release repariert sein wird, wenn #1216 ge-merge't wurde.

hoschiking, 2024-11-13:
Danke für den workaround, hat funktioniert :-)
Allerdings musste ich lediglich VE.Direct einschalten,  dann im DPL Solar-Passtrough ausschalten und dann wieder VE. Direct ausschalten, schon war die Abfrage der Full-Solar-Passtrough Spannungen im DPL nicht mehr vorhanden. Eine Eingabe von Schwellwerten war nicht notwendig.

closed 2024-11-17 by schlimmchen