hoylabs/OpenDTU-OnBattery #1399: [Request] Base load corrected voltage is on inverter DC input power

schlimmchen, 2024-11-16:
### Is your feature request related to a problem? Please describe.

The load corrected voltage is currently calculated based on the AC output power of the inverter. However, the inverter's AC output power has little to do with either the battery voltage nor the battery current draw.

The only thing we are interested in is the voltage drop of the battery due to its load. This allows us to delay stopping the discharge beyond the stop threshold. Once discharging stops, we can expect the battery voltage to bounce back up to the stop threshold.

The voltage drop is proportional to the current draw, as the (internal) resistance that causes the voltage drop is constant (we ignore the battery's temperature for now).

Hence, what we actually need to know to calculate a load corrected voltage is the battery's current draw. The inverter's AC output power does not correlate well with the battery current draw, as the battery voltage can be significantly different. We could ignore that as we use the load corrected voltage only when the battery is near the stop voltage threshold, so it is okay if the load correction factor is tuned based on that one battery voltage.

We then still have the inverter's efficiency to deal with, which depends on its output power and hence can be different each time when approaching the stop threshold, in which case the battery current draw will also be different.

### Describe the solution you'd like

For that reason, we should calculate the load corrected voltage based on the inverter's DC input power (not AC output power).

### Additional context

If we want to take this a step further, we should base the load corrected voltage offset/diff solely on the DC current at the inverter's inputs. This would allow us to estimate the battery's idle voltage using the same load correction factor at different battery voltages. We then need to convert the existing load correction factor. As long as we only need the load corrected voltage in the context of one battery voltage (the stop threshold), we might as well keep using the DC input power.

schlimmchen, 2024-11-30:
> For that reason, we should calculate the load corrected voltage based on the inverter's DC input power (not AC output power).

Depending on the situation, this also does not make sense. The battery could actually be in an unexpected state, e.g., the battery could actually be idle as the power the inverters consume is actually provided by a solar charger, or mostly by a solar charger, so the corrected voltage is too high. I mentioned this already in #676.

There is also #874.

The idea in #1111, which gets rid of the load-correction factor altogether, seems to be the best option to me, especially since setups without solar charge controller data and without a battery interface provide very limited means to mitigate the issues.

SW-Niko, 2024-12-23:
If you want to do it right, you have to use the battery voltage and the battery current for the calculation of the battery open circuit voltage (load corrected voltage).

[image]

For a 48VDC system with an 800W inverter, a solar power of 2000W and an internal battery resistance of 18mOhm, you can calculate following values:
With an open circuit voltage of: 52.00VDC 
Charging with 2000W: 52.69VDC 
Discharging with 800W: 51.73VDC 
Compensation still makes sense and you can reach the stop and start voltage much better if you use the battery open circuit voltage (load corrected voltage). 

I would suggest offering the compensation/calculation of the battery's open circuit voltage as an additional feature and keeping the current solution as a backup for all systems without access to battery voltage and current. In PR #1466  I have implemented this behavior for the battery provider “Smart Shunt”.

>The idea in https://github.com/hoylabs/OpenDTU-OnBattery/issues/1111, which gets rid of the load-correction factor altogether, seems to be the best option to me.

I'm still working on this feature. But in the meantime I realized we need both features. Because #1111 cannot improve the start voltage behavior.

gitisgreat2023, 2025-01-20:
Yesterday I noticed that DPL switched too late off based on the corrected battery voltage.
Looking at the log I discovered the correction voltage is based on the inverter output. However I have two inverters (controlled by two separate boards) connected to the battery. The load distribution is dynamic over both inverters as well.
So there is no obvious way for me to fix this issue.

Would it be possible, if the battery current/power is available, to base the correction voltage on that value? That would also eliminate other issues like solar charge controllers connected to the battery, while inverters are drawing power as well from the battery.

> Depending on the situation, this also does not make sense. The battery could actually be in an unexpected state, e.g., the battery could actually be idle as the power the inverters consume is [...]

gitisgreat2023, 2025-01-20:
@spcqike Yes JK BMS via serial.

The issue is in the DPL section. The Load correction factor is applied on the power drawn by the inverter, not the amount totally drawn from/supplied to the battery.

schlimmchen, 2025-02-01:
* If the battery does not report current readings,
    * base the load-corrected voltage on the DC input current of the inverter(s).
    * only use the load-corrected voltage when testing the stop threshold.
* If the battery does report current readings, using the current (with the correct sign) will fix the issue where the load-corrected voltage should be higher, but it currently isn't.