hoylabs/OpenDTU-OnBattery #1407: [Request] Validate usage of SunPosition.isSunsetAvailable()

AndreasBoehm, 2024-11-19:
### Is your feature request related to a problem? Please describe.

When i implemented the use of `SunPosition.isSunsetAvailable()` I assumed that this reports if the time of the ESP was set and is able to report day/night.

As it turns out this actually reports if sunset is a thing in the current timezone or if it never gets dark during the 'night'.

### Describe the solution you'd like

1. Document what `SunPosition.isSunsetAvailable()` actually returns.
2. Check usages of `SunPosition.isSunsetAvailable()` and make sure its correctly used.

### Additional context

Thomas explained pretty well what this method actually tells us: https://github.com/tbnobody/OpenDTU/pull/2198#issuecomment-2460792251

stefan123t, 2024-11-19:
@AndreasBoehm

there is a already a comment in SunPosition.cpp

https://github.com/hoylabs/OpenDTU-OnBattery/blob/61aa32ab5e5d0bfd876a4b008342c6b54f82d0d0/src/SunPosition.cpp#L110,L118

Wouldn‘t that be sufficient ?

If sunrise / sunset calculation fails, eg. too far north / south the sunset functionality is not available.

AndreasBoehm, 2024-11-19:
The problem is that this comment is not on the getter and thats also the reason why i interpreted the result wrong.

stefan123t, 2024-11-19:
Agreed, how do we document such details normally, do we use some kind of documentation automation like doxygen ?

stefan123t, 2024-11-19:
@spcqike sounds like a good plan for the Vue.js code of the Web App. 

We could use JSDoc for the Web App and Doxygen for the C++ code.

Here is a Stack Overflow Entry to use Doxygen for both [Generate JavaScript documentation with Doxygen](https://stackoverflow.com/questions/7320464/generate-javascript-documentation-with-doxygen)
The additional benefit would be only one documentation for both parts of the code being produced by one generator.

@AndreasBoehm sorry to sort of high-jack your issue for enhancing the documentation / auto-generate docs:

I have also tried to write an Swagger / OpenAPI specification for OpenDTU in https://github.com/tbnobody/OpenDTU/pull/2410
You may find my first stab at the OpenDTU-OnBattery REST API specs here: https://github.com/tbnobody/OpenDTU/issues/378#issuecomment-2227470895

It would be great if we could auto-generate these API specs from the REST API inline documentation too.
But apparently the Feature Request for Doygen to support OpenAPI is still open: [Compatibility with or support for REST APIs swagger / openAPI](https://github.com/doxygen/doxygen/issues/6613)

closed 2024-12-07 by AndreasBoehm

schlimmchen, 2024-12-07:
Guys, please relax. Machine-readable code documentation is close to the end of the "maybe-we-do-that-if-we-ever-find-the-time-wishlist".

Andreas' added comment is important here because this is "hard" to wrap your head around this.