hoylabs/OpenDTU-OnBattery #1409: [Request] Different DPL strategies for handling multiple inverters

schlimmchen, 2024-11-19:
### Is your feature request related to a problem? Please describe.

The implementation merged in #1216 prefers to change the limit of those inverters which promise to implement the needed change in AC output with a single limit update. If none can implement the change in one go, the inverters are sorted by the contribution they can make to the required change.

This tries to make sure that the DPL reacts as quickly as possible.

This also means that the DPL is happy to put inverters online, but hesitant to put them into standby.

However, people objected that this can lead to less than ideal situations, where multiple inverters are online producing at less efficiency while less inverters could be producing at the same total power with higher efficiency.

### Describe the solution you'd like

Without too much trouble, we could implement different strategies in the DPL. The respective function implementing a particular strategy would always receive a bunch of inverters as input as well as the desired change in output, and it would be responsible to determine which of those inverters are supposed to change their limit by how much.

### Additional context

I am happy to help implement different strategies and make them selectable (config switch, UI changes, etc.). If you think the DPL should implement a different strategy, please see
https://github.com/hoylabs/OpenDTU-OnBattery/blob/61aa32ab5e5d0bfd876a4b008342c6b54f82d0d0/src/PowerLimiter.cpp#L546-L588
for inspiration and describe the method you would like to see the DPL using. The more technical (flow-chart, pseudo-code) and detailed, the better.

Manos1966, 2024-11-21:
Hi @schlimmchen suchst du weiterhin nach [Tersters]?
(ich habe eine Test-Installation mit 2-3 Hoymiles online).

schlimmchen, 2024-11-21:
Naja, jetzt ist die Katze ja aus dem Sack. Unsere Nutzer werden das neue Release sicherlich ausprobieren und wir hören sicherlich, wenn es Probleme gibt.

spcqike, 2024-11-21:
Ich hab aktuell nur 1 wechselrichter und viel zu wenig Sonne. 

Das Feature testen werde ich dann ab April 2025 😄

Manos1966, 2024-11-21:
Jesus! 😱 😱 😱 😱 😱 
[image]

🥇 🥇 🥇 🥇 🥇 🥇 🥇 🥇 
🥂 🥂 🥂 🥂 🥂 🥂 🥂 🥂

andy75andreas, 2024-11-26:
Hallo,
irgendwie geht da was nicht mit der 20.11.2024.

ESP32-S3 8MB, Oled,Pylontec,Victron,SML,(W5500)

Ich habe alles so eingestellt wie in der Version vorher, aber es wird kein Strom produziert.
Was ist hier falsch?

Dump von der Konsole:
`[VE.Direct MPPT 12/11] serial input (242 Bytes):
19:36:24.441 > [VE.Direct MPPT 12/11] 3a 37 46 36 45 44 30 30 31 38 31 35 33 45 0a 3a
19:36:24.504 > [VE.Direct MPPT 12/11] 37 32 37 32 30 30 30 46 46 46 46 46 46 46 46 30
19:36:24.565 > [VE.Direct MPPT 12/11] 42 0a 3a 41 44 35 45 44 30 30 44 34 31 34 41 31
19:36:24.631 > [VE.Direct MPPT 12/11] 0a 3a 41 42 42 45 44 30 30 31 32 30 30 39 31 0a
19:36:24.682 > [VE.Direct MPPT 12/11] 0d 0a 50 49 44 09 30 78 41 30 37 33 0d 0a 46 57
19:36:24.738 > [VE.Direct MPPT 12/11] 09 31 36 34 0d 0a 53 45 52 23 09 48 51 32 32 34
19:36:24.804 > [VE.Direct MPPT 12/11] 38 48 39 41 34 33 0d 0a 56 09 35 33 33 32 30 0d
19:36:24.868 > [VE.Direct MPPT 12/11] 0a 49 09 30 0d 0a 56 50 56 09 31 39 30 0d 0a 50
19:36:24.919 > [VE.Direct MPPT 12/11] 50 56 09 30 0d 0a 43 53 09 30 0d 0a 4d 50 50 54
19:36:24.978 > [VE.Direct MPPT 12/11] 09 30 0d 0a 4f 52 09 30 78 30 30 30 30 30 30 30
19:36:25.033 > [VE.Direct MPPT 12/11] 31 0d 0a 45 52 52 09 30 0d 0a 4c 4f 41 44 09 4f
19:36:25.197 > [VE.Direct MPPT 12/11] 4e 0d 0a 48 31 39 09 35 34 38 35 0d 0a 48 32 30
19:36:25.255 > [VE.Direct MPPT 12/11] 09 39 30 0d 0a 48 32 31 09 33 30 38 0d 0a 48 32
19:36:25.327 > [VE.Direct MPPT 12/11] 32 09 37 39 0d 0a 48 32 33 09 32 36 35 0d 0a 48
19:36:25.387 > [VE.Direct MPPT 12/11] 53 44 53 09 34 37 0d 0a 43 68 65 63 6b 73 75 6d
19:36:25.437 > [VE.Direct MPPT 12/11] 09 f3
19:36:25.491 > [VE.Direct MPPT 12/11] Text Data 'PID' = '0XA073'
19:36:25.610 > [VE.Direct MPPT 12/11] Text Data 'FW' = '164'
19:36:25.656 > [VE.Direct MPPT 12/11] Text Data 'SER' = 'HQ2248------'
19:36:25.704 > [VE.Direct MPPT 12/11] Text Data 'V' = '53320'
19:36:25.749 > [VE.Direct MPPT 12/11] Text Data 'I' = '0'
19:36:25.797 > [VE.Direct MPPT 12/11] Text Data 'VPV' = '190'
19:36:25.853 > [VE.Direct MPPT 12/11] Text Data 'PPV' = '0'
19:36:25.909 > [VE.Direct MPPT 12/11] Text Data 'CS' = '0'
19:36:26.028 > [VE.Direct MPPT 12/11] Text Data 'MPPT' = '0'
19:36:26.089 > [VE.Direct MPPT 12/11] Text Data 'OR' = '0X00000001'
19:36:26.141 > [VE.Direct MPPT 12/11] Text Data 'ERR' = '0'
19:36:26.200 > [VE.Direct MPPT 12/11] Text Data 'LOAD' = 'ON'
19:36:26.283 > [VE.Direct MPPT 12/11] Text Data 'H19' = '5485'
19:36:26.346 > [VE.Direct MPPT 12/11] Text Data 'H20' = '90'
19:36:26.404 > [VE.Direct MPPT 12/11] Text Data 'H21' = '308'
19:36:26.463 > [VE.Direct MPPT 12/11] Text Data 'H22' = '79'
19:36:26.528 > [VE.Direct MPPT 12/11] Text Data 'H23' = '265'
19:36:26.604 > [VE.Direct MPPT 12/11] Text Data 'HSDS' = '47'
19:36:26.683 > [DPL::loop] ******************* ENTER **********************
19:36:26.762 > [DPL::loop] battery interface enabled, SoC: 99.0 %, StartTH: 90 %, StopTH: 20 %, SoC age: 1 s, ignore: no
19:36:26.852 > [DPL::getBatteryVoltage] BMS: 53.25 V, MPPT: 53.32 V, inverter 1164a------------: 53.40 V, returning: 53.25V
19:36:26.913 > [DPL::loop] dcVoltage: 53.25 V, loadCorrectedVoltage: 53.25 V, StartTH: 53.00 V, StopTH: 49.00 V
19:36:26.982 > [DPL::loop] StartTH reached: yes, StopTH reached: no, SolarPT disabled, use at night: yes
19:36:27.041 > [DPL::calcHouseholdConsumption] target consumption: 0 W, base load: 10 W
19:36:27.112 > [DPL::calcHouseholdConsumption] power meter value: 405.0 W, power meter valid: yes
19:36:27.174 > [DPL::calcHouseholdConsumption] inverter 1164a------------- is behind power meter producing 1 W
19:36:27.237 > [DPL::updateInverterLimits] requested: 406 W, producing: 0 W using 0 solar-powered inverters, diff: 406 W, hysteresis: 30 W
19:36:27.297 > [DPL::calcBatteryAllowance] power requested: 406 W
19:36:27.461 > [DPL::updateInverterLimits] requested: 406 W, producing: 1 W using 1 battery-powered inverters, diff: 405 W, hysteresis: 30 W
19:36:27.520 > [DPL::updateInverterLimits] will cover 1 W using battery-powered inverters
19:36:27.587 > [DPL inverter 1164a---------]:
19:36:27.641 >     battery-powered, producing 1 W
19:36:27.707 >     lower/current/upper limit: 10/0/500 W, output capability: 1600 W
19:36:27.771 >     sending commands enabled, reachable, disqualified
19:36:27.843 >     max reduction production/standby: 0/0 W, max increase: 0 W
19:36:27.899 >     target limit/output/state: -1 W (unchanged)/1 W/unchanged, 0 update timeouts
19:36:27.957 > [DPL::loop] consumption: 406 W, target output: 406 W (limited to 1500 W), solar inverters output: 0 W, battery allowance: 406 W, battery inverters output: 1 W
19:36:28.009 > [DPL::announceStatus] the system is stable, the last power limit is still valid
19:36:28.066 > [VE.Direct MPPT 12/11] Unhandled Hex Async Response, addr: 0xEDD5 (Charger Voltage), value: 0x000014D5, flags: 0x00
19:36:28.123 > [VE.Direct MPPT 12/11] Unhandled Hex Async Response, addr: 0xEDBB (Panel Voltage), value: 0x00000012, flags: 0x00
19:36:28.275 > Fetch inverter: 1164A-----------
19:36:28.323 > TX RealTimeRunData 865.00 MHz --> 15 A0 09 93 20 80 19 36 00 80 0B 00 67 46 15 29 00 00 00 00 00 00 00 00 DC 7D 97 
19:36:28.381 > [PowerMeterSerialSml] decoded energy import to -0.10
19:36:28.443 > [PowerMeterSerialSml] decoded energy import to 1514999.75
[... 300 lines cut ...]
    },
    "display": {
		"type": 3,
		"data": 9,
		"clk": 8,
		"cs": -1,
		"reset": -1
    },
    "victron": {
		"rx": 12,
		"tx": 11,
		"rx1": 10,
		"tx1": 17
    },
	"battery": {
		"rx": 18,
		"tx": 7
	},
	"powermeter": {
		"tx": 39,
		"rx": 40,
		"dere": -1
	}
  }
]`

drahdiwaberl, 2024-11-28:
following the discussions and inputs on discord, having 3 single phase inverters (2 PV, 1 battery) at my house, several rather high single phase loads and a limit of unbalanced loads of 3.7kW ..

if it were possible (and from what i read so far i assume it is) to make the DPL work 'per phase', i.e. having meter readings per phase and set 3 inverter(phase)s according to these, then this probably would make a lot of sense to have as a strategy. esp. for the setups people are using OpenDTU for.

arguments for preferring to use as little inverters as possible for efficiency reasons for sure are valid either, but i'm not sure this needs to be first priority with the small loads/powers the majority of users are dealing with in these setups.

my2c.

drahdiwaberl, 2024-11-28:
written it, went on thinking..

actually any strategy should prefer serving the most strained phase as directly as possible (so if phase 1 had a high load on and there's potential production there it should be used) to avoid unbalanced loads..

stefan123t, 2024-11-29:
So you are advocating for a per inverter phase setting for A/B/C.

schlimmchen, 2024-11-29:
@drahdiwaberl Your input is valuable, but I want to point out that you are mixing things here. This issue originally was about "how to I handle multiple inverters if multiple are eligible to fulfill the desired total output". That question in your scenario is only relevant if there are multiple inverters per group/phase/powermeter reading.

Right now I don't see that we will support assinging inverters to a group/phase but still allow to increase production in another group/phase to meet the total need. That would certainly have to wait until managing inverters per group/phase is implemented and works well.

I created #1428 to keep track of support for grouping of inverters per powermeter value.

dragricola, 2024-12-10:
@schlimmchen: 
Ich denke, es ist ein guter Wirkungsgrad mindestens genauso wichtig, wie eine schnellen Reaktionszeit.  Zu diesem Thema habe ich daher im photovoltaikforum einen Post abgesetzt und vom Forumsinventar AusmWW folgende Antwort bekommen, die mir sehr plausibel erscheint: 

Zitat: 

"Die Wirkungsgradkurve sieht für alle WR ähnlich aus, da diese von den verwendeten Halbleiter abhängt.
Im unteren Teillastbereich ist der Wirkungsgrad grottig, weil dort die abgerufene Leistung in der Größenordnung des Eigenverbrauchs liegt.
Bei sehr guten verwendeten Komponenten liegt der „Knick“ im Bereich von 5-10% der Nennleistung.
Wenn du auf der sicheren Seite liegen möchtest, schalte einen WR ab, wenn die abgegebene Leistung unter 15% der Nennleistung fällt.
Im Bereich von ca. 30% bis 50% ist der Wirkungsgrad am besten, weiter höher fällt dieser leicht ab.
Der Abfall des Wirkungsgrades nach oben ist relativ schwach, da würde ich eher den Focus auf eine möglichst geringe Anzahl an Schaltvorgängen achten.
Nach meinem Wissensstand würde sich für die beschriebene Konfiguration wenn, dann nur eine Regelung zur bestmöglichen Vermeidung des unteren Teillastbereichs lohnen. Das ergäbe dann die Abschaltbedingung.
Den 2. WR würde ich zuschalten, wenn der 1. WR bei ca. 80% seiner Nennlast ist.
Das ergibt dann eine relativ breite Regelhysterese die der Reduktion von Schaltvorgängen zugute kommt."

Zitat Ende

Man kann meine Anfrage und den thread gerne auch hier nachlesen:
https://www.photovoltaikforum.com/thread/239346-wirkungsgrad-als-funktion-der-leistung-bei-hoymiles-wechselrichtern-der-hm-serie/

Man könnte die zu aktivierenden Inverter nach folgenden Kriterien bestimmen:

Wenn mehr Leistung gefordert wird (Regelabweichung positiv): 
- Wenn kein Inverter läuft, schalte den Inverter mit max. 80% ein, der die geforderte Leistung mit einer Aussteuerung <80% liefern kann (In diesem Fall wäre auch eine Aussteuerung <15% zulässig). Wenn diese Leistung nicht ausreicht, schalte weitere Inverter nach gleichem Kriterium an.
- Wenn mehrere aber nicht alle Inverter laufen und alle eingeschalteten Inverter laufen mit 80%, schalte weiteren Inverter zu, der die geforderte Leistung mit einer Aussteuerung <80%  und >15% liefern kann. Wenn diese Leistung nicht ausreicht, schalte weitere Inverter sofern vorhanden nach gleichem Kriterium an. 
- Wenn alle Inverter mit 80% laufen, lasse höhere Aussteuergrade zu. 

Wenn weniger Leistung gefordert wird (Regelabweichung negativ):
- Wenn alle Inverter laufen und ein Inverter hat weniger als 15% Aussteuerung, schalte ihn ab.
- Wenn nicht alle Inverter laufen und mindestens ein Inverter hat weniger als 15% Aussteuerung und ein anderer Inverter könnte mit Aussteuerung >15% die Leistung liefern, schalte auf ihn um.
- Wenn nur ein Inverter läuft, der weniger als 15% Aussteuerung hat und ein anderer Inverter könnte mit Aussteuerung >15% die Leistung liefern, schalte auf ihn um.
- Wenn nur ein Inverter läuft der weniger als 15% Aussteuerung hat und kein anderer Inverter könnte mit Aussteuerung >15% die Leistung liefern, lasse Aussteuergrade <15% zu.

Ich hoffe, dass ich keinen Fall vergessen habe!

Der vorhandene Algorithmus zur Ermittlung der Inverteraussteuerungen könnte dann bleiben und müsste sich auf die aktivierten Inverter unter Berücksichtigung der Limits der Aussteuergrade (>80% nur wenn alle Inverter laufen, <15% nur wenn ein einzelner Inverter läuft oder laufen soll) beschränken. 

Eine solche Regelung dürfte auch dafür sorgen, dass die Inverter im Mittel etwa gleich ausgelastet werden und ausreichend vermeiden, dass sich ein Inverter überhitzt, da ab einer bestimmten Aussteuerung weitere Inverter entlastend zugeschaltet werden. Da sich die Inverter beim Betrieb an einer Batterie in einer kühleren Umgebung befinden als auf dem Dach unter einem PV-Modul ist die thermische Situation ohnehin nicht so kritisch.

dragricola, 2025-02-04:
@schlimmchen Zu "Different DPL strategies for handling multiple inverters #1409" hatte ich vor ca. 2 Monaten einen Vorschlag gemacht, den ich hiermit nochmals in Erinnerung rufen möchte.

schlimmchen, 2025-02-04:
Wo Code? :wink: Auf diesem Thema liegt bei mir gerade keine Priorität... Muss jemand anders beisteuern. Für mich funktioniert der DPL sehr gut. Mich machen ein paar Prozent unnötige Verluste nicht nervös...