hoylabs/OpenDTU-OnBattery #1410: [Request] Support for Modbus TCP energy meters.

PortWest86, 2024-11-20:
### Is your feature request related to a problem? Please describe.

There is a Modbus TCP-ready energy meter in the meter cabinet that I would like to integrate in order to control the output of the inverter.

### Describe the solution you'd like

An option for open configuration of the Modbus TCP slave with IP address, Modbus ID and register to be polled.

spcqike, 2024-11-20:
while this is a great request, i guess it takes some time (and a motivated developer) to get integratet

in the meantime, you can use a tasmota device to read out your meter.
tasmota SML supports modbus TCP and can publish your meter data via mqtt.

https://tasmota.github.io/docs/Smart-Meter-Interface/#meter-definition

stefan123t, 2024-11-22:
See this PR https://github.com/tbnobody/OpenDTU/pull/1893 upstream.

stefan123t, 2024-11-22:
And here https://github.com/hoylabs/OpenDTU-OnBattery/pull/1150

AndreasBoehm, 2025-03-18:
The linked PRs from @stefan123t are not related to this request, because they are focused on a modbus server rather than a client.

AndreasBoehm, 2025-03-19:
Can you tell us more about your energy meter @PortWest86 ? Model, documentation?