hoylabs/OpenDTU-OnBattery #1417: Hat Solar Paththrough aktuell ne Macke?

Snoopy-HSS, 2024-11-23:
### What happened?

Aktuell geht der Solar Paththrough mit meinem QVictron nicht.
Schalte ich es ein, regelt er ganz komische Werte.
Bedarf wäre 500 Watt, Solar hat 200W am Victron und er springt wischen 20 und 400 Watt hin und her, macht aber nie die Null am Stromzähler.

Schalte ich es aus, passt wieder alles.

Ist kein Bedarf im Haus, also Minuseinspeisung, speist er genau die 200W - 3% ein.

### To Reproduce Bug

Bei eingeschaltetem Paththrough keine Anpassun der Wechselrichter Leistung am Verbrauch, sondern ständig schwankende Werte

### Expected Behavior

Auch bei BT sollte wie bis vor ca. 10 Tagen wieder mindestens die Leistung am Victron eingespeist werden, aber auch so viel aus der Batterie, dass der DPL passt.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU-OnBattery?

2605412

### What firmware variant (PIO Environment)?

generic_esp32

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Snoopy-HSS, 2024-11-23:
[image]

[image]

Snoopy-HSS, 2024-11-23:
Solar Paththrough ausgeschaltet, ist alles wieder ok:

[image]

schlimmchen, 2024-11-24:
Danke für die Screenshots, aber hier sind Logs wichtig. Bitte DPL Verbose Logging einschalten und wenn das Problem besteht (Solar-Passthrough an) einen Mitschnitt der Webkonsole anfertigen und teilen.

Snoopy-HSS, 2024-11-25:
anbei:

07:21:41.216 > Nothing received, resend whole request
07:21:41.272 > Websocket: [/livedata][22] disconnect
07:21:41.326 > Websocket: [/vedirectlivedata][21] disconnect
07:21:41.374 > Websocket: [/batterylivedata][21] disconnect
07:21:43.240 > Success
07:21:43.293 > Fetch inverter: 114184617285
07:21:43.339 > [DPL inverter 114184617285]: actual limit is 27.6 % (221 W respectively), effective 7023 ms after update started, requested were 27.6 %
07:21:43.387 > [DPL::updateInverterLimits] requested: 185 W, producing: 89 W using 1 battery-powered inverters, diff: 96 W, hysteresis: 20 W
07:21:43.435 > [DPL::updateInverterLimits] will cover 316 W using battery-powered inverters
07:21:43.481 > [DPL inverter 114184617285]:
07:21:43.550 >     battery-powered, producing 89 W
07:21:43.777 >     lower/current/upper limit: 20/220/900 W, output capability: 800 W
07:21:43.827 >     sending commands enabled, reachable, eligible
07:21:43.878 >     max reduction production/standby: 69/89 W, max increase: 580 W
07:21:44.128 >     target limit/output/state: 316 W (update)/316 W/production, 0 update timeouts
07:21:44.196 > [DPL::announceStatus] passing through all solar power (full solar passthrough)
07:21:44.251 > [DPL inverter 114184617285]: sending limit of 39.5 % (316 W respectively), max output is 800 W
07:21:44.298 > [DPL::announceStatus] waiting for a start/stop/restart/limit command to complete
07:21:44.351 > Last missing
07:21:44.503 > Request retransmit: 3
07:21:44.551 > Last missing
07:21:44.598 > Request retransmit: 3
07:21:44.740 > Last missing
07:21:44.798 > Request retransmit: 3
07:21:44.851 > Last missing
07:21:44.900 > Request retransmit: 3
07:21:45.049 > Middle missing
07:21:45.124 > Request retransmit: 1
07:21:45.176 > Success
07:21:45.223 > Last missing
07:21:45.270 > Request retransmit: 11
07:21:45.317 > Last missing
07:21:45.363 > Request retransmit: 11
07:21:45.411 > Last missing
07:21:45.465 > Request retransmit: 11
07:21:45.680 > Last missing
07:21:45.728 > Request retransmit: 11
07:21:45.778 > Last missing
07:21:46.009 > Request retransmit: 11
07:21:46.067 > Last missing
07:21:46.615 > Retransmit timeout
07:21:46.702 > [ShellyACPlug] GET HTTP Error: connection refused
07:21:47.778 > Success
07:21:47.835 > Fetch inverter: 114184617285
07:21:47.883 > [DPL inverter 114184617285]: actual limit is 39.5 % (316 W respectively), effective 4425 ms after update started, requested were 39.5 %
07:21:47.930 > [DPL::updateInverterLimits] requested: 185 W, producing: 228 W using 1 battery-powered inverters, diff: -43 W, hysteresis: 20 W
07:21:47.976 > [DPL::updateInverterLimits] will cover 273 W using battery-powered inverters
07:21:48.022 > [DPL inverter 114184617285]:
07:21:48.072 >     battery-powered, producing 228 W
07:21:48.181 >     lower/current/upper limit: 20/316/900 W, output capability: 800 W
07:21:48.235 >     sending commands enabled, reachable, eligible
07:21:48.293 >     max reduction production/standby: 208/228 W, max increase: 484 W
07:21:48.345 >     target limit/output/state: 273 W (update)/273 W/production, 0 update timeouts
07:21:48.494 > [DPL::announceStatus] passing through all solar power (full solar passthrough)
07:21:48.545 > [DPL inverter 114184617285]: sending limit of 34.1 % (273 W respectively), max output is 800 W
07:21:48.592 > [DPL::announceStatus] waiting for a start/stop/restart/limit command to complete
07:21:48.640 > Last missing
07:21:48.687 > Request retransmit: 2
07:21:48.737 > Last missing
07:21:48.796 > Request retransmit: 2
07:21:48.844 > Last missing
07:21:48.891 > Request retransmit: 2
07:21:48.940 > Last missing
07:21:49.043 > Request retransmit: 3
07:21:49.101 > Success
07:21:49.567 > Middle missing
07:21:49.620 > Request retransmit: 3
07:21:49.699 > Middle missing
[... 197 lines cut ...]
07:22:20.521 >     battery-powered, producing 310 W
07:22:20.574 >     lower/current/upper limit: 20/406/900 W, output capability: 800 W
07:22:20.684 >     sending commands enabled, reachable, eligible
07:22:20.787 >     max reduction production/standby: 290/310 W, max increase: 394 W
07:22:20.836 >     target limit/output/state: 281 W (update)/281 W/production, 0 update timeouts
07:22:20.885 > [DPL::announceStatus] passing through all solar power (full solar passthrough)
07:22:20.995 > [DPL inverter 114184617285]: sending limit of 35.1 % (281 W respectively), max output is 800 W
07:22:21.272 > [DPL::announceStatus] waiting for a start/stop/restart/limit command to complete
07:22:21.327 > Last missing
07:22:21.371 > Request retransmit: 2
07:22:21.420 > Last missing
07:22:21.468 > Request retransmit: 3
07:22:21.516 > Success
07:22:21.704 > Middle missing
07:22:21.758 > Request retransmit: 1
07:22:21.807 > Middle missing
07:22:21.851 > Request retransmit: 3
07:22:21.913 > Middle missing
07:22:21.959 > Request retransmit: 3
07:22:22.028 > Middle missing
07:22:22.261 > Request retransmit: 4
07:22:22.318 > Middle missing
07:22:22.372 > Request retransmit: 6
07:22:22.449 > Middle missing
07:22:22.498 > Retransmit timeout

schlimmchen, 2024-11-25:
> macht aber nie die Null am Stromzähler.

Also zum einen sprichst du laut Logs von **Full** Solar-Passthrough, d.h. der PowerMeter spielt hier keine Rolle. Das ist ein wichtiger Unterschied zur Solar-Passthrough.

Zum anderen sehe ich, dass dein batteriebetriebener Wechselrichter irgendeine Ausgangsleistung einstellt, aber meistens nicht annähernd die gewünschte:

Limit: 221W => produziert: 89W
Limit: 316W => produziert: 228W
Limit: 273W => produziert: 274W <= hier siehts gut aus und das nächste Limit wird dann auch korrekt errechnet
Limit: 182W => produziert: 282W

usw.

Ich glaube zu wissen, warum das passiert: Die Full Solar-Passthrough Implementierung wird aufgerufen **bevor** sichergestellt ist, dass alle Inverter neue Statistiken geliefert haben **nachdem** sie das letzte Limit Update verarbeitet haben.

Magst du einmal folgenden Diff anwenden bei dir und testen (die Zeilen sind lediglich hinter die Schleife verschoben)?

```diff
--- a/src/PowerLimiter.cpp
+++ b/src/PowerLimiter.cpp
@@ -170,39 +170,39 @@ void PowerLimiterClass::loop()
 
     if (_reloadConfigFlag) {
         reloadConfig();
         return announceStatus(Status::ConfigReload);
     }
 
     if (_inverters.empty()) {
         return announceStatus(Status::InverterInvalid);
     }
 
-    if (Mode::UnconditionalFullSolarPassthrough == _mode) {
-        return fullSolarPassthrough(Status::UnconditionalSolarPassthrough);
-    }
-
-    if (isFullSolarPassthroughActive()) {
-        return fullSolarPassthrough(Status::FullSolarPassthrough);
-    }
-
     uint32_t latestInverterStats = 0;
 
     for (auto const& upInv : _inverters) {
         auto oStatsMillis = upInv->getLatestStatsMillis();
         if (!oStatsMillis) {
             return announceStatus(Status::InverterStatsPending);
         }
 
         latestInverterStats = std::max(*oStatsMillis, latestInverterStats);
     }
 
+    if (Mode::UnconditionalFullSolarPassthrough == _mode) {
+        return fullSolarPassthrough(Status::UnconditionalSolarPassthrough);
+    }
+
+    if (isFullSolarPassthroughActive()) {
+        return fullSolarPassthrough(Status::FullSolarPassthrough);
+    }
+
     // if the power meter is being used, i.e., if its data is valid, we want to
     // wait for a new reading after adjusting the inverter limit. otherwise, we
     // proceed as we will use a fallback limit independent of the power meter.
     // the power meter reading is expected to be at most 2 seconds old when it
     // arrives. this can be the case for readings provided by networked meter
     // readers, where a packet needs to travel through the network for some
     // time after the actual measurement was done by the reader.
     if (PowerMeter.isDataValid() && PowerMeter.getLastUpdate() <= (latestInverterStats + 2000)) {
         return announceStatus(Status::PowerMeterPending);
     }
```

Bei Solar-Passthrough (kein **Full**) tritt das Problem so nicht auf, und das hätte ich auch bemerkt. Für Full Solar-Passthrough war kein Wetter die letzten Monate, daher ist mir das entgangen.

Snoopy-HSS, 2024-11-25:
Prüfe das gleich mal, aber ist bei mir auch beim "normalen" Path-Through so...

Der Victron macht 214 Watt, das Haus brauch 415 und die DTU bringt die Random Werte dem Wechselrichter bei...

Snoopy-HSS, 2024-11-25:
Ja, dass sieht wieder gut aus. -30 Watt als Ziel eingestellt.
Hab mal einen 250W Verbraucher an und ausgeschaltet während des logs. Regelt direkt wieder nach,
So war es bis vor 2 Wochen etwa auch.

Vielen Dank für den schnellen Tip.
Soll ich es in meinen Mergerequest für den Shellystecker mit rein packen?

15:24:07.261 > [PowerMeterUdpSmaHomeManager] Leistung = 135.1 (timestamp 1672824829)
15:24:07.315 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 144.4 (timestamp 1672824829)
15:24:07.365 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 256.1 (timestamp 1672824829)
15:24:07.415 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -265.4 (timestamp 1672824829)
15:24:07.496 > [PowerMeterUdpSmaHomeManager] Leistung = 64.2 (timestamp 1672825824)
15:24:07.546 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 145.8 (timestamp 1672825824)
15:24:07.595 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 185.1 (timestamp 1672825824)
15:24:07.641 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -266.8 (timestamp 1672825824)
15:24:07.697 > Success
15:24:08.067 > Fetch inverter: 114184617285
15:24:08.111 > [DPL inverter 114184617285]: actual limit is 49.2 % (394 W respectively), effective 2275 ms after update started, requested were 49.2 %
15:24:08.164 > [DPL::announceStatus] waiting for sufficiently recent inverter data
15:24:08.212 > Last missing
15:24:08.266 > Request retransmit: 2
15:24:08.312 > [PowerMeterUdpSmaHomeManager] Leistung = 91.0 (timestamp 1672826829)
15:24:08.362 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 144.5 (timestamp 1672826829)
15:24:08.411 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 212.2 (timestamp 1672826829)
15:24:08.462 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -265.8 (timestamp 1672826829)
15:24:08.511 > Last missing
15:24:08.562 > Request retransmit: 2
15:24:08.617 > Last missing
15:24:08.661 > Request retransmit: 3
15:24:08.712 > Success
15:24:08.762 > [DPL::announceStatus] waiting for sufficiently recent power meter reading
15:24:08.845 > [PowerMeterUdpSmaHomeManager] Leistung = 77.6 (timestamp 1672827824)
15:24:08.896 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 142.3 (timestamp 1672827824)
15:24:09.067 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 198.3 (timestamp 1672827824)
15:24:09.127 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -263.0 (timestamp 1672827824)
15:24:09.262 > Fetch inverter: 114184617285
15:24:09.811 > Last missing
15:24:10.069 > Request retransmit: 2
15:24:10.118 > [PowerMeterUdpSmaHomeManager] Leistung = 76.1 (timestamp 1672828829)
15:24:10.163 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 143.1 (timestamp 1672828829)
15:24:10.212 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 196.9 (timestamp 1672828829)
15:24:10.262 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -264.0 (timestamp 1672828829)
15:24:10.314 > Last missing
15:24:10.360 > Request retransmit: 2
15:24:10.410 > Last missing
15:24:10.462 > Request retransmit: 2
15:24:10.513 > Last missing
15:24:10.561 > Request retransmit: 2
15:24:10.610 > Last missing
15:24:10.661 > Request retransmit: 3
15:24:10.711 > Success
15:24:10.822 > [PowerMeterUdpSmaHomeManager] Skipped unknown measurement: 116 1 241 37
15:24:11.265 > Fetch inverter: 114184617285
15:24:11.813 > Last missing
15:24:11.862 > Request retransmit: 3
15:24:11.985 > [PowerMeterUdpSmaHomeManager] Leistung = 67.0 (timestamp 1672829824)
15:24:12.087 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 146.5 (timestamp 1672829824)
15:24:12.144 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 187.3 (timestamp 1672829824)
15:24:12.195 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -266.8 (timestamp 1672829824)
15:24:12.246 > [DPL::loop] ******************* ENTER **********************
15:24:12.295 > [DPL::loop] battery interface enabled, SoC: 79.0 %, StartTH: 25 %, StopTH: 5 %, SoC age: 0 s, ignore: no
15:24:12.344 > [DPL::getBatteryVoltage] BMS: 56.92 V, MPPT: 56.29 V, inverter 114184617285: 56.00 V, returning: 56.92V
15:24:12.396 > [DPL::loop] dcVoltage: 56.92 V, loadCorrectedVoltage: 57.32 V, StartTH: 52.00 V, StopTH: 47.00 V
15:24:12.444 > [DPL::loop] StartTH reached: yes, StopTH reached: no, SolarPT enabled, use at night: yes
15:24:12.495 > [DPL::calcHouseholdConsumption] target consumption: -30 W, base load: 100 W
15:24:12.545 > [DPL::calcHouseholdConsumption] power meter value: 67.0 W, power meter valid: yes
15:24:12.595 > [DPL::calcHouseholdConsumption] inverter 114184617285 is behind power meter producing 403 W
15:24:12.647 > [DPL::updateInverterLimits] requested: 500 W, producing: 0 W using 0 solar-powered inverters, diff: 500 W, hysteresis: 20 W
15:24:12.695 > [DPL::calcBatteryAllowance] power requested: 500 W
15:24:12.746 > [DPL::updateInverterLimits] requested: 500 W, producing: 403 W using 1 battery-powered inverters, diff: 97 W, hysteresis: 20 W
[... 118 lines cut ...]
15:24:23.435 > Request retransmit: 3
15:24:23.480 > Last missing
15:24:23.532 > Retransmit timeout
15:24:23.845 > [PowerMeterUdpSmaHomeManager] Leistung = 24.9 (timestamp 1672842829)
15:24:23.901 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 399.4 (timestamp 1672842829)
15:24:24.067 > [PowerMeterUdpSmaHomeManager] Leistung L2 = -128.4 (timestamp 1672842829)
15:24:24.115 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -246.1 (timestamp 1672842829)
15:24:24.163 > Fetch inverter: 114184617285
15:24:24.552 > Middle missing
15:24:24.604 > Request retransmit: 2
15:24:24.965 > Middle missing
15:24:25.014 > Request retransmit: 2
15:24:25.065 > Middle missing
15:24:25.116 > Request retransmit: 2
[... cut ...]
15:24:25.365 > Success
15:24:25.413 > [DPL::announceStatus] waiting for sufficiently recent power meter reading
15:24:25.847 > [PowerMeterUdpSmaHomeManager] Leistung = 31.0 (timestamp 1672844819)
15:24:25.897 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 400.7 (timestamp 1672844819)
15:24:25.949 > [PowerMeterUdpSmaHomeManager] Leistung L2 = -123.0 (timestamp 1672844819)
15:24:26.072 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -246.7 (timestamp 1672844819)
15:24:26.130 > Fetch inverter: 114184617285

Snoopy-HSS, 2024-11-25:
Anderer Fehler.

Wenn die Batterie jetzt den Full Path Through Wert erreicht, wird der Inverter auf die 214 von Victron -3% fest geregelt, egal ob mehr benötigt wird oder nicht. Er nimmt keine Strom aus der Batterie dazu.

Last missing
16:24:06.555 > Retransmit timeout
16:24:06.607 > Fetch inverter: 114184617285
16:24:06.961 > Last missing
16:24:07.008 > Request retransmit: 2
16:24:07.057 > [PowerMeterUdpSmaHomeManager] Leistung = 653.2 (timestamp 1676426089)
16:24:07.108 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 270.2 (timestamp 1676426089)
16:24:07.158 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 540.0 (timestamp 1676426089)
16:24:07.208 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -156.9 (timestamp 1676426089)
16:24:07.258 > [DPL::updateInverterLimits] requested: 196 W, producing: 208 W using 1 battery-powered inverters, diff: -12 W, hysteresis: 20 W
16:24:07.307 > Last missing
16:24:07.359 > Request retransmit: 2
16:24:07.409 > Last missing
16:24:07.462 > Request retransmit: 3
16:24:07.507 > Last missing
16:24:07.564 > Request retransmit: 3
16:24:07.623 > Last missing
16:24:07.673 > Request retransmit: 3
16:24:07.724 > Last missing
16:24:07.773 > Retransmit timeout
16:24:07.824 > Fetch inverter: 114184617285
16:24:08.080 > All missing
16:24:08.126 > Nothing received, resend whole request
16:24:08.654 > [PowerMeterUdpSmaHomeManager] Leistung = 675.4 (timestamp 1676427084)
16:24:08.714 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 269.2 (timestamp 1676427084)
16:24:09.554 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 563.3 (timestamp 1676427084)
16:24:09.614 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -157.1 (timestamp 1676427084)
16:24:09.659 > [DPL::updateInverterLimits] requested: 196 W, producing: 208 W using 1 battery-powered inverters, diff: -12 W, hysteresis: 20 W
16:24:09.846 > [ShellyACPlug] GET HTTP Error: connection refused
16:24:09.892 > Middle missing
16:24:09.944 > Request retransmit: 2
16:24:09.992 > Middle missing
16:24:10.042 > Request retransmit: 2
16:24:10.091 > Success
16:24:10.144 > Fetch inverter: 114184617285
16:24:10.190 > [PowerMeterUdpSmaHomeManager] Leistung = 665.6 (timestamp 1676428089)
16:24:10.241 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 269.0 (timestamp 1676428089)
16:24:10.292 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 553.9 (timestamp 1676428089)
16:24:10.344 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -157.3 (timestamp 1676428089)
16:24:10.390 > [DPL::updateInverterLimits] requested: 196 W, producing: 208 W using 1 battery-powered inverters, diff: -12 W, hysteresis: 20 W
16:24:10.441 > Last missing
16:24:10.492 > Request retransmit: 2
16:24:10.541 > Last missing
16:24:10.593 > Request retransmit: 2
16:24:10.641 > Last missing
16:24:10.691 > Request retransmit: 2
16:24:10.740 > Last missing
16:24:10.792 > Request retransmit: 2
16:24:10.844 > Last missing
16:24:10.895 > Request retransmit: 3
16:24:10.944 > Last missing
16:24:10.992 > Retransmit timeout
16:24:11.044 > Fetch inverter: 114184617285
16:24:11.091 > [PowerMeterUdpSmaHomeManager] Leistung = 627.4 (timestamp 1676429084)
16:24:11.284 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 269.2 (timestamp 1676429084)
16:24:11.559 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 515.5 (timestamp 1676429084)
16:24:11.609 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -157.3 (timestamp 1676429084)
16:24:11.659 > [DPL::updateInverterLimits] requested: 196 W, producing: 208 W using 1 battery-powered inverters, diff: -12 W, hysteresis: 20 W
16:24:11.709 > Middle missing
16:24:11.761 > Request retransmit: 2
16:24:11.810 > Middle missing
16:24:11.922 > Request retransmit: 2
16:24:11.977 > Middle missing
16:24:12.209 > Request retransmit: 2
16:24:12.261 > Middle missing
16:24:12.310 > Request retransmit: 2
[... 239 lines cut ...]
16:24:29.261 > Request retransmit: 2
16:24:29.310 > Last missing
16:24:29.360 > Request retransmit: 2
16:24:29.410 > Last missing
16:24:29.462 > Request retransmit: 2
16:24:29.513 > Last missing
16:24:29.565 > Request retransmit: 2
16:24:29.613 > Last missing
16:24:29.660 > Request retransmit: 3
16:24:29.740 > Success
16:24:29.793 > Fetch inverter: 114184617285
16:24:29.843 > [PowerMeterUdpSmaHomeManager] Leistung = 557.3 (timestamp 1676448079)
16:24:29.893 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 245.4 (timestamp 1676448079)
16:24:29.942 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 447.6 (timestamp 1676448079)
16:24:29.992 > [PowerMeterUdpSmaHomeManager] Leistung L3 = -135.6 (timestamp 1676448079)
16:24:30.043 > [DPL::updateInverterLimits] requested: 196 W, producing: 204 W using 1 battery-powered inverters, diff: -8 W, hysteresis: 20 W
16:24:30.093 > All missing
16:24:30.142 > Nothing received, resend whole request
16:24:30.195 > Last missing
16:24:30.245 > Request retransmit: 2
16:24:30.294 > Last missing
16:24:30.344 > Request retransmit: 2
16:24:30.393 > Last missing
16:24:30.444 > Request retransmit: 3
16:24:30.496 > [PowerMeterUdpSmaHomeManager] Leistung = 547.6 (timestamp 1676449084)

schlimmchen, 2024-11-25:
> Prüfe das gleich mal, aber ist bei mir auch beim "normalen" Path-Through so...

Hm, das kann aber nicht sein bzw. ist das Verschieben der Code-Zeilen wie von mir vorgeschlagen dann völlig irrelevant.

Wie gesagt kann ich das nicht nachvollziehen. Schau mal hier exemplarisch was von heute:
[image]

Obwohl die solare Leistung stark schwankt, wurden die Inverter (bzw. der eine) wie erwartet so eingestellt, dass die Grid Power um die 20W schwankte, mit vertretbaren Abweichungen.

Kannst du zusätzlich einen Log teilen, während non-Full Solar-Passthrough läuft und der Inverter in seiner Ausgangsleistung stark schwankt?

> Soll ich es in meinen Mergerequest für den Shellystecker mit rein packen?

Nein. Wir vermischen nicht Buffixes und Features im selben PR. Ich schieb demnächst einen passenden commit in den development Branch.

> Wenn die Batterie jetzt den Full Path Through Wert erreicht, wird der Inverter auf die 214 von Victron -3% fest geregelt, egal ob mehr benötigt wird oder nicht. Er nimmt keine Strom aus der Batterie dazu.

Fail :man_facepalming: Das ist richtig. Das hab ich total verkackt. Da muss ich nochmal drüber nachdenken, wie man das ins neue Design einbauen kann ohne es zu zerhackstückeln. Dazu öffne ich ein neues Issue.

Snoopy-HSS, 2024-11-25:
Ok, neuer Stand.
Alte Firmware(2024.11.20). Starke Schwankungen **nur** im Full Path Through. Anscheinend hatte ich hier einen falschen Wert konfiguriert oder die Batterie war genau beim Einspielen des Bugfixes umgesprungen von FPT in PT mode...

Bugfix Firmware: Nach verschieben der Codezeilen wie von dir beschrieben, nur noch VE-Direct MPPT Wert -3% im Full Path Through betrieb.
Im Path Through Betrieb bei beiden FW Varianten alles ok.

schlimmchen, 2024-11-25:
Alles klar, danke für die Rückmeldung. Dann ist dieses Issue hier geschlossen via f8f54aca und das verbleibende Problem ist ein anderes und wird in #1422 weiterbehandelt.