hoylabs/OpenDTU-OnBattery #1421: Dynamic Power Limiter - maximales Leistungslimit nach Neustart wieder auf 800 Watt

knopers1, 2024-11-25:
### What happened?

maximales Leistungslimit auf 200 Watt gesetzt und gespeichert. Nach Neustart steht der Wert auf maximum 800 Watt. 

### To Reproduce Bug

Der Fehler lässt sich immer reproduzieren.

### Expected Behavior

der eingestellte Wert sollte nach Neustart 200 Watt betragen.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.10.22

### What firmware variant (PIO Environment)?

generic_esp32s3

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

knopers1, 2024-11-28:
@schlimmchen 
Danke Dir! Ich habe jetzt die Release 2024.11.20 aufgespielt, das Problem scheint nicht mehr zu existieren. 
Das Problem in der alten Release 2024.10.22 ist mir nie aufgefallen, da der WR immer auf volle 800 Watt eingestellt war. Jetzt im Winter musste ich meine Pylon Batterie abklemmen und ins "Warme" umziehen. Dabei habe ich meine alten AGM wieder angeschlossen und dabei die max. Leistung auf 200 Watt gesetzt. 
Wie auch immer... Das Problem ist mit der neuen Release beseitigt.

schlimmchen, 2024-11-28:
Danke für die Rückmeldung.

closed 2024-11-28 by schlimmchen