hoylabs/OpenDTU-OnBattery #1427: [Bug] DPL: Inverter produziert nicht

schlimmchen, 2024-11-26:
Hallo,
irgendwie geht da was nicht mit der 20.11.2024.

ESP32-S3 8MB, Oled,Pylontec,Victron,SML,(W5500)

Ich habe alles so eingestellt wie in der Version vorher, aber es wird kein Strom produziert.
Was ist hier falsch?

<details>
<summary>Dump von der Konsole:</summary>
```[VE.Direct MPPT 12/11] serial input (242 Bytes):
19:36:24.441 > [VE.Direct MPPT 12/11] 3a 37 46 36 45 44 30 30 31 38 31 35 33 45 0a 3a
19:36:24.504 > [VE.Direct MPPT 12/11] 37 32 37 32 30 30 30 46 46 46 46 46 46 46 46 30
19:36:24.565 > [VE.Direct MPPT 12/11] 42 0a 3a 41 44 35 45 44 30 30 44 34 31 34 41 31
19:36:24.631 > [VE.Direct MPPT 12/11] 0a 3a 41 42 42 45 44 30 30 31 32 30 30 39 31 0a
19:36:24.682 > [VE.Direct MPPT 12/11] 0d 0a 50 49 44 09 30 78 41 30 37 33 0d 0a 46 57
19:36:24.738 > [VE.Direct MPPT 12/11] 09 31 36 34 0d 0a 53 45 52 23 09 48 51 32 32 34
19:36:24.804 > [VE.Direct MPPT 12/11] 38 48 39 41 34 33 0d 0a 56 09 35 33 33 32 30 0d
19:36:24.868 > [VE.Direct MPPT 12/11] 0a 49 09 30 0d 0a 56 50 56 09 31 39 30 0d 0a 50
19:36:24.919 > [VE.Direct MPPT 12/11] 50 56 09 30 0d 0a 43 53 09 30 0d 0a 4d 50 50 54
19:36:24.978 > [VE.Direct MPPT 12/11] 09 30 0d 0a 4f 52 09 30 78 30 30 30 30 30 30 30
19:36:25.033 > [VE.Direct MPPT 12/11] 31 0d 0a 45 52 52 09 30 0d 0a 4c 4f 41 44 09 4f
19:36:25.197 > [VE.Direct MPPT 12/11] 4e 0d 0a 48 31 39 09 35 34 38 35 0d 0a 48 32 30
19:36:25.255 > [VE.Direct MPPT 12/11] 09 39 30 0d 0a 48 32 31 09 33 30 38 0d 0a 48 32
19:36:25.327 > [VE.Direct MPPT 12/11] 32 09 37 39 0d 0a 48 32 33 09 32 36 35 0d 0a 48
19:36:25.387 > [VE.Direct MPPT 12/11] 53 44 53 09 34 37 0d 0a 43 68 65 63 6b 73 75 6d
19:36:25.437 > [VE.Direct MPPT 12/11] 09 f3
19:36:25.491 > [VE.Direct MPPT 12/11] Text Data 'PID' = '0XA073'
19:36:25.610 > [VE.Direct MPPT 12/11] Text Data 'FW' = '164'
19:36:25.656 > [VE.Direct MPPT 12/11] Text Data 'SER' = 'HQ2248------'
19:36:25.704 > [VE.Direct MPPT 12/11] Text Data 'V' = '53320'
19:36:25.749 > [VE.Direct MPPT 12/11] Text Data 'I' = '0'
19:36:25.797 > [VE.Direct MPPT 12/11] Text Data 'VPV' = '190'
19:36:25.853 > [VE.Direct MPPT 12/11] Text Data 'PPV' = '0'
19:36:25.909 > [VE.Direct MPPT 12/11] Text Data 'CS' = '0'
19:36:26.028 > [VE.Direct MPPT 12/11] Text Data 'MPPT' = '0'
19:36:26.089 > [VE.Direct MPPT 12/11] Text Data 'OR' = '0X00000001'
19:36:26.141 > [VE.Direct MPPT 12/11] Text Data 'ERR' = '0'
19:36:26.200 > [VE.Direct MPPT 12/11] Text Data 'LOAD' = 'ON'
19:36:26.283 > [VE.Direct MPPT 12/11] Text Data 'H19' = '5485'
19:36:26.346 > [VE.Direct MPPT 12/11] Text Data 'H20' = '90'
19:36:26.404 > [VE.Direct MPPT 12/11] Text Data 'H21' = '308'
19:36:26.463 > [VE.Direct MPPT 12/11] Text Data 'H22' = '79'
19:36:26.528 > [VE.Direct MPPT 12/11] Text Data 'H23' = '265'
19:36:26.604 > [VE.Direct MPPT 12/11] Text Data 'HSDS' = '47'
19:36:26.683 > [DPL::loop] ******************* ENTER **********************
19:36:26.762 > [DPL::loop] battery interface enabled, SoC: 99.0 %, StartTH: 90 %, StopTH: 20 %, SoC age: 1 s, ignore: no
19:36:26.852 > [DPL::getBatteryVoltage] BMS: 53.25 V, MPPT: 53.32 V, inverter 1164a------------: 53.40 V, returning: 53.25V
19:36:26.913 > [DPL::loop] dcVoltage: 53.25 V, loadCorrectedVoltage: 53.25 V, StartTH: 53.00 V, StopTH: 49.00 V
19:36:26.982 > [DPL::loop] StartTH reached: yes, StopTH reached: no, SolarPT disabled, use at night: yes
19:36:27.041 > [DPL::calcHouseholdConsumption] target consumption: 0 W, base load: 10 W
19:36:27.112 > [DPL::calcHouseholdConsumption] power meter value: 405.0 W, power meter valid: yes
19:36:27.174 > [DPL::calcHouseholdConsumption] inverter 1164a------------- is behind power meter producing 1 W
19:36:27.237 > [DPL::updateInverterLimits] requested: 406 W, producing: 0 W using 0 solar-powered inverters, diff: 406 W, hysteresis: 30 W
19:36:27.297 > [DPL::calcBatteryAllowance] power requested: 406 W
19:36:27.461 > [DPL::updateInverterLimits] requested: 406 W, producing: 1 W using 1 battery-powered inverters, diff: 405 W, hysteresis: 30 W
19:36:27.520 > [DPL::updateInverterLimits] will cover 1 W using battery-powered inverters
19:36:27.587 > [DPL inverter 1164a---------]:
19:36:27.641 >     battery-powered, producing 1 W
19:36:27.707 >     lower/current/upper limit: 10/0/500 W, output capability: 1600 W
19:36:27.771 >     sending commands enabled, reachable, disqualified
19:36:27.843 >     max reduction production/standby: 0/0 W, max increase: 0 W
19:36:27.899 >     target limit/output/state: -1 W (unchanged)/1 W/unchanged, 0 update timeouts
19:36:27.957 > [DPL::loop] consumption: 406 W, target output: 406 W (limited to 1500 W), solar inverters output: 0 W, battery allowance: 406 W, battery inverters output: 1 W
19:36:28.009 > [DPL::announceStatus] the system is stable, the last power limit is still valid
19:36:28.066 > [VE.Direct MPPT 12/11] Unhandled Hex Async Response, addr: 0xEDD5 (Charger Voltage), value: 0x000014D5, flags: 0x00
19:36:28.123 > [VE.Direct MPPT 12/11] Unhandled Hex Async Response, addr: 0xEDBB (Panel Voltage), value: 0x00000012, flags: 0x00
19:36:28.275 > Fetch inverter: 1164A-----------
19:36:28.323 > TX RealTimeRunData 865.00 MHz --> 15 A0 09 93 20 80 19 36 00 80 0B 00 67 46 15 29 00 00 00 00 00 00 00 00 DC 7D 97 
19:36:28.381 > [PowerMeterSerialSml] decoded energy import to -0.10
[... 308 lines cut ...]
		"data": 9,
		"clk": 8,
		"cs": -1,
		"reset": -1
    },
    "victron": {
		"rx": 12,
		"tx": 11,
		"rx1": 10,
		"tx1": 17
    },
	"battery": {
		"rx": 18,
		"tx": 7
	},
	"powermeter": {
		"tx": 39,
		"rx": 40,
		"dere": -1
	}
  }
]```
</details>

_Originally posted by @andy75andreas in https://github.com/hoylabs/OpenDTU-OnBattery/issues/1409#issuecomment-2501722870_

schlimmchen, 2024-11-26:
```
19:36:27.587 > [DPL inverter 1164a---------]:
19:36:27.641 >     battery-powered, producing 1 W
19:36:27.707 >     lower/current/upper limit: 10/0/500 W, output capability: 1600 W
19:36:27.771 >     sending commands enabled, reachable, disqualified
19:36:27.843 >     max reduction production/standby: 0/0 W, max increase: 0 W
19:36:27.899 >     target limit/output/state: -1 W (unchanged)/1 W/unchanged, 0 update timeouts
```

Der Inverter kann nicht verwendet werden, weil der DPL darauf wartet, dass bekannt wird, welches Limit dort eingestellt ist.

Hält dieser Zustand dauerhaft an oder nur fünf Minuten nach dem Start deiner OpenDTU-OnBattery?

Wie lang ist die Uptime/Betriebszeit laut Web UI unter Info -> System?

schlimmchen, 2024-11-26:
@andy75andreas Ich habe dein Problem in dieses Issue verschoben. Bitte starte keine Diskussion um ein neues Problem in einem existierenden Issue. Neues Problem => neues Issue. Wenn ein Issue keins ist, machen wir es wieder zu, kein Problem. Die Ganzzahlen um Issues durchzunummerieren gehen uns sobald nicht aus. Wovon aber nicht genug da ist: Kapazität mehrere Themen in ein und dem gleichen Issue auseinander zu halten :wink:

andy75andreas, 2024-11-26:
Hallo "schlimmchen", alles klar und kein Problem.

Hier die Daten der Batterie:
[image]

Betriebszeit: 0 Tage 00:48:16

Dieser Zustand ist seit dem Update auf die aktuelle Version.
Es spiel im Übrigen auch keine Rolle ob der W5500 in den Einstellungen aktiviert ist oder nicht.

schlimmchen, 2024-11-26:
Hm. Im Live-View ist der Inverter gelb (nicht rot), oder sogar blau? Was sagt "Letzte Aktualisierung"?

Als Workaround kannst du bitte einmal ein beliebiges Limit manuell an den Inverter schicken, z.B. 100W, über die Web UI. Dann wird bekannt sein, dass 100W eingestellt sind, und der DPL kann diesen Inverter verwenden.

Was ich nicht verstehe ist, warum OpenDTU nicht irgendwann das Limit abfragt. Vielleicht wird nur ein Versuch pro Neustart unternommen, und wenn der fehlschlägt, bleibt das Limit unbekannt? Das müsste ich mal im Code nachlesen. Es erscheint mir aber nötig, das der DPL selbstständig nach einer gewissen Zeitspanne das lower power limit setzt, sodass das gesetzte Limit bekannt ist.

andy75andreas, 2024-11-26:
Hallo,

manuell auf 10% eingestellt = 160W.
Und ab da Regelt der DPL automatisch.

Das war definitiv bei der vorherigen Version nicht so!

[image]

schlimmchen, 2024-11-26:
> Das war definitiv bei der vorherigen Version nicht so!

Ja, ich weiß? Soll ich mich jetzt schämen?

Wenn du OpenDTU-OnBattery neu startest, kommt das Problem dann wieder und bleibt für mindestens fünf Minuten? Es dauert eine Zeit lang nach einem Neustart, aber nach fünf Minuten sollte es dann losgehen.

andy75andreas, 2024-11-26:
Sollte keine Anschuldigung sein.

Nach Neustart:

Umrichter steht bei 0%, produziert aber 30W.
Umrichter bleibt bei 0%.
Manuell auf 10% und der DPL regelt.

[image]
[image]

schlimmchen, 2024-11-26:
> Umrichter bleibt bei 0%.
Manuell auf 10% und der DPL regelt.

Hast du zwischen diesen beiden Schritten fünf Minuten verstreichen lassen?

Hm, ich glaube es ist egal, denn das ist schon mehrfach aufgefallen, und es ist jedenfalls Mist, wenn die OpenDTU-OnBattery neu startet und dann (aus ungeklärten Gründen) der DPL niemals anfängt zu regeln weil er darauf wartet zu erfahren, welches Limit eingestellt ist. Er muss im Zweifel selbst ein befehlen. Das werde ich einbauen.

andy75andreas, 2024-11-26:
Noch ne Frage zur Regelung mit 2 WR.
Bei einem WR wird alle 5s ein Wert gesendet und bei 2 WR? 
Ich habe zB meinen 2. WR der direkt an der Somme hängt in der Kommunikation vom DTU ausgeschaltet damit der nicht immer abgefragt wird. Einmal im Monat mache ich den kurz an um die Werte zu lagen.
Wenn das dann doppelt so lange dauert wäre dann eine Nulleispeisung überhaupt noch möglich, da das System dann viel zu träge wird?

schlimmchen, 2024-11-26:
> Bei einem WR wird alle 5s ein Wert gesendet

Das kommt ganz auf deine Einstellungen an. Setzt das PowerMeter Intervall auf 1s (wenn du einen hast, der Polling macht) und setz das DTU Intervall auf 1s, dann kann der DPL auch häufiger als alle 5s ein neues Limit setzen, auch wenn ein zweiter WR abgefragt wird.

andy75andreas, 2024-11-26:
Hallo,

SML sendet immer 1s (Optischer Leser).
DTU mit 2s geht.

schlimmchen, 2024-11-26:
Also mir ist das jetzt mindestens drei Mal passiert an meinem Test-Setup und jedes Mal hat sich das Problem nach spätestens fünf Minuten von selbst erledigt, weil dann doch das aktuelle Limit vom WR erfragt wurde. Merkwürdig. Aber auch fünf Minuten warten müssen ist nicht ideal, daher bekräftige ich nochmal, dass der DPL "einfach" das lower power limit setzen sollte, wenn er bemerkt, dass der inverter kein bekanntes limit meldet.

mkneppe, 2024-11-29:
Ich habe mal in meinem iobroker addiert, dass immer wenn der DPL das Limit auf 0 setzt, dass er wieder auf 100 % setzt.
2024-11-29 09:16:42.557 - warn: javascript.0 (2180) script.js.BKW.openDTU_Ausgleichen: 100 % wegen 0 gesetzt
2024-11-29 09:21:45.373 - warn: javascript.0 (2180) script.js.BKW.openDTU_Ausgleichen: 100 % wegen 0 gesetzt
2024-11-29 09:31:42.337 - warn: javascript.0 (2180) script.js.BKW.openDTU_Ausgleichen: 100 % wegen 0 gesetzt
2024-11-29 09:36:43.342 - warn: javascript.0 (2180) script.js.BKW.openDTU_Ausgleichen: 100 % wegen 0 gesetzt
2024-11-29 09:41:42.332 - warn: javascript.0 (2180) script.js.BKW.openDTU_Ausgleichen: 100 % wegen 0 gesetzt
2024-11-29 09:46:43.370 - warn: javascript.0 (2180) script.js.BKW.openDTU_Ausgleichen: 100 % wegen 0 gesetzt
Es ist immer ein vielfaches der genannten 5 Minuten. 
Kann das ein Konfigurationsfehler von meiner Seite sein ?
Innerhalb der 5 Minuten regelt er korrekt. Nach den 5 Minuten stellt die "opendtu on battery" den WR wieder auf 0

schlimmchen, 2024-11-30:
Erledigt in be9e2aac.

closed 2024-11-30 by schlimmchen