hoylabs/OpenDTU-OnBattery #1428: [Request] DPL: Support grouping of inverters

schlimmchen, 2024-11-29:
### Describe the solution you'd like

Allow to assign inverters to a specific power meter value, i.e., a phase, such that setups with multiple inverters but split over multiple phases can be governed such that they only produce as much as is consumed on that particular phase.

Andrix82, 2025-05-14:
Hallo! 
Wie ist denn hier der Status, wird das Thema noch verfolgt ?
Hätte ich einen UseCase dafür und könnte auch SW testen.
Habe ein Setup mit 3*HM1500 auf 3 Phasen aufgeteilt und Powermeter per Phase vom Shelly 3EM.
Sind zwar aktuell nur solar-powered aber denke für den DPL sollte das ja keinen besonderen Unterschied machen.

AndreasBoehm, 2025-05-15:
Wir haben das Thema auf unserer ToDo-Liste, hat aber derzeit keine Priorität. Wir haben einige andere Themen die wir aktuell bearbeiten.