hoylabs/OpenDTU-OnBattery #1432: Fehlerhafte Temperaturanzeige Huawei R4850G2

Solarteur, 2024-12-02:
### What happened?

Yesterday it was quite cold in the garage, the initial temperature on the Huawei R4850G2 was initially displayed as **4194303 °C**, then after a few minutes of operation it showed **1 °C**, which is closer to reality.
[image]

### To Reproduce Bug

Operating the Huawei R4850G2 at low temperatures. I only have this device and therefore cannot rule out that the error comes from Huawei.

According to the specification, the "operating temperature range" is -40 °C - +75°C.

### Expected Behavior

Display also of sub-zero temperatures.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.11.20

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Anything else?

For the sake of completeness, no priority.

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2024-12-02:
@Solarteur Unfortunately, I still don't have such an AC charger (maybe I will gift one to myself this christmas :wink:), so I need you to please test the [firmware built in the context of PR#1433](https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/12115922101?pr=1433). Can you do that?

Solarteur, 2024-12-02:
Sure, I'll do this test. Since it's warmer now, I won't be able to say anything about the results until the temperatures drop.

Solarteur, 2024-12-02:
With PR#1433, temperatures below 0 °C are also displayed correctly - thank you!

[image]

closed 2024-12-02 by Solarteur

schlimmchen, 2024-12-02:
Awesome! Thanks for testing!

PR #1433 (2024-12-02): Fix: Huawei AC charger sends signed values
as shown in #1432, the Huawei AC charger sends signed values to communicate the temperature. given that the value range for all other data points is expected to be less than ~2 million, we now interpret all data as signed values.

Closes #1432.