hoylabs/OpenDTU-OnBattery #1438: Time Calibration ends in DTU Restart

LinuxUnix80, 2024-12-03:
### What happened?

Hi,

The Time Calibration will let Restart the DTU:

09:16:19	09:16:19	2	Zeitabgleich
09:16:24	09:16:24	2	Zeitabgleich

I dont know, if its happend, because of two Time changes in short time. It happend to fusion board v2.5 and even on a selfmade esp32-s3 setting.

### To Reproduce Bug

DTU must trigger (minimun) 2 time Calibration in a short time

### Expected Behavior

Maybe delay between time Calibration changes. I expect no Restart for time Calibration. 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024-11-20

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

stefan123t, 2024-12-03:
@LinuxUnix80 Ich verstehe nicht was das Issue soll ?

Event Code 2 Time Calibration ist das Ereignis das der Wechselrichter an die DTU meldet. Die warn_code.json wurde aus der S-Miles Installer App des Herstellers extrahiert.

Was dieses Ereignis auslöst wissen wir bis dato nicht. Es tritt immer mal wieder auf und hat evtl etwas mit dem Zeitstempel in den RealTimeRunDebug Abfragen zu tun, dh bei schlechter Verbindungsqualität oder Problemen mit der Stromversorgung des Inverters aus den PV / DC Anschlüssen zB früh morgens / abends aber auch bei max Einspeisung gegen mittag tritt das Ereignis zeitweise häufiger auf.

Evtl müssen wir auch alle x Sekunden / Minuten ein bestimmtes Kommando an den Wechselrichter schicken damit er das Gefühl hat immer noch mit der DTU verbunden und die Zeit von DTU und WR im Sync sind ?

Früher haben wir es übrigens einfach nur Kommunikationsfehler genannt.

Aber wenn Du keine Erklärung / einen Verbesserungsvorschlag hast, weiß ich nicht wo hier ein Problem sein soll bzw wie wir Dir helfen können?

LinuxUnix80, 2024-12-03:
Hi,

erstmal vielen Dank für deine Antwort!

Das nervige ist halt, wenn dies abends passiert, der WR kaum noch Strom produziert und die DTU Restartet und keine neuen Werte mehr bekommt, dann alle Zähler auf 0 sind in Home-Assiant.

Es gab mal eine Version, von der regulären DTU (muss 2023 gewesen, Release habe ich mir leider nicht gemerkt), da lief die DTU ohne Probleme wochenlang durch.

Der Fehler kann ja ruhig auftreten, sollte aber nicht die DTU rebooten.

stefan123t, 2024-12-03:
@LinuxUnix80 wenn Du ein Reboot Problem hast, dann solltest Du es machen, 
wie schlimmchen es hier beschrieben hat: https://github.com/hoylabs/OpenDTU-OnBattery/issues/1437#issuecomment-2514550276

Die Beschreibung zum USB / Serial Logging findest Du in der offiziellen Anleitung zu OpenDTU.
Follow the link to the documentation to setup for USB / serial logging:
https://www.opendtu.solar/firmware/howto/serial_console/

schlimmchen, 2024-12-09:
@LinuxUnix80 Ist dir bewusst, dass wir hier auf deine Hilfe angewiesen sind? Du musst bitte Logs des Neustarts beischaffen, sonst können wir nichts tun. Bei mir beispielsweise tritt das Problem nicht auf.

LinuxUnix80, 2024-12-09:
Moin! Ja, ich Versuch es bis zum Wochenende fertig zu machen. Problem ist die Distanz. DTU ist in der Garage (20m weit weg vom PC).

schlimmchen, 2024-12-09:
Absolut verständlich, dass das lästig ist, geht mir genauso. Leider können wir die Backtraces nicht anders einfangen. Danke für deine Mühe! Hoffentlich können wir die Ursache feststellen und abstellen.

LinuxUnix80, 2024-12-14:
So, ich habe es nun endlich geschafft, dass Seriell Monitoring Log anzufertigen. Es findet ein Core Panic statt:

```bash

[PowerMeterHttpJson] New total: 370.00
[DPL::announceStatus] waiting for sufficiently recent inverter data
[PowerMeterHttpJson] New total: 398.00
[PowerMeterHttpJson] New total: 346.00
[DPL::announceStatus] waiting for sufficiently recent inverter data
[PowerMeterHttpJson] New total: 368.00
[PowerMeterHttpJson] New total: 382.00
[DPL::announceStatus] waiting for sufficiently recent inverter data
[PowerMeterHttpJson] New total: 364.00
[PowerMeterHttpJson] New total: 371.00
[DPL::announceStatus] waiting for sufficiently recent inverter data
[PowerMeterHttpJson] New total: 361.00
[PowerMeterHttpJson] New total: 362.00
[DPL::announceStatus] waiting for sufficiently recent inverter data
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x40380baf  PS      : 0x00060a34  A0      : 0x820b69b2  A1      : 0x3fcebe10  
A2      : 0x3fc96140  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060a23  
A6      : 0x00060a20  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0xffffffff  
A10     : 0x3fca5384  A11     : 0x00000001  A12     : 0x00000006  A13     : 0x00000003  
A14     : 0x02c96140  A15     : 0x00ffffff  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0xffffffff  

Backtrace: 0x40380bac:0x3fcebe10 0x420b69af:0x3fcebe50 0x4208fa95:0x3fcebe70 0x42077811:0x3fcebeb0 0x42026dc0:0x3fcebef0 0x421520dd:0x3fcebf50 0x4205b8be:0x3fcebf70 0x4205ba39:0x3fcebf90 0x4205ba52:0x3fcebfb0 0x42086fbd:0x3fcebfd0

Core  0 register dump:
PC      : 0x420b6712  PS      : 0x00060134  A0      : 0x820b69b5  A1      : 0x3fcace60  
A2      : 0x00c1903f  A3      : 0x01000000  A4      : 0x3fca5384  A5      : 0x00060123  
A6      : 0x00060120  A7      : 0x00000001  A8      : 0x60008800  A9      : 0x00000001  
A10     : 0x3fcef940  A11     : 0x3fcef904  A12     : 0x00000000  A13     : 0x4215b01c  
A14     : 0x02c96140  A15     : 0x00ffffff  SAR     : 0x00000018  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x420b670f:0x3fcace60 0x420b69b2:0x3fcace80 0x42094575:0x3fcacea0 0x4210e90a:0x3fcacec0 0x4210e7cb:0x3fcacee0 0x4210a33d:0x3fcacf00 0x4210172f:0x3fcacf20 0x4215b01c:0x3fcacf60

ELF file SHA256: e874923d4036bc27

E (17130) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4209564e
SPIWP:0xee
mode:DOUT, clock div:2
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
SHA-256 comparison failed:
Calculated: ca4adcc736b98868c479048883edac1969cfd9dde0951723205cf1625628f181
Expected: 27accf466d6a3ffbdd2fe4a9dcdf2a85d67a20eb4b1f2dced8b0c8a88771aee2
Attempting to boot anyway...
entry 0x403c98d0
E (513) esp_core_dump_flash: No core dump partition found!
E (513) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
Initialize FS... done
Reading configuration... Check for default DTU serial... done
done
Reading language pack... done
Reading PinMapping... found valid mapping done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
[SyslogLogger] Disable
[SyslogLogger] Syslog not enabled
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Display... done
Initialize LEDs... done
Initialize Hoymiles interface... NRF: Connection successful
CMT: Connection successful
  Setting country mode... 
  Setting CMT target frequency... 
  Setting radio PA level... 
CMT TX power set to 3 dBm
  Setting DTU serial... 
  Setting poll interval... 
  Setting verbosity... 
  Adding inverter: 116495103970 - Garagensolar done
  Adding inverter: 114494802376 - Garagen-Solar-800W done
done
Initialize Huawei AC charger interface... 
Invalid pin config
Switch to WiFi mode
Setting Hostname... done
Configuring WiFi STA using new credentials... done
Configuring WiFi STA static IP... done
[SyslogLogger] Disable
[SyslogLogger] Syslog not enabled
[DPL::calcNextInverterRestart] automatic inverter restart disabled
[DPL::announceStatus] reloading DPL configuration
[  1001][E][WiFiClient.cpp:249] connect(): connect on fd 49, errno: 118, "Host is unreachable"
```

schlimmchen, 2024-12-14:
Wunderbar! 💪

Erinnert mich an tbnobody#2456. Ich schau trotzdem noch nach, ob im Backtrace etwas vom AsyncWebserver bzw. async_tcp zu sehen ist.

Ist es plausibel, dass du irgendetwas per Web abgerufen hast an der OpenDTU-OnBattery, als das passiert ist? Vielleicht F5 in einem Browser gedrückt oder so?

LinuxUnix80, 2024-12-14:
Aktiv habe ich zu dieser Zeit nichts getan, allerdings war auf dem Loggin-Host ein Browser-Tab geöffnet (die ganze Zeit). Ich habe mal alle Fenster geschlossen und Logge weiter.

Sobald es ein erneuten Reboot gibt, poste ich das Log wieder.

LinuxUnix80, 2024-12-14:
Ein zweiter Reboot (gegen Abend als die Inverter nicht mehr produziert haben). Etwas längerer Auszug, zwecks Kontext:

```code
DPL inverter SERIENNUMMER2]: Starting inverter...
TX PowerControl 868.00 MHz --> 51 95 10 39 70 80 16 05 12 81 00 00 B0 01 2C 
Interrupt received
RX 868.00 MHz --> D1 95 10 39 70 80 16 05 12 81 00 00 00 00 24 00 39 | -53 dBm
Interrupt received
Interrupt received
Interrupt received
Interrupt received
Interrupt received
Interrupt received

[PowerMeterHttpJson] New total: 925.00
[DPL::announceStatus] waiting for a start/stop/restart/limit command to complete
RX Period End
Success
Fetch inverter: SERIENNUMMER1
TX RealTimeRunData 868.00 MHz --> 15 94 80 23 76 80 16 05 12 80 0B 00 67 5D 9E E3 00 00 00 00 00 00 00 00 90 54 DD 
Interrupt received
RX 868.00 MHz --> 95 94 80 23 76 80 16 05 12 01 00 01 01 85 01 86 00 00 00 00 00 02 00 02 00 01 57 | -45 dBm
Interrupt received
RX 868.00 MHz --> 95 94 80 23 76 80 16 05 12 02 BA 5F 00 01 B1 10 00 00 00 00 08 98 13 89 00 00 18 | -45 dBm
Interrupt received
RX 868.00 MHz --> 95 94 80 23 76 80 16 05 12 83 00 00 00 00 00 00 00 27 00 01 5D 64 C9 | -45 dBm
RX Period End
Success

Interrupt received
Interrupt received
Interrupt received
Interrupt received
Interrupt received
Interrupt received

Fetch inverter: SERIENNUMMER2
TX RealTimeRunData 868.00 MHz --> 15 95 10 39 70 80 16 05 12 80 0B 00 67 5D 9E E6 00 00 00 00 00 00 00 00 C0 6B 3A 
Interrupt received
RX 868.00 MHz --> 95 95 10 39 70 80 16 05 12 01 00 01 01 4B 01 4B 00 02 00 01 00 07 00 05 00 04 DD | -54 dBm
Interrupt received
RX 868.00 MHz --> 95 95 10 39 70 80 16 05 12 02 1B EE 00 04 1F 80 00 00 00 00 01 4B 01 4A 00 03 B6 | -54 dBm
Interrupt received
[PowerMeterHttpJson] New total: 942.00
RX 868.00 MHz --> 95 95 10 39 70 80 16 05 12 03 00 02 00 0B 00 09 00 04 20 4F 00 04 26 FB 00 00 69 | -53 dBm
Interrupt received
RX 868.00 MHz --> 95 95 10 39 70 80 16 05 12 04 00 00 08 9A 13 88 00 00 00 00 00 00 00 00 00 23 F6 | -54 dBm
Interrupt received
RX 868.00 MHz --> 95 95 10 39 70 80 16 05 12 85 00 00 C1 08 94 | -53 dBm
RX Period End
Success

[DPL inverter SERIENNUMMER2]: Starting inverter...
TX PowerControl 868.00 MHz --> 51 95 10 39 70 80 16 05 12 81 00 00 B0 01 2C 
Interrupt received
RX 868.00 MHz --> D1 95 10 39 70 80 16 05 12 81 00 00 00 00 24 00 39 | -54 dBm
RX Period End
Success
Fetch inverter: SERIENNUMMER1
TX RealTimeRunData 868.00 MHz --> 15 94 80 23 76 80 16 05 12 80 0B 00 67 5D 9E E9 00 00 00 00 00 00 00 00 30 2A 09 
Interrupt received
RX 868.00 MHz --> 95 94 80 23 76 80 16 05 12 01 00 01 01 86 01 87 00 00 00 00 00 02 00 02 00 01 55 | -45 dBm
Interrupt received
RX 868.00 MHz --> 95 94 80 23 76 80 16 05 12 02 BA 5F 00 01 B1 10 00 00 00 00 08 9D 13 88 00 00 1C | -45 dBm
Interrupt received
RX 868.00 MHz --> 95 94 80 23 76 80 16 05 12 83 00 00 00 00 00 00 00 27 00 01 FB A4 AF | -45 dBm
RX Period End
Success
TX AlarmData 868.00 MHz --> 15 94 80 23 76 80 16 05 12 80 11 00 67 5D 9E E9 00 00 00 00 00 00 00 00 EA 31 D2 
Interrupt received
[... 88 lines cut ...]
    solar-powered, standing by at 0 W
    lower/current/upper limit: 20/0/2000 W, output capability: 0 W
    sending commands enabled, reachable, disqualified
    max reduction production/standby: 0/0 W, max increase: 0 W
    target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
[DPL inverter SERIENNUMMER1]:
    solar-powered, standing by at 0 W
    lower/current/upper limit: 20/0/800 W, output capability: 0 W
    sending commands enabled, reachable, disqualified
    max reduction production/standby: 0/0 W, max increase: 0 W
    target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
[DPL::calcBatteryAllowance] power requested: 100 W
[DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (0 W): 0 W
[PowerMeterHttpJson] Value 1: HTTP Error: connection refused
[DPL::updateInverterLimits] requested: 0 W, producing: 0 W using 0 battery-powered inverters, diff: 0 W, hysteresis: 20 W
[DPL::loop] consumption: 100 W, target output: 100 W (limited to 2800 W), solar inverters output: 0 W, battery allowance: 0 W, battery inverters output: 0 W
[DPL::announceStatus] the system is stable, the last power limit is still valid
[DPL::loop] ******************* ENTER **********************
[DPL::calcHouseholdConsumption] target consumption: -150 W, base load: 100 W
[DPL::calcHouseholdConsumption] power meter value: 0.0 W, power meter valid: no
[DPL::updateInverterLimits] requested: 100 W, producing: 0 W using 2 solar-powered inverters, diff: 100 W, hysteresis: 20 W
[DPL::updateInverterLimits] will cover 0 W using solar-powered inverters
[DPL inverter SERIENNUMMER2]:

```

stefan123t, 2024-12-15:
I specifically looked at the following part:

```
WiFi disconnected: 8
Try reconnecting
[19621387][E][WiFiSTA.cpp:357] disconnect(): disconnect failed!
[19621393][E][WiFiSTA.cpp:305] begin(): config failed
Network lost connection
Disconnected from MQTT.
Disconnect reason:TCP_DISCONNECTED
``` 

I did not spot the `WiFi disconnected: 8` reason yet, but it is the ESP-IDF return code as added here:
https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2325247906

stefan123t, 2024-12-15:
The other subsequent oddity (and therefor unrelated to the previous crashes) I see in both logs is this

```
[DPL::calcNextInverterRestart] automatic inverter restart disabled
[DPL::announceStatus] reloading DPL configuration
[   958][E][WiFiClient.cpp:249] connect(): connect on fd 49, errno: 118, "Host is unreachable"
```

LinuxUnix80, 2024-12-20:
Guten Morgen,

soeben wieder ein Absturz, nachdem ein (autom.) Zeitabgleich ausgeführt wurde:

[image]

[image]

```code
[DPL inverter INV1]: actual limit is 49.2 % (984 W respectively), effective 2239 ms after update started, requested were 49.2 %
[DPL::announceStatus] waiting for sufficiently recent inverter data
Fetch inverter: 116495103970
TX RealTimeRunData 868.00 MHz --> 15 95 10 39 70 80 16 05 12 80 0B 00 67 65 31 7A 00 00 00 00 00 00 00 00 14 71 FF 
Interrupt received
RX 868.00 MHz --> 95 95 10 39 70 80 16 05 12 01 00 01 01 AF 01 AA 00 8A 00 89 02 54 02 47 00 04 C9 | -53 dBm
Interrupt received
RX 868.00 MHz --> 95 95 10 39 70 80 16 05 12 02 1D B6 00 04 21 45 00 23 00 23 01 A9 01 A9 00 87 96 | -53 dBm
Interrupt received
RX 868.00 MHz --> 95 95 10 39 70 80 16 05 12 03 00 89 02 3B 02 44 00 04 22 1F 00 04 28 CC 00 21 D5 | -53 dBm
Interrupt received
RX 868.00 MHz --> 95 95 10 39 70 80 16 05 12 04 00 22 08 9D 13 8B 08 A6 00 00 00 64 03 E8 00 4B 99 | -53 dBm
Interrupt received
RX 868.00 MHz --> 95 95 10 39 70 80 16 05 12 85 00 01 43 3F 20 | -53 dBm
RX Period End
Success

[DPL::announceStatus] waiting for sufficiently recent power meter reading
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x40380baf  PS      : 0x00060e34  A0      : 0x820b69b2  A1      : 0x3fcebe10  
A2      : 0x3fc96140  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060e23  
A6      : 0x00060e20  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0xffffffff  
A10     : 0x3fca5384  A11     : 0x00000001  A12     : 0x00000006  A13     : 0x00000003  
A14     : 0x02c96140  A15     : 0x00ffffff  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0xffffffff  

Backtrace: 0x40380bac:0x3fcebe10 0x420b69af:0x3fcebe50 0x4208fa95:0x3fcebe70 0x42077811:0x3fcebeb0 0x42026dc0:0x3fcebef0 0x421520dd:0x3fcebf50 0x4205b8be:0x3fcebf70 0x4205ba39:0x3fcebf90 0x4205ba52:0x3fcebfb0 0x42086fbd:0x3fcebfd0

Core  0 register dump:
PC      : 0x420b670f  PS      : 0x00060d34  A0      : 0x820b69b5  A1      : 0x3fcc96b0  
A2      : 0x00c19048  A3      : 0x01000000  A4      : 0x3fca5384  A5      : 0x00060d23  
A6      : 0x00060d20  A7      : 0x00000001  A8      : 0x60008800  A9      : 0x00000001  
A10     : 0x3fcef940  A11     : 0x3fcef904  A12     : 0x00000000  A13     : 0x4215b01c  
A14     : 0x02c96140  A15     : 0x00ffffff  SAR     : 0x0000001b  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x420b670c:0x3fcc96b0 0x420b69b2:0x3fcc96d0 0x42094575:0x3fcc96f0 0x4210e90a:0x3fcc9710 0x4210e7cb:0x3fcc9730 0x4210a33d:0x3fcc9750 0x4210172f:0x3fcc9770 0x4215b01c:0x3fcc97b0

ELF file SHA256: e874923d4036bc27

E (7089) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4209564e
SPIWP:0xee
mode:DOUT, clock div:2
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
SHA-256 comparison failed:
Calculated: ca4adcc736b98868c479048883edac1969cfd9dde0951723205cf1625628f181
Expected: 27accf466d6a3ffbdd2fe4a9dcdf2a85d67a20eb4b1f2dced8b0c8a88771aee2
Attempting to boot anyway...
entry 0x403c98d0
E (513) esp_core_dump_flash: No core dump partition found!
E (513) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
Initialize FS... done
Reading configuration... Check for default DTU serial... done
done

Reading language pack... done
Reading PinMapping... found valid mapping done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
[SyslogLogger] Disable
[SyslogLogger] Syslog not enabled
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Display... done
Initialize LEDs... done
Initialize Hoymiles interface... NRF: Connection successful

CMT: Connection successful
  Setting country mode... 
  Setting CMT target frequency... 
  Setting radio PA level... 
CMT TX power set to 3 dBm
  Setting DTU serial... 
  Setting poll interval... 
  Setting verbosity... 
  Adding inverter: INV1 - Garagensolar done
  Adding inverter: INV2 - Garagen-Solar-800W done
done

Initialize Huawei AC charger interface... 
Invalid pin config
Switch to WiFi mode
Setting Hostname... done
Configuring WiFi STA using new credentials... done
Configuring WiFi STA static IP... done
[SyslogLogger] Disable
[SyslogLogger] Syslog not enabled
[DPL::calcNextInverterRestart] automatic inverter restart disabled
[DPL::announceStatus] reloading DPL configuration
[   985][E][WiFiClient.cpp:249] connect(): connect on fd 49, errno: 118, "Host is unreachable"

[DPL::loop] ******************* ENTER **********************
[DPL::calcHouseholdConsumption] target consumption: -800 W, base load: 100 W
[DPL::calcHouseholdConsumption] power meter value: 0.0 W, power meter valid: no
[DPL::updateInverterLimits] requested: 100 W, producing: 0 W using 2 solar-powered inverters, diff: 100 W, hysteresis: 20 W
[PowerMeterHttpJson] Value 1: HTTP Error: connection refused
[DPL::updateInverterLimits] will cover 0 W using solar-powered inverters
[DPL inverter INV1]:
```

schlimmchen, 2024-12-20:
Dein erster und dritter Log sagen, dass sich irgendetwas beim Auslesen der Temperatur verklemmt hat.

<details>
<summary>Backtrace Absturz 1</summary>

```
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x40380baf  PS      : 0x00060a34  A0      : 0x820b69b2  A1      : 0x3fcebe10  
A2      : 0x3fc96140  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060a23  
A6      : 0x00060a20  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0xffffffff  
A10     : 0x3fca5384  A11     : 0x00000001  A12     : 0x00000006  A13     : 0x00000003  
A14     : 0x02c96140  A15     : 0x00ffffff  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0xffffffff  

Backtrace: 0x40380bac:0x3fcebe10 0x420b69af:0x3fcebe50 0x4208fa95:0x3fcebe70 0x42077811:0x3fcebeb0 0x42026dc0:0x3fcebef0 0x421520dd:0x3fcebf50 0x4205b8be:0x3fcebf70 0x4205ba39:0x3fcebf90 0x4205ba52:0x3fcebfb0 0x42086fbd:0x3fcebfd0

Core  0 register dump:
PC      : 0x420b6712  PS      : 0x00060134  A0      : 0x820b69b5  A1      : 0x3fcace60  
A2      : 0x00c1903f  A3      : 0x01000000  A4      : 0x3fca5384  A5      : 0x00060123  
A6      : 0x00060120  A7      : 0x00000001  A8      : 0x60008800  A9      : 0x00000001  
A10     : 0x3fcef940  A11     : 0x3fcef904  A12     : 0x00000000  A13     : 0x4215b01c  
A14     : 0x02c96140  A15     : 0x00ffffff  SAR     : 0x00000018  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x420b670f:0x3fcace60 0x420b69b2:0x3fcace80 0x42094575:0x3fcacea0 0x4210e90a:0x3fcacec0 0x4210e7cb:0x3fcacee0 0x4210a33d:0x3fcacf00 0x4210172f:0x3fcacf20 0x4215b01c:0x3fcacf60

ELF file SHA256: e874923d4036bc27

top backtrace:

0x40380bac: esp_ptr_external_ram at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/soc/include/soc/soc_memory_types.h:115
 (inlined by) spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:99
 (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:301
0x420b69af: vPortEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
 (inlined by) temp_sensor_get_raw_value at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32s3/sar_periph_ctrl.c:181
0x4208fa95: temp_sensor_read_celsius at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/esp32s3/rtc_tempsensor.c:146 (discriminator 2)
0x42077811: CpuTemperatureClass::read() at /home/beki/Documents/OpenDTU-OnBattery/lib/CpuTemperature/src/CpuTemperature.cpp:40
0x42026dc0: MqttHandleDtuClass::loop() at /home/beki/Documents/OpenDTU-OnBattery/src/MqttHandleDtu.cpp:47
0x421520dd: void std::__invoke_impl<void, void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&>(std::__invoke_memfun_deref, void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&) at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/invoke.h:73
 (inlined by) std::__invoke_result<void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&>::type std::__invoke<void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&>(void (MqttHandleDtuClass::*&)(), MqttHandleDtuClass*&) at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/invoke.h:95
 (inlined by) void std::_Bind<void (MqttHandleDtuClass::*(MqttHandleDtuClass*))()>::__call<void, , 0u>(std::tuple<>&&, std::_Index_tuple<0u>) at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/functional:400
 (inlined by) void std::_Bind<void (MqttHandleDtuClass::*(MqttHandleDtuClass*))()>::operator()<, void>() at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/functional:484
 (inlined by) std::_Function_handler<void (), std::_Bind<void (MqttHandleDtuClass::*(MqttHandleDtuClass*))()> >::_M_invoke(std::_Any_data const&) at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:297
0x4205b8be: std::function<void ()>::operator()() const at /home/beki/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:687
0x4205ba39: Scheduler::execute() at /home/beki/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/TaskScheduler/src/TaskScheduler.h:1552
 (inlined by) Scheduler::execute() at /home/beki/Documents/OpenDTU-OnBattery/.pio/libdeps/generic_esp32s3_usb/TaskScheduler/src/TaskScheduler.h:1351
0x4205ba52: loop() at /home/beki/Documents/OpenDTU-OnBattery/src/main.cpp:200
0x42086fbd: loopTask(void*) at /home/beki/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
WARNING: trace file checksum 'e874923d4036bc27' refers to a different ELF file.
         we are working with ELF file with checksum '807b8fd251ae627dbcb6a48044440fcc0f344881d61afde17bedf8070c5d9a83'.

bottom backtrace:
[... cut ...]
Für mich deutet das auf ein Hardwareproblem hin. Entweder ein Defekt am ESP32-S3 intern, sodass der Temperatursensor nicht fertig wird oder das Bit flippt, oder was banales wie eine schlechte Stromversorgung. Stichpunkt Stromversorgung: Wie ist dein Board mit Strom versorgt? Kannst du mal eine Weile ein anderes Netzteil und ein anderes Kabel verwenden? Gerne 2A Netzteil oder mehr, und ein Kabel dessen Dicke potentiell darauf hindeutet, dass es die 5V auch bei 1A bis zum Fusion-Board transportieren kann?

schlimmchen, 2025-01-08:
@LinuxUnix80 Hattest du Gelegenheit, zumindest die Spannungsversorgung deines Boards zu überprüfen? Ist der Fehler vielleicht sogar nicht mehr aufgetreten?

LinuxUnix80, 2025-01-08:
Moin, ich hatte ja zwecks serielles Monitoring, die DTU am USB Port vom Mini PC.

Ich habe mal DPL ausgeschaltet (da im Winter nicht so sinnvoll). Wenn DPL aus ist, tritt dieses Problem nicht auf.

LinuxUnix80, 2025-01-29:
Also mit der neuesten Version läuft es aktuell sehr stabil. Die DTU hat schon 102 Stunden ohne Reboot!

closed 2025-01-29 by LinuxUnix80

schlimmchen, 2025-01-31:
Danke für deine Rückmeldung und das selbstständige Schließen des Issues 💪