hoylabs/OpenDTU-OnBattery #1440: [Request] Add Huawei Charger R4875G1 support (additional to R4850G2)

gitisgreat2023, 2024-12-03:
### Is your feature request related to a problem? Please describe.

I would like to use the Huawei R4875G1 in my OpenDTU-onBattery setup. Currently on the Huawei R4850G2 is supported.

### Describe the solution you'd like

It seems only [a scaling factor](https://www.akkudoktor.net/t/4000w-45-58v-laderegler-esphome-and-mqtt-control-and-adjust-current-and-voltage-parameters/24393) has to be adapted. And maybe some CAN IDs.

See [here](https://github.com/mjpalmowski/CAN-BUS-control-R4875G1-with-ESPHome-and-MQTT/blob/main/R4875G2.YAML)  for a YAML code for the R4875G1.

### Additional context

I'm happy to test the code.

AndreasBoehm, 2025-01-21:
We could try to autodetect the exact model of the huawei charger and set the scaling factors accordingly

https://github.com/mjpalmowski/CAN-BUS-control-R4875G1-with-ESPHome-and-MQTT/blob/b8f4bd32a802e63911a6bcb3a90f889bfd6fc157/R48xxGx%20autodetect-autoscaling-dual-inverter-example.yaml#L215-L238

gitisgreat2023, 2025-01-22:
Perfect... happy to test a build (4MB_nota please). Just ordered a R4875G1... :-)

schlimmchen, 2025-01-22:
@AndreasBoehm In that project there is pure gold. This one got me excited a couple of days ago: https://github.com/mjpalmowski/CAN-BUS-control-R4875G1-with-ESPHome-and-MQTT/discussions/4#discussioncomment-11681537 Had no time yet to explore it... I desperately want that to we know the model, and can at least infer the correct scaling factors. However, I also saw that those guys are onto reading the factor out at runtime. I will probably stumble over that as well.

@gitisgreat2023 I suspect your unit will not set and/or report the correct current limit. Let us know. I will ask you to test stuff once I am ready with something. Thank you for your interest in testing!

schlimmchen, 2025-02-02:
@gitisgreat2023 Progress is slow, but there is progress. How is your hardware setup coming along? Please let me know when you are ready. You should be able to talk to the R4875 just fine, but the max current value will not match the actual output.

gitisgreat2023, 2025-02-03:
@schlimmchen the Huawei arrived, but I have to wait for the adapter still though, that might take still some time. I'll write when I've wired it up...
Hopefully I managed to get the G2 running on the Pro PCT, because I want to test and run the G1 on the "old" board.... as you noticed I still struggle to get the G2 running on the Pro PCB ;-)

commit b477db70 (2023-11-16): Fix #1440: Add CSS code to stop numbers changing the size of their container
PR #1806 (2025-05-21): Feature: Improved support for R4850 and support for R48xx in general
This changeset improves upon the integration for R4850 power supplies by implementing new features/functions like

* reading board properties
* setting fan control modes
* shutdown over CAN
* processing ACKs from the PSU
* auto-setting offline limits (user-configurable)

and adds general support for (hopefully) all Huawei R48xx power supplies.

[image]

Closes #1544 (set (user-configurable) offline limits automatically)
Closes #1440 (support for R48xx in general, here: R4875)
Closes #435 (there is now a MQTT topic that allows [... cut]
closed 2025-05-21 by schlimmchen