hoylabs/OpenDTU-OnBattery #1445: DPL verbose logging eingeschaltet

Exodusnick, 2024-12-05:
### What happened?

Von FredonZen Der DPL gibt teilweise bis zu 300W....
https://github.com/tbnobody/OpenDTU/discussions/2153#discussioncomment-11420999

### To Reproduce Bug

Hoymiles 16004t / hoymiles 20004t firmware 1.1.12
Aktuelle Open dtu on Batterie version 2024.11.20

Mit der hoymiles 0.0.27 funktioniert es ohne Probleme.

### Expected Behavior

Keine Nulleinspeisung mehr möglich 
Gibt bis zu 300 w ins Netz ein 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.11.20

### What firmware variant (PIO Environment)?

generic_esp32s3

### Relevant log/trace output

```Shell
Keine vorhanden.
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2024-12-05:
Waiting for @FredonZen's Logs.

spcqike, 2024-12-05:
öhm, kann man den Titel passender gestallten? sowas wie "Nulleinspeisung mit WR Firmware 1.1.12 nicht möglich" oder so?

Exodusnick, 2024-12-05:
> öhm, kann man den Titel passender gestallten? sowas wie "Nulleinspeisung mit WR Firmware 1.1.12 nicht möglich" oder so?

Erledigt 👍

Exodusnick, 2024-12-07:
Hallo https://github.com/FredonZen wir benötigen dein Log mit der aktuellen hoymiles Firmware, um den Fehler zu finden warum Der DPL teilweise bis zu 300W ins Netz zurück gibt.

FredonZen, 2024-12-07:
> Hallo @FredonZen <https://github.com/FredonZen> wir benötigen dein Log

Hallo, leider muss ich warten, bis mal wieder die Sonne scheint. VG

Thaui57, 2024-12-08:
Habe die Batterie zwar abgeschlossen aber die Nulleinspeisung regelt relative genau um den Null-Punkt. Obwohl die Wintermonate eine echte Herausforderung sind bei der Schwachlast. 1.1.12 scheint keinen Einfluss zu haben.

schlimmchen, 2024-12-09:
@FredonZen Ich hab noch eine Idee, wie wir das abkürzen können: Kannst du uns mal Screenshots erstellen von deinen DPL-Einstellungen?

FredonZen, 2024-12-09:
Hallo, sehr gern. Vielen Dank im voraus für den Einsatz!

schlimmchen, 2024-12-09:
Waren da Screenshots als Anhänge in deiner E-Mail, oder kommt demnächst was?

FredonZen, 2024-12-09:
Ja waren sie. Versuche es nochmal einzeln. Hier 1. Bild

FredonZen, 2024-12-09:
> 2. Bild
> 22:59:

3. Bild

Exodusnick, 2024-12-09:
Du musst die direkt auf GitHub einführen per Email funktioniert das net wir können die nicht sehen .

Einfach per paste add Files die Bilder hochladen.

spcqike, 2024-12-10:
Bilder sind hier 
https://github.com/tbnobody/OpenDTU/discussions/2153#discussioncomment-11516814

schlimmchen, 2024-12-10:
Danke, @spcqike, hab ich schon bemerkt, aber leider ist "kompensiere Verschattung" ausgeschaltet. Das wäre nämlich logisch, dass man den Schalter ausmachen muss mit der neuen Firmware, die ja von selbst die vorhandene Eingangsleistung so umsetzt, dass tatsächlich die gewünschte Ausgangsleistung (das Limit) erreicht wird.

spcqike, 2024-12-10:
Das wäre auch meine erste Vermutung gewesen. 

vielleicht verhalten sich die WR nun in manchen Last Szenarien anders? So wie mein HMS-2000 Lasten <15% gar nicht sauber ausregeln kann und da immer zu wenig liefert. 

Dem kommt man aber nur mit passenden Logs hinterher (: 

also, @FredonZen bitte einmal das verbose logging einschalten und einen Konsolen log Auszug bereitstellen, wo das Problem der nicht sauberen Regelung auftritt.

FredonZen, 2024-12-10:
Hallo, sollte es die Sonne mal wieder zulassen, werde ich das Log hier
posten. Ich bin total geflasht von Eurem Support! Danke VG

Jonrembo, 2024-12-20:
Hallo zusammen,
Ich hab das gleiche Problem mit der neuen Firmware. Gibt es dazu schon eine Lösung? vorher hat alles super funktioniert.

11:53:24.807 > Success
11:53:25.204 > [PowerMeterHttpJson] New total: -184.00
11:53:25.226 > [DPL::loop] ******************* ENTER **********************
11:53:25.235 > [DPL::calcHouseholdConsumption] target consumption: 0 W, base load: 130 W
11:53:25.242 > [DPL::calcHouseholdConsumption] power meter value: -184.0 W, power meter valid: yes
11:53:25.279 > [DPL::calcHouseholdConsumption] inverter xxxxxxx is behind power meter producing 758 W
11:53:25.324 > [DPL::updateInverterLimits] requested: 574 W, producing: 758 W using 1 solar-powered inverters, diff: -184 W, hysteresis: 5 W
11:53:25.336 > [DPL inverter xxxxxx]: 2/4 mppts are producing, scaling from 574 to 1148 W
11:53:25.339 > [DPL::updateInverterLimits] will cover 574 W using solar-powered inverters
11:53:25.352 > [DPL inverter xxxxxx]:
11:53:25.383 >     solar-powered, producing 758 W
11:53:25.409 >     lower/current/upper limit: 40/1152/800 W, output capability: 1600 W
11:53:25.425 >     sending commands enabled, reachable, eligible
11:53:25.433 >     max reduction production/standby: 718/718 W, max increase: 65178 W
11:53:25.440 >     target limit/output/state: 1148 W (update)/574 W/production, 0 update timeouts
11:53:25.449 > [DPL::calcBatteryAllowance] power requested: 0 W
11:53:25.522 > [DPL::updateInverterLimits] requested: 0 W, producing: 0 W using 0 battery-powered inverters, diff: 0 W, hysteresis: 5 W
11:53:25.530 > [DPL::loop] consumption: 574 W, target output: 574 W (limited to 800 W), solar inverters output: 574 W, battery allowance: 0 W, battery inverters output: 0 W
11:53:25.539 > [DPL inverter xxxxxxx]: sending limit of 71.8 % (1148 W respectively), max output is 1600 W
11:53:25.550 > TX ActivePowerControl 865.00 MHz --> 51 A0 09 99 D5 80 16 04 92 81 0B 00 02 CD 00 01 E7 50 47 
11:53:25.585 > [DPL::announceStatus] waiting for a start/stop/restart/limit command to complete
11:53:25.624 > Interrupt received
11:53:25.633 > RX 865.00 MHz --> D1 A0 09 99 D5 80 16 04 92 81 00 00 0B 00 14 07 AD | -60 dBm
11:53:27.316 > RX Period End
11:53:27.327 > Success
11:53:27.368 > Fetch inverter: xxxxxxxx
11:53:27.378 > [DPL inverter xxxxxxxx]: actual limit is 71.7 % (1147 W respectively), effective 2041 ms after update started, requested were 71.8 %
11:53:27.387 > [DPL::announceStatus] waiting for sufficiently recent inverter data

Liebe Grüße,
Jon

Exodusnick, 2024-12-20:
Nein aber dank dein log kann @schlimmchen vielleicht den Fehler finden .
Vielen Dank für dein log👍

Jonrembo, 2024-12-20:
wow super für die schnelle Antwort! Wenn ihr da noch mehr braucht sagt gerne bescheid!

schlimmchen, 2024-12-20:
> das gleiche Problem mit der neuen Firmware

Nämlich? Ist dein Problem, dass der WR zu viel Leistung abgibt, also mehr als dein Haushalt benötigt?

> [DPL inverter xxxxxx]: 2/4 mppts are producing, scaling from 574 to 1148 W

Du hast "Verschattung durch Überskalierung ausgleichen" offensichtlich eingeschaltet -- oder nicht? Das musst du ausmachen mit der neuen Firmware, weil die "Skalierung" der WR schon selbst vornimmt.

Ich denke wir sollten das Skalieren pauschal unterbinden wenn wir einen WR vorfinden, der eine entsprechende Firmware hat. Siehe #1472.

Jonrembo, 2024-12-20:
ja genau. so ist es.

Verschattung durch überskalierung ist natürlich ausgeschaltet. 
[image]

schlimmchen, 2024-12-20:
> [DPL inverter xxxxxx]: 2/4 mppts are producing, scaling from 574 to 1148 W

Du hast nur zwei Eingänge an Module angeschlossen? Dann wird trotzdem skaliert...

@FredonZen Hast du an deinem WR alle Eingänge mit Modulen versorgt, oder sind da Eingänge ungenutzt?

Jonrembo, 2024-12-20:
Ja genau, ich nutze aktuell nur 2 Eingänge.

Exodusnick, 2024-12-23:
See here
hoylabs#1473

Josef1966, 2024-12-31:
Gibt es Unterschiede in der Performance, wenn man anstelle von 2 Eingängen alle 4 Eingänge am HMS-1600-4T nutzen würde?

spcqike, 2024-12-31:
Klar. 

Wenn nur 2 Eingänge belegt sind kann der WR nur knapp 800W liefern. Ein HMS-1600 sollte nur 400W pro Eingang verarbeiten.

Josef1966, 2024-12-31:
Ja klar, aber kann man das im Hinblick auf die Nulleinspeisung mit der WR-Firmware 1.1.12 problemlos variieren? Bei mir hat der WR bei 4er-Belegung zu viel ins Netz eingespeist, also die Nulleinspeisung hat nicht funktioniert. Ich betreibe das Ganze aber auch in Kombination mit einem Zendure SolarFlow Hub2000 und AB2000-Akku.
Ich weiß nicht, ob dieses Problem hier auch schon mitbehandelt wurde.

spcqike, 2024-12-31:
Was willst du variieren? 

Der WR versucht das eingestellte Limit zu erzeugen. Dabei kann er, soweit ich weiß, bis zu MAX/n pro Eingang liefern, wobei n die Anzahl der verbauten MPPT ist. Also 1600/4 bzw 400W beim HMS-1600 

die alte wechselrichter Firmware hat das gesetzte Limit einfach gleich auf alle Eingänge verteilt. Du wolltest 800W? Dann durfte jeder MPPT bis zu 200W liefern. Egal ob er mehr konnte oder nicht. 

„Dagegen“ wurde in openDTU das überskalieren eingebaut. Du brauchst 800W, hast aber nur 2 Eingänge belegt? Dann wurde 1600W als Limit übertragen, was wiederum 400W pro MPPT erzeugt und damit eben 800W bei nur zwei belegten Eingängen liefert. 

Nun hat hoymiles die Firmware angepasst, so dass das feste Limit der MPPT aufgehoben wurde. Bei 800W können die einzelnen MPPT nun unterschiedlich viel beisteuern, je nachdem was am Eingang grade verfügbar ist. Aber, wahrscheinlich weiterhin maximal 400W pro MPPT.

nun kann es sein, dass die in openDTU vorhandenen Regelungen noch nicht ideal mit der neuen Firmware der WR arbeiten, also bspw noch zu hohe Limits überragen. Das wird ja bereits in verschiedenen issues, Diskussionen und PRs besprochen und angegangen. 

Wie gesagt, was willst du variieren? So ganz verstehe ich deine Frage noch nicht (: 

Aber allgemein gesprochen: alle Eingänge zu belegen ist besser, als nur die Hälfte zu nutzen. Allein schon weil die Ströme pro Eingang halbiert und die Verluste damit geviertelt werden.

Josef1966, 2025-01-01:
Aktuelle Situation: DPL arbeitet nicht, es wird über 200 Watt ins Netz eingespeist, obwohl Limit im DPL auf -15 Watt begrenzt ist. Am HMS-1600-4T sind 2 Eingänge belegt. WR hängt an Zendure SolarFlow Hub2000 und AB2000-Akku.

[image]
[Log.txt](https://github.com/user-attachments/files/18285281/Log.txt)
[image]

spcqike, 2025-01-01:
Ich denke du hast die Lösung für dein Problem bereits gefunden. Aber noch nicht umgesetzt 

https://github.com/hoylabs/OpenDTU-OnBattery/pull/1473#issuecomment-2566908008

Josef1966, 2025-01-01:
Ok, ich würde ja am liebsten alle 4 strings verwenden, aber bisher dachte ich immer, dann wären die Probleme noch grösser. So hab ich es auf jeden Fall in den verschiedenen Foren gelesen, also dass man beim HMS-1600 besser nur 2 anstelle von 4 Eingängen belegt (Aufteilung Hub-Ausgang auf WR-Eingänge).
Welche Aussagen gibts hier zu diesem Thema?

Exodusnick, 2025-01-01:
Hab ich dir doch oben geschrieben die Aufteilung 
Bei Verwendung von zwei Eingänge 
Entweder oben links und rechts oder unten links und rechts ...... Und nicht alle beide auf einer Seite.

Exodusnick, 2025-01-01:
Ja kannst du .👌

Exodusnick, 2025-01-01:
> Ja kannst du .👌

Ich habe diese sind am besten 
[image]

Exodusnick, 2025-01-01:
Sorry zu schnell gedrückt und ausversehen das issue geöffnet 😂

closed 2025-01-01 by Exodusnick

Josef1966, 2025-01-01:
Die habe ich schon :-)
Ich muss mir nur noch weitere Anschluss-Kabel-Verlängerungen besorgen, es sollen ja alle 4 Eingänge wegen der Symmetrie mit gleich langen Kabeln angeschlossen werden. Und mit den Y-Verteilern alleine komme ich nicht gleichmässig auf beide Seiten des WR.

Sorbat, 2025-01-01:
> Ja kannst du .
> 
> Ich benutzt diese hier wenn du Fragen solltest welche am besten sind 😉 
> 
> https://amzn.eu/d/dqQFv5S
> 
> 
> 
> 

Warum sind diese am Besten?

Exodusnick, 2025-01-01:
Weil ich vorher diese hatte.
https://amzn.eu/d/dqQFv5S