hoylabs/OpenDTU-OnBattery #1453: [Bug] Huawei AC charger interface must not rely on "power" pin

schlimmchen, 2024-12-10:
### What happened?

Trying to connect my R4850G2.

### To Reproduce Bug

Create a valid pin mapping for object "huawei" but forget to define the power pin. The pin config will be determined to be invalid.

### Expected Behavior

Controlling the "power" pin is optional and not specifying the pin in the `pin_mapping.json` shall not prevent initialization of the communication interface.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.11.20

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Manos1966, 2024-12-12:
Indeed.... (Wiki)

[image]

schlimmchen, 2024-12-12:
I saw that note, which is why I figured it out fast. But it does not change the fact that it should work without said pin.

Manos1966, 2024-12-13:
The best compliment one could expect! (that our Wiki has information even for the most experienced of us)

I agree 100% percent with you and I am delighted to hear that you are spending time to improve this module.

I am very happy with it (Huawei), once every three weeks I charge my Pylontechs to 50% - Very easy! 💯 

Out of context: One of the **best** options you guys made available: "Use battery at night **even if only partially charged**" 🥇 
With one flick of the switch (no need to touch the DPL values) I have the system in winter running:
 - With option INACTIVE, and START at 95% it can days but I will have the batteries full (important for the SOC% re-calibration, Pylontechs have less of a problem but other BMSs need it regulalry).
 - With option ACTIVE, and START at 95% I can use the battery no matter how full they got
 - In between, if the winter weather is terrible for days, I can activate the HUAWEI to keep the batteries charged

Excellent! 🥇 

(warum schreiben wir in Englisch? 🤦‍♂️ )

schlimmchen, 2025-01-18:
https://github.com/hoylabs/OpenDTU-OnBattery/blob/ea7ab874fd763c1ccee4f2e9e7c34533b38682a5/src/gridcharger/huawei/Controller.cpp#L80-L85

According to `git blame` I addressed this in 01a42b231, which is part of my effort to support the internal TWAI to talk to the Huawei PSU.

closed 2025-01-18 by schlimmchen