hoylabs/OpenDTU-OnBattery #1483: Battery limit not applied consistently, leading to inverter flickering on/off

ranma, 2024-12-26:
### What happened?

In release 2024.11.20 (latest), calcBatteryAllowance sometimes does not correctly apply battery limits, additionally it looks like start threshold (90%) and discharge even if partially full (off) were not applied as expected, since the battery was <90% the whole day:

Relevant log excerpt (only got last ~1 hour or so):
```
$ egrep "will cover|calcBatteryAllow" ~/OpenDTU.log
2024-12-26T15:17:19.031722+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:19.037855+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:20.057608+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:20.069026+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:21.108985+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:21.119856+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::updateInverterLimits] will cover 200 W using battery-powered invert>
2024-12-26T15:17:37.967869+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:37.973637+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:37.986608+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::updateInverterLimits] will cover 0 W using battery-powered inverters
2024-12-26T15:17:48.208731+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:48.214849+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:48.413243+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:48.422881+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:48.950031+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:48.957620+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:50.074476+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:50.082122+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:51.384988+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:51.390971+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:52.243860+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:52.298244+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:53.427772+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:53.434751+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:54.398986+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:54.405097+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:55.433727+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:55.444385+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:56.464993+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:56.476356+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:57.498877+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:57.506602+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:58.871537+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:58.880356+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:17:59.810887+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:17:59.817610+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:18:00.856482+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:18:00.884876+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::updateInverterLimits] will cover 200 W using battery-powered invert>
2024-12-26T15:18:17.883497+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:18:17.891235+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:18:17.898022+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::updateInverterLimits] will cover 0 W using battery-powered inverters
2024-12-26T15:18:37.986880+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:18:37.995110+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:18:38.226010+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:18:38.237613+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:18:38.799411+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:18:38.804365+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:18:39.828618+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:18:39.844490+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::updateInverterLimits] will cover 200 W using battery-powered invert>
2024-12-26T15:18:47.859114+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:18:47.878367+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::updateInverterLimits] will cover 234 W using battery-powered invert>
2024-12-26T15:18:57.866343+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:18:57.876984+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:18:57.883849+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::updateInverterLimits] will cover 0 W using battery-powered inverters
2024-12-26T15:19:18.570975+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
2024-12-26T15:19:18.588855+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (>
2024-12-26T15:19:18.709375+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W
[...]
```

### To Reproduce Bug

Unclear, may be specific to my configuration? TBD.

### Expected Behavior

- Inverter should not flicker on/off rapidly
- Inverter should not be enabled if battery < start threshold

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

81b1e6e

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

ranma, 2024-12-26:
[OpenDTU.log](https://github.com/user-attachments/files/18254343/OpenDTU.log)

ranma, 2024-12-26:
Mit der aktuellen Version vom 22.11. hab' ich auch P robleme, er hat heute morgen und jetzt abends mit der Inverterleistung geflackert, obwohl der aus sein sollte (SoC < Startschwelle und nachts entladen aus):

```
[DPL::loop] ******************* ENTER **********************
2024-12-26T15:32:58.875250+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::loop] battery interface enabled, SoC: 40.3 %, StartTH: 90 %, StopTH: 30 %, SoC age: 0 s, ignore: no
2024-12-26T15:32:58.883136+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::getBatteryVoltage] BMS: 48.97 V, MPPT: 48.53 V, inverter 114184614039: 48.90 V, returning: 48.97V
2024-12-26T15:32:58.888151+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::loop] dcVoltage: 48.97 V, loadCorrectedVoltage: 49.06 V, StartTH: 51.50 V, StopTH: 45.94 V
2024-12-26T15:32:58.899241+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::loop] StartTH reached: no, StopTH reached: no, SolarPT enabled, use at night: no
2024-12-26T15:32:58.912855+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcHouseholdConsumption] target consumption: 10 W, base load: 200 W
2024-12-26T15:32:58.926049+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcHouseholdConsumption] power meter value: 1010.0 W, power meter valid: yes
2024-12-26T15:32:58.931730+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcHouseholdConsumption] inverter 114184614039 is behind power meter producing 95 W
2024-12-26T15:32:58.946864+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::updateInverterLimits] requested: 200 W, producing: 0 W using 0 solar-powered inverters, diff: 200 W, hysteresis: 20 W
2024-12-26T15:32:58.965111+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::calcBatteryAllowance] power requested: 200 W 
2024-12-26T15:32:58.976221+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::updateInverterLimits] requested: 200 W, producing: 95 W using 1 battery-powered inverters, diff: 105 W, hysteresis: 20 W
2024-12-26T15:32:58.980490+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::updateInverterLimits] will cover 210 W using battery-powered inverters
2024-12-26T15:32:58.986360+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL inverter 114184614039]:
2024-12-26T15:32:58.996225+00:00 OpenDTU-3480B8 OpenDTU[74b53816]    battery-powered, producing 95 W
2024-12-26T15:32:59.006234+00:00 OpenDTU-3480B8 OpenDTU[74b53816]    lower/current/upper limit: 40/105/800 W, output capability: 800 W
2024-12-26T15:32:59.011351+00:00 OpenDTU-3480B8 OpenDTU[74b53816]    sending commands enabled, reachable, eligible
2024-12-26T15:32:59.018252+00:00 OpenDTU-3480B8 OpenDTU[74b53816]    max reduction production/standby: 55/95 W, max increase: 695 W
2024-12-26T15:32:59.023101+00:00 OpenDTU-3480B8 OpenDTU[74b53816]    target limit/output/state: 210 W (update)/210 W/production, 0 update timeouts
2024-12-26T15:32:59.035103+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::loop] consumption: 1095 W, target output: 200 W (limited to 200 W), solar inverters output: 0 W, battery allowance: 200 W, battery inverters output: 210 W
2024-12-26T15:32:59.043725+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL inverter 114184614039]: sending limit of 26.2 % (210 W respectively), max output is 800 W
2024-12-26T15:32:59.045227+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::announceStatus] waiting for a start/stop/restart/limit command to complete
2024-12-26T15:32:59.401935+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Last missing
2024-12-26T15:32:59.411863+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Request retransmit: 12
2024-12-26T15:32:59.508656+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Last missing
2024-12-26T15:32:59.517617+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Request retransmit: 12
2024-12-26T15:32:59.614936+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Middle missing
2024-12-26T15:32:59.627250+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Request retransmit: 1
2024-12-26T15:32:59.728108+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Middle missing
2024-12-26T15:32:59.735995+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Request retransmit: 1
2024-12-26T15:32:59.837176+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Middle missing
2024-12-26T15:32:59.838015+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Request retransmit: 2
2024-12-26T15:32:59.967983+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Middle missing
2024-12-26T15:32:59.980118+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Retransmit timeout
2024-12-26T15:33:01.999327+00:00 OpenDTU-3480B8 OpenDTU[74b53816] All missing
2024-12-26T15:33:02.017187+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Nothing received, resend whole request
2024-12-26T15:33:03.998840+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Success
2024-12-26T15:33:04.004254+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Fetch inverter: 114184614039
2024-12-26T15:33:04.011386+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL inverter 114184614039]: actual limit is 26.2 % (210 W respectively), effective 5282 ms after update started, requested were 26.2 %
[...]
2024-12-26T15:33:19.304987+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::loop] consumption: 1079 W, target output: 200 W (limited to 200 W), solar inverters output: 0 W, battery allowance: 200 W, battery inverters output: 204 W
2024-12-26T15:33:19.335469+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Fetch inverter: 114184614039
2024-12-26T15:33:19.539007+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Last missing
2024-12-26T15:33:19.553625+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Request retransmit: 2
2024-12-26T15:33:19.646549+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Last missing
[... cut ...]
2024-12-26T15:33:24.663282+00:00 OpenDTU-3480B8 OpenDTU[74b53816] Success
2024-12-26T15:33:24.670742+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [DPL::announceStatus] waiting for sufficiently recent power meter reading
2024-12-26T15:33:27.733817+00:00 OpenDTU-3480B8 OpenDTU[74b53816] [PowerMeterHttpJson] New total: 1099.51
```

Im Monitoring sieht das so aus, morgens:
[image]

Abends:
[image]

Abends habe ich dann das Limit "Maximale Gesamtausgangsleistung" im DPL von 800W auf 200W runtergesetzt.
[image]
[image]

ranma, 2024-12-26:
For the time being I've downgraded to 2024.10.20 (however until the other day I was running an older self-compiled version with custom patches).

SW-Niko, 2024-12-27:
Hallo @ranma,
ich arbeite gerade an einer ganz anderen Sache aber ich ich kann so ein Verhalten ebenfalls beobachten.

Es schaut so aus als ob die Inverter-AC Mess Daten nicht rechtzeitig bereit stehen. Beispiel:

- Wir starten mit 81W AC 

- Ein neues Limit wird an die Inverter übertragen
12:20:12.194 > [DPL inverter 112182846421]: sending limit of 18.3 % (55 W respectively), max output is 300 W

- Die Regelung wartet bis die Inverter ok melden.
12:20:12.254 > [DPL] waiting for a start/stop/restart/limit command to complete

- Das Powermeter misst neue Werte.
12:20:12.312 > [PowerMeterHttpJson] New total: 37.16
12:20:13.278 > [PowerMeterHttpJson] New total: 60.00
12:20:14.244 > [DPL inverter 112182846421]: actual limit is 18.3 % (55 W respectively), effective 2024 ms after update started, requested were 18.3 %
Alles Ok. Ich kann erkennen das die Inverter das neue Limit umgesetzt haben.

- Aber wenn ich jetzt über die Inverter Stats die aktuelle Inverter AC Leistung einlese. (über getCurrentOutputAcWatts() )
Dann bekomme ich 78W zurück und nicht wie erwartet 55W.

Die weitere Regelung verwendet aber die 78W und kommt damit ins Schwingen.

ThomasCr, 2024-12-28:
Bei mir schaltet der Inverter mit der [2024.11.20](https://github.com/hoylabs/OpenDTU-OnBattery/releases/tag/2024.11.20) auch zu früh wieder an: Habe den DPL eingestellt dass er bei 80% SoC starten soll und bei 45% stoppt.

Aber er schaltet den Inverter schon 53% Ladung ein...

Genau so hatte ich ein merkwürdiges Phänomen festgestellt, dass er sich gar nicht anschaltet, wenn ich nicht manuell in der Konfig die ich exportiere den Wert von `solar_passthrough_enabled` auf `false` setze und wieder hochlade.

Das hatte ich hier schon geschrieben: https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1471

ThomasCr, 2024-12-28:
Bei mir schaltet der Inverter mit der [2024.11.20](https://github.com/hoylabs/OpenDTU-OnBattery/releases/tag/2024.11.20) auch zu früh wieder an: Habe den DPL eingestellt dass er bei 80% SoC starten soll und bei 45% stoppt. voltage_start_threshold

Aber er schaltet den Inverter schon 53% Ladung ein... und das eigentlich immer!

Genau so hatte ich ein merkwürdiges Phänomen festgestellt, dass er sich gar nicht anschaltet, wenn ich nicht manuell in der Konfig die ich exportiere den Wert von `solar_passthrough_enabled` auf `false` setze und wieder hochlade.

Das hatte ich hier schon geschrieben: https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1471

Das sind die powerlimiter-Werte aus dem Konfig Dump:

```
    "powerlimiter": {
        "enabled": true,
        "verbose_logging": true,
        "solar_passthrough_enabled": false,
        "solar_passthrough_losses": 3,
        "battery_always_use_at_night": true,
        "target_power_consumption": 50,
        "target_power_consumption_hysteresis": 10,
        "base_load_limit": 100,
        "ignore_soc": false,
        "battery_soc_start_threshold": 80,
        "battery_soc_stop_threshold": 45,
        "voltage_start_threshold": 50,
        "voltage_stop_threshold": 49,
        "voltage_load_correction_factor": 0.001,
        "full_solar_passthrough_soc": 100,
        "full_solar_passthrough_start_voltage": 66,
        "full_solar_passthrough_stop_voltage": 66,
        "inverter_serial_for_dc_voltage": "XXX191403XXX",
        "inverter_channel_id_for_dc_voltage": 0,
        "inverter_restart_hour": -1,
        "total_upper_power_limit": 800,
        "inverters": [
            {
                "serial": "XXX191403XXX",
                "is_governed": true,
                "is_behind_power_meter": true,
                "is_solar_powered": false,
                "use_overscaling_to_compensate_shading": false,
                "lower_power_limit": 10,
                "upper_power_limit": 1500
            }
        ]
    },
```

stefan123t, 2024-12-30:
>   12:20:14.244 > [DPL inverter 112182846421]: actual limit is 18.3 % (55 W respectively), effective 2024 ms after update started, requested were 18.3 %
>   Alles Ok. Ich kann erkennen das die Inverter das neue Limit umgesetzt haben.

Diese Info sollte die OpenDTU mit SystemConfigPara eingesammelt haben.

Jetzt müsste für alle Inverter ein RealTimeRunDebug laufen damit ihr auch die aktuellen Inverter Werte in de OpenDTU habt.

> * Aber wenn ich jetzt über die Inverter Stats die aktuelle Inverter AC Leistung einlese. (über getCurrentOutputAcWatts() )
>   Dann bekomme ich 78W zurück und nicht wie erwartet 55W.
> 
> Die weitere Regelung verwendet aber die 78W und kommt damit ins Schwingen.

Hier hat wohl der DPL loop die beiden o.g. Kommandos überholt und regelt bevor der neu akzeptierte Regelpunkt angefahren wurde. 

Dies hängt evtl. natürlich auch vom Delta und dem RampUp/RampDown Werten im GridProfile ab.

ranma, 2024-12-30:
Das eigentliche Problem ist ja das der Inverter gar nicht an sein sollte. Solarleistung ist zu klein und Battier-Ladestand ist zu niedrig => Inverter sollte aus bleiben. Trotzdem sehe ich in 2024.11.20 das eher morgens/abends mal gerne mit bis zu 800W ins Netz speist, auch wenn er nicht sollte.

SW-Niko, 2024-12-31:
@ranma 
Ja, sorry. Ich dachte Anfangs die beiden Probleme hängen zusammen. Aber da habe ich mich getäuscht.

@stefan123t 

>Dies hängt evtl. natürlich auch vom Delta und dem RampUp/RampDown Werten im GridProfile ab.

Richtig, aber in meinem Fall ist das nicht das Problem. Ich sehe ja an den Powermeter Messdaten das der Inverter das neue Limit eingestellt hat. Nur die Messdaten vom Inverter sind noch nicht auf aktuellen Stand.
Ich habe jetzt das Inverter Poll Intervall von 4 Sek auf 2 Sek reduziert.
[image]
Das scheint auf den ersten Blick besser zu laufen. 
Ich beobachte das mal eine Weile und wenn mir noch was negatives auffällt dann mach ich ein neues Issue auf.

ThomasCr, 2025-01-04:
Also bei mir ist der Poll Intervall des Stromzählers schon auf eine Sekunde.
Hilft es wenn ich ein Log bereitstelle und wenn ja, was muss aktiviert sein (ausführliche Protokollierung von welchen Funktionen), damit alles enthalten ist, was nötig ist?

AndreasBoehm, 2025-01-04:
@ranma @ThomasCr Könntet ihr nochmal mit einem build des aktuellen development branches testen? Es hat sich nochmal einiges getan im DPL code seit dem letzten release.

Builds findet ihr hier: https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/12599153511

AndreasBoehm, 2025-01-07:
Ich kann nun bestätigen das es auch mit der aktuellen development branch ein Problem gibt.

Ausgangs Situation:
- Batterie SoC unterhalb start threshold
- Limit wurde beachtet, WR war aus oder hat durch SolarPassthrough nur die Leistung genutzt die von den MPPTs kommt
- build: custom build basierend auf der development branch (ohne änderungen die den PowerLimiter betreffen)

Firmware update:
- Neues firmware build der development branch installiert
- Nach dem neustart der DTU wird jetzt auch Strom aus der Batterie entnommen obwohl start threshold nicht erreicht wurde
- Im log sehe ich `battery discharge allowed` -> das dürfte nicht sein

Log
```
10:34:14.194 > [DPL] battery interface enabled, SoC 40.3 % (used), age 0 s (valid)
10:34:14.202 > [DPL] BMS: 52.45 V, MPPT: 52.45 V, inverter 1164a00a6340: 52.50 
10:34:14.209 > [DPL] battery voltage 52.45 V, load-corrected voltage 52.81 V @ 274 W, factor 0.00130 1/A
```

**battery discharge allowed**
```
10:34:14.213 > [DPL] battery discharge allowed, start 52.50 V or 50 %, stop 52.00 V or 40 %
```

```
10:34:14.221 > [DPL] full solar-passthrough dormant, start 56.00 V or 55 %, stop 54.00 V
```

**start NOT reached**
```
10:34:14.223 > [DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night: no
```

```
10:34:14.226 > [DPL] total max AC power is 2300 W, conduction losses are 3 %
10:34:14.230 > [DPL] targeting 0 W, base load is 150 W, power meter reads 13.0 W (valid)
10:34:14.233 > [DPL] inverter 1164a00a6340 is behind power meter producing 274 W
10:34:14.236 > [DPL] requesting 287 W from 0 solar-powered inverters currently producing 0 W (diff 287 W, hysteresis 5 W)
10:34:14.240 > [DPL] battery allowance is 3937/3627 W DC/AC, solar power is 172/158 W DC/AC, requested are 287 W AC
10:34:14.244 > [DPL] requesting 287 W from 1 battery-powered inverter currently producing 274 W (diff 13 W, hysteresis 5 W)
10:34:14.247 > [DPL] will cover 297 W using 1 battery-powered inverter
10:34:14.251 > [DPL inverter 1164a00a6340]:
10:34:14.254 >     battery-powered, producing 274 W
10:34:14.257 >     lower/current/upper limit: 52/284/2000 W, output capability: 2000 W
10:34:14.260 >     sending commands enabled, reachable, eligible
10:34:14.262 >     max reduction production/standby: 222/274 W, max increase: 1716 W
10:34:14.274 >     target limit/output/state: 297 W (update)/297 W/production, 0 update timeouts
10:34:14.279 >     MPPTs AC power: a: 69 W b: 69 W c: 69 W d: 68 W
10:34:14.283 > [DPL inverter 1164a00a6340]: sending limit of 14.9 % (297 W respectively), max output is 2000 W
```

spcqike, 2025-01-07:
> Für mich sieht es so aus als ob wir beim start der DTU den state nicht korrekt ermitteln können.

war das nicht schon immer so? Woher soll die DTU wissen, ob wir grade am "Aufladen" oder am "Entladen" waren?
Allerdings sollte dieses initiale Problem auch nur ein mal auftreten. solange die DTU stabil läuft, oder?

AndreasBoehm, 2025-01-07:
Gute Frage.

Allerdings ist es seltsam das eine kombination aus diesen beiden Ausgaben möglich ist
`battery discharge allowed`
`start NOT reached, stop NOT reached`

SW-Niko, 2025-01-07:
> Für mich sieht es so aus als ob wir beim start der DTU den state nicht korrekt ermitteln können.

> war das nicht schon immer so? Woher soll die DTU wissen, ob wir grade am "Aufladen" oder am "Entladen" waren?
Allerdings sollte dieses initiale Problem auch nur ein mal auftreten. solange die DTU stabil läuft, oder?

Das hat @spcqike schon richtig beschrieben. Ein Threshold benötigt immer eine Vorgeschichte. Es ist ja wichtig über welche Schwelle man in den Bereich zwischen Stop und Start gekommen ist.
Beim Start-Up des Systems ist das unbekannt und man hat sich entschieden definiert immer mit "Aus" zu starten. Also der Threshold Bereich wurde über die Stop-Schwelle erreicht.

Vielleicht sollten wir uns den Startvorgang mal genauer an schauen. Ich hatte vor kurzen ein Problem, weil beim Start-Up die Daten vom Batterie Provider noch nicht da waren als gültig gekennzeichnet waren. Bei einem Threshold könne ein einziger falscher Wert gleich ein kippen bewirken.

schlimmchen, 2025-01-18:
@ranma Hast du auch Auswertungen über die Uptime deiner DTU mit der neuen Firmware und während das Problem bestand? Ist die ständig neu gestartet?

Tut mir leid, dass dieses Issue liegen geblieben ist. Allerdings ist in der Zwischenzeit schon wieder einiger passiert am DPL, und ich kann die Ausgaben nicht mehr zum aktuellen Code verknüpfen. Insb. `calcBatteryAllowance` wurde deutlich besser mit Ausgaben ausgestattet, was in deinem Fall ja genau das richtige ist. Bisher war es ja offenbar niemandem möglich, das Problem zu verstehen. Daher muss ich dir zumuten, es mit dem aktuellen Release noch einmal zu versuchen und neue Logs vom Fehlerfall, falls er denn wieder auftritt, zu liefern. Kannst du das bitte machen?

Nebenbemerkung: "start not reached, stop not reached, battery discharge allowed" ist sehr wohl gültig, auch mit "night time discharging" ausgeschaltet, nämlich wenn die Startschwelle erreicht wurde in der Vergangenheit und wir uns deshalb in einem Entladezyklus befinden. Ich vermute Niko hat genau das schon umschrieben?

ranma, 2025-02-05:
> [@ranma](https://github.com/ranma) Hast du auch Auswertungen über die Uptime deiner DTU mit der neuen Firmware und während das Problem bestand? Ist die ständig neu gestartet?

Bin momentan im Urlaub, kann leider erst gegen ende Februar wieder schauen.

closed 2025-03-05 by github-actions[bot]