hoylabs/OpenDTU-OnBattery #1488: [HTTP ERROR] Bad Request

denis-robel, 2024-12-29:
### What happened?

I want to make an OTA update from installed version [2024.09.11](https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/2024.09.11) to actual release 2024.11.20.

When I upload the firmware file (opendtu-onbattery-generic_esp32s3_usb.bin)
 I get always [HTTP ERROR] Bad Request and no update process is starting.

My hardware is:

Chip-Modell	ESP32-S3
Chip-Revision	0
Chip-Kerne	2
CPU-Frequenz	240 MHz
CPU-Temperatur	37,1 °C
Flash-Speichergröße	16.777.216 Byte (16 MB)

### To Reproduce Bug

don't know

### Expected Behavior

OTA update went fine

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.09.11

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
OTA Error

[HTTP ERROR] Bad Request
```

### Anything else?

-

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

spcqike, 2024-12-29:
Please check 
https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1486

schlimmchen, 2024-12-29:
See https://opendtu-onbattery.net/firmware/howto/upgrade_8mb/

Closing as invalid.

closed 2024-12-29 by schlimmchen

denis-robel, 2024-12-29:
> Please check #1486

thx a lot, it works. 
I was not sure if this is relevant for me because I started with fresh install of 2024.09.11 in September.

Happy new year

Denis

commit fa7641f7 (2023-11-16): Fix: Power not set to 0 when not reachable at EOD

If the inverter polling was disabled before it got unreachable the values where not set to zero
Fix: #1488