hoylabs/OpenDTU-OnBattery #1496: ESP32S3 + W5500 trennt LAN Verbindung

Tpaydt, 2025-01-02:
### What happened?

Ich benutze ein ESP32S3 + W5500 die LAN Verbindung funktioniert gut aber nach ein paar Tagen trennt sich die Verbindung und bleibt getrennt. Ziehe ich das LAN-Kabel raus verbindet sich die DTU sofort mit WLAN und arbeitet ganz normal weiter.
Hat jemand auch sowas feststellen können?

### To Reproduce Bug

W5500 Lan Verbindung trennt sich nach ein paar Tagen bis man diese wider manuell verbindet.

### Expected Behavior

Falls es an der Software liegt evtl eine Prüfung der LAN Verbindung einbauen.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.11.20

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Anything else?

Ich weiß nicht wie ich das Logen soll. Bin über einen Tipp dankbar.

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-01-02:
Weißt du ob das immer nach der gleichen Anzahl an Tagen passiert?

Hab noch ein POE Shield für mein Fusion Board hier liegen, dann nehme ich das jetzt als Anlass es endlich in Betrieb zu nehmen. Ich werde in ein paar Tagen berichten.

Tpaydt, 2025-01-02:
Hey,

ganz sicher bin ich mir nicht aber der Zeitraum kommt mir identisch vor.
Manchmal merke ich es nicht wenn die Verbindung getrennt wird, da ja aktuell kaum Sonne scheint gucke ich nicht so oft danach.

Danke dir für die Mühe.

AndreasBoehm, 2025-01-05:
Kannst du mir noch mehr über dein Setup erzählen? Welche Geräte hast du an der DTU angebunden und wie? 

Waren es Tage oder eher Wochen bis die DTU nicht mehr erreichbar war?

Tpaydt, 2025-01-05:
ESP32-S3 16MB + nRF24 + W5500 + OLED 2,4 Display SSD1309
W5500 kam als letztes dazu seitdem es integriert wurde.
Hoymiles HM600 mit Nulleinspeisung über Shelly3EM, keine Batterie.

Es sing Tage bis die Verbindung über LAN abbricht, mit WLAN kein Problem.

Die Firmware läuft auch stabil, den sobald ich das LAN-Kabel abziehe läuft die DTU übe WLAN weiter.

AndreasBoehm, 2025-01-07:
Danke für die Infos.

Meine DTU ist jetzt seit 4 Tagen und 6 Stunden über den W5500 online und es gab noch keine Aussetzer.

Hattest du wieder einen Aussetzer?

Tpaydt, 2025-01-08:
Hey

Also ich hatte leider das sich die DTU Neustart hatte dadurch hatte sich die LAN Verbindung automatisch neu verbunden.

CPU 0 Vbat Power on reset
Was hatte es zu bedeuten?

Aktuell läuft die DTU 1Tag und 22 Stunden, ich Check es so gut wie geht.

Danke schön mal für den Support 

Sobald der Fehler wieder kommt melde ich mich.

AndreasBoehm, 2025-01-08:
Das klingt nach einem Stromausfall bzw Unterspannung, ist das verwendet USB Netzteil eventuell zu schwach?

Tpaydt, 2025-01-08:
5V mit 2A

Tpaydt, 2025-01-09:
Jetzt hatte ich ein 
CPU 0 Software reset CPU   Fehler vor 30min.

LAN Verbindung ist immer noch aktiv.

AndreasBoehm, 2025-01-10:
Den "Fehler" sehe ich auch, aber das sagt nichts aus, bei mir kommt es daher das ich ein Update geflasht hab.

Kannst du die DTU an einen Computer hängen und permanent ein Log ziehen bis der Fehler wieder auftritt?
https://opendtu-onbattery.net/firmware/howto/serial_console/

Tpaydt, 2025-01-10:
Ja das lässt sich einrichten

Tpaydt, 2025-01-12:
Moin,

der Fehler war wieder da ich konnte es logen.

[putty.log](https://github.com/user-attachments/files/18389079/putty.log)

Zeile 35726 kommt es zur Unterbrechung der LAN Verbindung

Zeile 57194  habe ich das LAN Kabel abgezogen, danach ging es über WLAN weiter, es war 01:00 Uhr.

Firmware 10.01.2025

schlimmchen, 2025-01-12:
Danke fürs Log! Leider steht nichts drin... Sehr schade. Man sieht, dass die Netzwerkverbindung gestört ist, aber nichts zum Grund.

Wenn du das Kabel ziehst, verbindet sich OpenDTU-OnBattery per Wi-Fi. Und wenn du es dann wieder steckst, geht's dann über LAN weiter?

Bei mir gab es so einen Ausfall am W5500 auch noch nicht. Mein OpenDTU Fusion läuft oft unbeachtet sehr viele Tage einfach durch ohne Probleme. Auch hat sonst noch niemand von solchen Problemen berichtet, auch Upstream nicht. Daher bin ich, auch weil du schon von einem Power-On-Reset berichtet hast, recht schnell dabei, deiner Hardware die Schuld zu geben.

Dein Netzteil kann 2A liefern, das klingt nicht schlecht. Aber wie viel Spannung kommt denn an deinem Board an wenn gerade viel los ist? Sowohl das Netzteil als auch das Kabel können bei höheren Strömen plötzlich einen Spannungseinbruch verursachen, der problematisch ist. Das ist leider bei vielen Netzteilen und Kabeln so. Und welche Hardware hast du denn? Ein OpenDTU Fusion, oder etwas anderes? Meine Vorschläge sind: Anderes Netzteil versuchen, anderes Kabel versuchen (eins das "dick" zu sein scheint oder für schnelles Laden geeignet sein soll), 5V Stützkondensator und 3V3 Stützkondensator am Board verlöten.

Ich schließe nicht aus, dass die Software ne Macke haben könnte, aber wie geschildert würde ich dann erwarten, dass es schon mehr solcher Meldungen gibt...

Tpaydt, 2025-01-12:
Hardware ist
ESP32-S3 16MB + nRF24 + W5500 + OLED 2,4 Display SSD1309

Ja wenn ich das LAN Kabel wieder einstecke geht die Verbindung wieder über LAN

Netzteil: Samsung USB Handylader 2A 5V

Was für Netzteil benutzt ihr den so?

Ich habe noch ein anderes W5500 Board werde es mal tauschen.

Thema Spannungsabfall: W5500 und Display bekommen 5V direkt vom Netzteil nicht über das ESP.

Stützkondensator werde ich mir mal annehmen.

Frage ist wie viel Leute wirklich ein W5500 in Betrieb haben, also nicht das Fusion Board.

Danke erstmal für den Support.

SG-1980, 2025-02-06:
Hallo,

ich habe ein vergleichbares Problem.

Nach ein paar Tagen (5-20) ist das Gerät über das Lan nicht mehr erreichbar.
Ich sehe dass versucht wird sich ins WLan einzuwählen. Das scheitert an der Entfernung.
Die Logs liefern keinen Hinweis auf einen Fehler.
Die LEDs des W5500 blinken weiter.

Mir ist aufgefallen, dass der Fehler bei Versorgung über einen USB-Hub (Rechner) ca. 1 mal in 24h auftritt.
Bei Stromversorgung über ein USB Netzteil (5Port 65W) ca. 1 mal in 6h.

Aktuell setze ich einen Schaltregler 3,3V ein. Mit diesem ist es etliche Tage stabil aber nicht final.
Das scheint also auch nicht das finale Problem zu sein.

Reconnect des Lan-Kabels bringt bei mir nichts.

Ich habe bereits die SPI-Frequenz auf 10MHz / 8MHz geändert und auch ein paar zusätzliche Parameter angepasst.
Das scheint alles keinen Einfluss auf das Problem zu haben. Ebenso verschiedene W5500 und ESP32 ohne Änderung.

Meine aktuelle Vermutung aus einem unbekannten Grund (z.B. Spannung, SPI, Code) der W5500 aus dem Tackt kommt.
Ich überlege, wie man zum Testen am einfachsten den W5500 neu startet (reinitialisiert), wenn die Verbindung für XSekunden weg fällt.

Mein Problem: Ich habe einen zusätzlichen Service in der DTU eingebaut, damit das Balkonkraftwerk bei Notstromversorgung vom Stromnetz getrennt wird. Bei Lan-Ausfall geht daher das Balkonkraftwerk offline und ich brauche eine Leiter.

Wenn wer nen Tipp oder eine Idee hat.

Schon mal Danke vorab, tolles Projekt.

Tpaydt, 2025-02-06:
Ich konnte mein Problem auch noch nicht lösen, ein anderes W5500 Modul hat keine Besserung gebracht. Aktuell nutze ich das WLAN läuft seit 10 Tagen durch.
@SG-1980 
Wie versorgst du dein W5500 mit Spannung? Über das ESP32?

Tpaydt, 2025-02-06:
Ich versorge auch alles direkt mit 5V 2A mit Netzteil.
Ich habe diese Board.

[image]

SG-1980, 2025-02-06:
Davon hatte ich noch keins.
Bei mir ist es ein 38er board, habe davon diverse im Einsatz auch mit W5500. Bisher alles stabil, teils seit Jahren.

Kannst du eventuell noch die Reset und Interupt pins des W5500 schreiben um abzugleichen ob wir da auch gleich sind?
Müsste in der Hardware config auf der Gui stehenn

[image]

Tpaydt, 2025-02-06:
CS PIN 42
int PIN 44
miso PIN 41
mosi PIN 40
rst PIN 43
sclk PIN 39

SG-1980, 2025-02-06:
OK. Danke, dann sollte es an den Pins auch nicht liegen

SG-1980, 2025-02-09:
Hallo,

nach einigem Suchen, habe ich eine Vermutung wo ein Problem liegen könnte,
(Kann aber auch sein, dass ich im Code was nicht verstehe)

Im Code findet sich ein SPIManager.
Mittels diesem wird der SPI-Bus für Komponenten "zugewiesen" (SpiManager::alloc_device).
In meinem Fall für das Funkmodul (Cmt) und der W5500.

Über die Methode (get_shared_bus) wird anhand einer ID gesucht ob der Bus schon existiert. Wenn nicht wird ein neuer angelegt (wenn noch einer frei ist). Nach debug, habe ich den Eindruck, das für beide Module nur ein bus "erzeugt wird"
(verschiedene MOSI, MISO pins aber nur ein bus). Das Funkmodul und der W5500 teilen sich dann einen logischen SPI-Bus.

Anpassung: W5500::setup -> SpiManagerInst.alloc_device("",  geändert nach -> SpiManagerInst.alloc_device("ETH", 

Da es Tage dauern kann bis der Fehler auftritt, nicht sicher ob das die Lösung ist.

Da ich von der AVRStudio Seite komme ist für mich der SPI Bus was mit Registern und definierten Pins und ich war etwas überrascht, dass hier ein Bus (Object) mit zwei Sätzen von Pins (MISO, MOSI) funktioniert.
Daher wäre toll wenn wer prüfen könnte ob das oben geschriebene Sinn macht.

AndreasBoehm, 2025-02-10:
> ...das für beide Module nur ein bus "erzeugt wird" (verschiedene MOSI, MISO pins aber nur ein bus). Das Funkmodul und der W5500 teilen sich dann einen logischen SPI-Bus.

Das ist bewusst so implementiert worden, daher auch die Methode `get_shared_bus`. Wenn jedes SPI-Device einen eigenen Bus nutzt dann müssten wir uns stark einschränken, ein ESP32-S3 hat nur 2 SPI-Bus.

SPI-Devices:
- W5500 (shareable SPI)
- CMT (shareable SPI)
- NRF24 (non-sharable SPI)
- MCP2515 (z.B. Huawei CAN, non-shareable SPI)

Da die Kombination aus CMT, NRF24 und W5500 bei mir ohne Probleme läuft behaupte ich das es nicht an einem geteilten SPI-Bus liegen kann.

SG-1980, 2025-02-11:
Der aussage, kann nicht am SPI BUS liegen kann ich (leider) zustimmen. Ist Heute erneut ausgefallen, nach etwas mehr als 2 Tagen.

So langsam gehen mir die Ideen aus. Vrorschläge um das Thema einzugrenzen willkommen.

schlimmchen, 2025-02-12:
Hat hier jemand das OpenDTU Fusion mit dem PoE hat und hat dieses Problem?

SG-1980, 2025-02-18:
Habe das ganze jetzt auf eine Lochraster Platine gelötet.
Seit Inbetriebnahme stabil. Optimistisch das das so bleibt.

Allerdings hat der Wechselrichter seit Gestern Morgen keine Verbindung mehr zu DTU.
Auch die Nacht + Stromtrennung Wechselrichter und heute DTU haben das bisher nicht geändert.
Gibt aber schon einige Tickets und Einträge zu dem Thema (nur leider keine Lösung), daher nichts für diesen Thread.

closed 2025-03-20 by github-actions[bot]

SG-1980, 2025-03-20:
Wie anfangs geschrieben tritt der Fehler bei mir unregelmäßig auf.
Das System ist nun seit mehr als 20 Tagen ohne Ausfall im Einsatz (länger als jemals zuvor).

Daher hier was ich geändert habe (leider kein Beweis, dass dies das Problem ist aber vielleicht ein Hinweis für Andere).

W5500 läuft auf Bus2
CMT läuft auf Bus3

Die Änderungen wurden vorgenommen, da nach meinem Verständnis alle die einen stabilen Betrieb berichteten auf einem Board unterwegs waren wo es keine ISP 3 gibt. 

Hinweis zur Konfig (Parameter identisch mit der konfig beim AUftreten der Fehler):
- Der SPI zum W5500 läuft mit 10 MHz
- Pins des NRF24 für die W5500 Anbindung genutzt
- Stromversorgung 3,3V Schaltregler (nicht der Spannungsregler vom ESP)