hoylabs/OpenDTU-OnBattery #1499: VE.Direkt, Potentielle Probleme bei fragmentierten Daten (Multible Frames)

SW-Niko, 2025-01-05:
### What happened?

Beim testen von neuen Funktionen habe ich 2 potentielle Probleme im Victron Frame-Handler festgestellt. 

1. Fragmentierte Daten auf der VE-Direct Schnittstelle

Einige Victron Geräte senden einen Datensatz verteilt über mehrere Frames. Aktuell erhalten die Daten allerdings bei jedem Frame einen neuen gültigen Zeitstempel. Der Smart Shunt zum Beispiel sendet 2 Frames für einen Datensatz und damit erhält man auch zwei Zeitstempel für einen Datensatz anstatt einem.

Solange die Daten nur angezeigt werden ist das unkritisch, aber bei einer weiteren Verarbeitung des Datensatzes kann das zu Problemen führen.

Zum Beispiel:
- Beim Start-Up kann es vorkommen das die Daten einen gültigen Zeitstempel erhaltenen obwohl der Datensatz noch nicht komplett ist.
- Eine Auswertung der Daten auf Basis des mehrfachen Zeitstempels bei identischen Daten verschwendet Prozessorzeit.
- Fehlerhaft Berechnungen wenn Daten von unterschiedlichen Datensätzen verwendet werden.
- Schwingneigung bei zeitkritischen Regelungen.

2. Funktion isDataValid() beim Smart Shunt funktioniert nicht
Eine der Bedingungen für gültige Daten ist der Empfang der Seriennummer. Aber der Smart Shunt sendet keine Seriennummer.
[image]

### To Reproduce Bug

Schwierig. Über die aktuell möglichen Ausgaben nicht zu erkennen.  

### Expected Behavior

1. Fragmentierte Daten auf der VE-Direct Schnittstelle
Ein Datensatz erhält nur einem Zeitstempel beim Erhalt des letzten Frames.

2. Funktion isDataValid() beim Smart Shunt funktioniert nicht
Die vorhandene Funktion isDataValid() sollte auch beim Smurt Shunt richtig funktionierten.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU-OnBattery?

d7e96e1

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Anything else?

Ich teste gerade einen möglichen Bugfix und mache einen PR wenn ich damit keine Probleme feststelle.

Beide Probleme sehe ich aktuell noch als unkritisch. Aber je mehr die Daten für Berechnungen und Regelungen verwendet werden desto größer wird die Gefahr.

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-01-06:
Punkt 2 ist ein Problem, klare Sache. Meiner Meinung nach ist allerdings das erfolgreiche Empfangen eines Datensatzes, also die Verifikation der Prüfsumme, der richtige Grund bei `isDataValid()` "wahr" zu liefern.

Zu Punkt 1: Ich verstehe das Problem, aber mir gefällt die Lösung überhaupt nicht. Victron schreibt nirgendwo dass es diese Art Fragmentierung überhaupt gibt (oder doch?) und sicherlich erst recht nicht, dass im "ersten" Frame immer die einen und im "zweiten" Frame immer die anderen Daten enthalten wären (oder doch?). Würde es dir (und ich behaupt es geht vor allem mal um die Erweiterungen, die du beisteuern möchtest) helfen, wenn alle Daten einen Zeitstempel (mit `millis()`-Auflösung) haben?

SW-Niko, 2025-01-07:
> Verifikation der Prüfsumme, der richtige Grund bei isDataValid() "wahr" zu liefern.

Klar, wenn Punkt 1 die richtig funktioniert dann reicht das auch.

> Victron schreibt nirgendwo dass es diese Art Fragmentierung überhaupt gibt (oder doch?)

Ja das ist richtig. In der Victron Dokumentation findet man nur folgenden Abschnitt.
[image]

"Multiple blocks are send containing different fields" Etwas mehrdeutig... Aber wenn man sich die Daten anschaut wird es klarer:

11:10:12.317 > [VE.Direct SmartShunt 15/16] Frame: 1
11:10:12.362 > [VE.Direct SmartShunt 15/16] Text Data 'PID' = '0XA389'
11:10:12.419 > [VE.Direct SmartShunt 15/16] Text Data 'V' = '26197'
11:10:12.468 > [VE.Direct SmartShunt 15/16] Text Data 'I' = '1517'
11:10:12.516 > [VE.Direct SmartShunt 15/16] Text Data 'P' = '40'
11:10:12.591 > [VE.Direct SmartShunt 15/16] Text Data 'CE' = '-60007'
11:10:12.638 > [VE.Direct SmartShunt 15/16] Text Data 'SOC' = '439'
11:10:12.712 > [VE.Direct SmartShunt 15/16] Text Data 'TTG' = '-1'
11:10:12.764 > [VE.Direct SmartShunt 15/16] Text Data 'ALARM' = 'OFF'
11:10:12.809 > [VE.Direct SmartShunt 15/16] Text Data 'AR' = '0'
11:10:12.856 > [VE.Direct SmartShunt 15/16] Text Data 'BMV' = 'SMARTSHUNT 500A/50MV'
11:10:12.903 > [VE.Direct SmartShunt 15/16] Text Data 'FW' = '0416'
11:10:12.949 > [VE.Direct SmartShunt 15/16] Text Data 'MON' = '0'

11:10:14.298 > [VE.Direct SmartShunt 15/16] Frame: 2
11:10:14.344 > [VE.Direct SmartShunt 15/16] Text Data 'H1' = '-91952'
11:10:14.392 > [VE.Direct SmartShunt 15/16] Text Data 'H2' = '-91952'
11:10:14.545 > [VE.Direct SmartShunt 15/16] Text Data 'H3' = '-69405'
11:10:14.594 > [VE.Direct SmartShunt 15/16] Text Data 'H4' = '2'
11:10:14.641 > [VE.Direct SmartShunt 15/16] Text Data 'H5' = '0'
11:10:14.687 > [VE.Direct SmartShunt 15/16] Text Data 'H6' = '-477500'
11:10:14.747 > [VE.Direct SmartShunt 15/16] Text Data 'H7' = '12005'
11:10:14.793 > [VE.Direct SmartShunt 15/16] Text Data 'H8' = '29330'
11:10:14.839 > [VE.Direct SmartShunt 15/16] Text Data 'H9' = '1885148'
11:10:14.889 > [VE.Direct SmartShunt 15/16] Text Data 'H10' = '4'
11:10:14.934 > [VE.Direct SmartShunt 15/16] Text Data 'H11' = '0'
11:10:14.981 > [VE.Direct SmartShunt 15/16] Text Data 'H12' = '0'
11:10:15.027 > [VE.Direct SmartShunt 15/16] Text Data 'H15' = '0'
11:10:15.075 > [VE.Direct SmartShunt 15/16] Text Data 'H16' = '0'
11:10:15.122 > [VE.Direct SmartShunt 15/16] Text Data 'H17' = '1254'
11:10:15.172 > [VE.Direct SmartShunt 15/16] Text Data 'H18' = '1356'

> und sicherlich erst recht nicht, dass im "ersten" Frame immer die einen und im "zweiten" Frame immer die anderen Daten enthalten wären (oder doch?).

Nein, dazu habe ich nichts gefunden. Ich kann nur sagen das es bei meinen Geräten so ist. Beim Smart Shunt sind in den beiden Frames immer die gleichen Daten und immer in der gleichen Reihenfolge. Meine Laderegler verhalten sich identisch senden aber immer nur einen Frame.

> Würde es dir (und ich behaupt es geht vor allem mal um die Erweiterungen, die du beisteuern möchtest) helfen, wenn alle Daten einen Zeitstempel (mit millis()-Auflösung) haben?

Ja das geht natürlich auch. Jedem Wert einen eigenen Zeitstempel zu geben halte ich aber nicht für zielführend. Damit verlagern wir die Auswertung nur auf die höheren Schichten.

Jetzt muss ich nur noch erklären warum ich den Zeitstempel auf den letzten Frame gesetzt habe.
Das mache ich damit alle zeitlich zusammengehörenden Daten den gleiche Zeitstempel haben. Wenn Victron es richtig umgesetzt hat dann sammeln sie zuerst alle Daten in einen Zwischenspeicher und senden dann die Daten über einen oder mehrere Frames.
Dieser zeitliche Zusammenhang  ist bei einigen Daten sehr wichtig, falls Victron wie auch wir, die Daten für Berechnungen und Regelungen verwendet.
Widerstand = Spannung / Strom. Aber das Ergebnis kann ziemlich falsch sein wenn die Messzeitpunkte von Spannung und Strom zeitlich um 1 Sekunde auseinander liegen!
Ok, das ist eine reine Vermutung, aber ich würde es so machen.

Was hältst du davon wenn wir immer dem Frame, der die V-Daten (Spannung) enthält als Zeitstempel verwenden? Also dem ersten Frame und nicht dem letzten Frame verwenden?
Damit haben alle zeitkritischen Daten den best möglichsten und einen einheitlichen Zeitstempel. Und bei den unkritischen Daten (Historische Daten) ist es zeitlich gesehen egal ob die Daten zum Frame davor oder danach gehören.
Das würde auch noch ein weiteres Problem lösen. Wenn die Anzahl der Frames stetig wechselt dann wird es schwierig den letzten Frame zu ermitteln. Das ist eine Schachstelle in meiner Lösung die mir auch nicht gefällt. Habe ich noch nicht gesehen, kann ich aber auch nicht ausschließen.

closed 2025-05-01 by SW-Niko