hoylabs/OpenDTU-OnBattery #1528: [Request] Zuordnung von Wechselrichtereingängen (Batterie vs. Modul)

Keuler272, 2025-01-13:
### Is your feature request related to a problem? Please describe.

Hallo, zunächst kurz mein Setup.

Dachanlage: 12 kWp

Balkonkraftwerk:
- 4* Trina 450Wp an Hoymiles HMS 1800-4T
   - 2 Module an Noah 2000 angeschlossen
   - 2 Module direkt am WR angeschlossen

- OpenDTU on Battery (2024.11.20)

Noah Einstellungen:
- Batterie zuerst (8:00-16:59)
- Verbraucher zuerst (17:00-7:59)

Problem:
Wenn die Dachanlage im Überschuss produziert (Netzbezug negativ), schaltet sich der Wechselrichter des Balkonkraftwerks aus. Somit können die beiden am Hoymiles direkt angeschlossenen Module nicht mehr in's Hausnetz einspeisen. Außerdem würden die beiden über die Batterie angeschlossenen Module, sobald die Batterie auf 100% aufgeladen ist, auch nicht mehr ihren Überschuss an das Hausnetz abgeben können.

Versuchte Lösung:
Habe in HA als Automation festgelegt, dass die DTU bei Sonnenaufgang in den DPL Modus "1" wechseln soll. Bei einem Neustart des Wechselrichters schaltet sich dieser aber sofort wieder ab. Nur die manuelle Deaktivierung (Haken entfernen) von DPL und ein manueller(!) Neustart der DTU löst das Problem.

### Describe the solution you'd like

Ist es möglich, dass die Eingänge des Wechselrichters zugeordnet werden können?

Beispielsweise Eingang 1 und 2 sind mit der Batterie verbunden ("von Batterie gespeist"), wohingegen Eingänge 3 und 4 direkt an PV Modulen angeschlossen sind ("von Solarmodul gespeist") und dementsprechend soll der WR bei Überschuss nicht abschalten, sodass über Eingang 3 und 4 weiter ins Hausnetz eingespeist werden kann?

schlimmchen, 2025-01-13:
Das Projekt unterstützt den "Mischbetrieb" von Wechselrichter ausdrücklich **nicht**. Aus der Doku:

[image]

Selbst wenn man diese Zuordnung implementieren würde, was einen hohen Aufwand darstellt, hilft das meiner Einschätzung nach nichts. Welches Limit soll der DPL denn am WR einstellen, wenn man weiß, dass die Hälfte der Eingänge Leistung liefern kann, die andere aber nicht. Schließlich unterstützen die WR es ihrerseits nicht, dass man ihnen sagt, aus welchen Eingängen sie Leistung beziehen dürfen.

Daher kann dein Wunsch, wenn ich alles richtig verstanden habe, nicht umgesetzt werden.

Warum kannst du den WR nicht insgesamt als "von Solarmodulen betrieben" einstellen? Diese Batterie wird ja vermutlich nur dann Leistung liefern, wenn sie hinreichend voll ist bzw. je nachdem, wie sie eingestellt ist. Das sollte doch pasen?

closed 2025-01-13 by schlimmchen

Keuler272, 2025-01-13:
Danke für die rasche Antwort. Da habe ich nicht genau genug in die Doku reingeschaut - entschuldige.

Die von dir vorgeschlagene Lösung passt grundsätzlich, sodass sich der Wechselrichter nicht mehr abschaltet. Allerdings habe ich im DPL als minimales Leistungslimit 150W eingestellt. Darauf wird dann leider auch im Modus "von Solarmodulen betrieben" das Leistungslimit des Wechselrichters begrenzt, sobald die zwei direkt angeschlossenen Module das erste mal am Tag in Summe mehr als 150W produzieren.

Wähle ich ein höheres minimales Leistungslimit (z.B. 400W), würde ich oft am Abend und v.a. Nachts natürlich deutlich mehr einspeisen, als im Hausnetz benötigt wird. Ist es möglich, das minimale Limit in Home Assistant per Automation zu steuern? Dann würde ich es herabsetzen, sobald die Batterie einspeist.

spcqike, 2025-01-13:
warum setzt du überhaupt ein so hohes minimales Limit?

hast du kein powermeter konfiguriert? der WR sollte nur so viel erzeugen, wie der Haushalt braucht.

schlimmchen, 2025-01-13:
> Da habe ich nicht genau genug in die Doku reingeschaut - entschuldige.

Nicht nötig. Mir reicht es völlig, wen du zu verstehen scheinst, dass dein Wunsch nicht erfüllt werden kann.

> sobald die zwei direkt angeschlossenen Module das erste mal am Tag in Summe mehr als 150W produzieren.

Was meinst du damit? Wenn dein WR als "solarbetrieben" eingestellt ist im DPL, und das minimale Leistungslimit beträgt 150W, und es wird weniger als 150W im Haushalt gebraucht, dann bleibt der WR angeschaltet mit einem Limit von 150W und was die Module liefern, geht auch ins Haus. Das ist anders als bei WR, die als "batteriebetrieben" eingestellt sind. Die werden unterhalb des minimalen Leistungslimits in den Standby versetzt. Nicht aber, wenn der WR "solarbetrieben" ist.

Kommen wir damit einen Schritt weiter?

Keuler272, 2025-01-13:
> warum setzt du überhaupt ein so hohes minimales Limit?
> 
> hast du kein powermeter konfiguriert? der WR sollte nur so viel erzeugen, wie der Haushalt braucht.

Doch, ich habe einen Lesekopf installiert und dieser liefert auch Daten in die DTU. Dachte es wäre positiv für die Stabilität des Betriebs, das Limit nicht zu niedrig zu wählen. Das Minimum meines nächtlichen Strombedarfs liegt zudem zwischen 170 und 200W.

> und was die Module liefern, geht auch ins Haus

das stimmt, allerdings nicht mehr als die z.B. eingestellten 150W, obwohl beide Module gerade in dem Moment in Summe z.B 400W erzeugen würden. Sie werden somit vom Wechselrichter beschnitten. Außer ich setze das Limit hoch.

> Kommen wir damit einen Schritt weiter?

Ich versuche nochmal mein Wunschszenario darzulegen:
- Annahme: blauer Himmel im Mai, später Vormittag
- Wunschzustand:
  - Batterie wird mit der Leistung zweier Module geladen (~800W)
  - Die beiden restlichen Module speisen parallel dazu 800W in's Hausnetz bzw. da diese Leistung gerade nicht benötigt wird, in's Stromnetz ein

- Ist-Zustand:
  - Batterie wird mit der Leistung zweier Module geladen (~800W) --> passt
  - Die beiden restlichen Module speisen parallel dazu nur 150W in's Hausnetz bzw. in's Stromnetz, da der Wechselrichter aufgrund des Limits nicht mehr durchlässt.

Ich muss dazu sagen, dass der Strom meines Balkonkraftwerks zusammen mit dem der PV Dachanlage vergütet wird. Dementsprechend bin ich darauf aus, möglichst auch den Strom, der durch direkt am Wechselrichter hängenden Module produziert wird, in's Netz einzuspeisen.

Macht es meinen Wunschgedanken klarer?

spcqike, 2025-01-13:
> Die beiden restlichen Module speisen parallel dazu nur 150W in's Hausnetz bzw. in's Stromnetz, da der Wechselrichter aufgrund des Limits nicht mehr durchlässt.

das sollte nicht so sein. wenn das Haus in dem Moment 400W braucht, solltest du da auch 400W liefern

> Dementsprechend bin ich darauf aus, möglichst auch den Strom, der durch direkt am Wechselrichter hängenden Module produziert wird, in's Netz einzuspeisen.

das wird in deinem Setup jedoch schwierig und ist durch die openDTU nicht abbildbar.

du könntest den unconditional DPL mode (2) setzen
https://opendtu-onbattery.net/firmware/mqtt_topics/#mode

Damit sollte tagsüber aller Solarstrom der Paneele (VOraussetzung: die Batterie gibt nichts raus) eingespeist werden. 

Abends musst du den DPL mode wieder auf normal (0) setzen, so dass der DPL die Einspeisung wieder passend zum Powermeter ausregelt.

spcqike, 2025-01-13:
> Das "Problem" ist die PV Dachanlage (12 kWp) [...]

ok, macht Sinn.

> Also zum Beispiel 800W, wenn diese eingestellt sind.

Macht auch Sinn, da wir ja, im Falle eines 800VA BKW, eben nicht mehr als das einspeisen wollen/dürfen.

Wenn der unconditional Mode 2 für dich hier so passt, ist ja gut :) ja mit HA solltest du das gut automatisiert steuern können. nein im WebUI sieht man den Modus nicht. höchstens im Konsolen Log. Aber auch da bin ich mir grad nicht sicher.

Keuler272, 2025-01-13:
Bisher scheint es zu klappen. 17 Uhr wurde automatisch von DPL Mode 2 auf 0 gestellt und die Batterie speiste brav gemäß Hausverbrauch ein. 

Danke für die super Unterstützung! 

Lediglich an den Parametern muss ich wohl noch etwas feilen. Hab das Gefühl, dass der Hoymiles manchmal nicht ganz nachkommt (bei schnell wechselnden Verbrauchswerten z.B. wenn der Fernseher läuft). Hysterese liegt momentan bei 15W. Passt das?

Der Lesekopf liefert etwa jede Sekunde einen neuen Wert für den Hausverbrauch, der Wechselrichter sendet alle 5 Sekunden und das ziemlich treffsicher. Jedoch gibt es einige Wiederanforderungen vom gesendeten Fragmenten. 

[image]

Ist es empfehlenswert das Sendeintervall des WR von 5 Sekunden zu reduzieren und muss ich ggfs. die Sendestärke der DTU noch erhöhen, damit es weniger Wiederanforderungen gibt?

schlimmchen, 2025-01-16:
> Ist es empfehlenswert das Sendeintervall des WR von 5 Sekunden zu reduzieren

Ja, stell pauschal auf 1 Sekunde -- meine Meinung.

> muss ich ggfs. die Sendestärke der DTU noch erhöhen, damit es weniger Wiederanforderungen gibt?

Vermutlich eher erniedrigen. Wir haben schon viel diskutiert darüber, dass die DTU oft zu nahe an den WR ist. Jedenfalls gibt es noch Optimierungspotential.

Keuler272, 2025-01-29:
Hallo, mittlerweile läuft es recht gut mit dem DPL Mode 2 und 0 im Wechsel.

Was mir aber noch aufgefallen ist: mein Wechselrichter ist - nicht nur, aber vor allem während Phasen mit häufigen Fluktuation im Stromverbrauch (z.B. wenn der Fernseher läuft), manchmal ein paar Sekunden nicht erreichbar. So wie ich es sehe, schaltet er nicht komplett ab und macht auch keinen Neustart, aber in der DTU ist die Leistung annähernd 0.

Ist das ein eher Kommunikationsproblem und ich sollte an der Signalstärke drehen - oder ein Steuerungsproblem und ich sollte an Hysterese etc. Änderungen durchführen?

[image]

Keuler272, 2025-01-30:
Ebenso erreicht die DTU den Wechselrichter nicht (Not reachable), wenn aus einem konstanten Grundverbrauch raus plötzlich viel Leistung benötigt wird (in unterem Beispiel springt die Zisternenpumpe mir 700W an).

[image]

Ist das ein bekanntes Verhalten oder kann ich irgendwo optimieren, sodass die Kommunikation zwischen DTU und Wechselrichter aufrecht erhalten wird?

spcqike, 2025-01-31:
> So wie ich es sehe, schaltet er nicht komplett ab und macht auch keinen Neustart, aber in der DTU ist die Leistung annähernd 0.

kann ich nicht nachvollziehen. der WR sollte nicht einfach abschalten oder drosseln, außer er bekommt den Befehl dazu. Wenn er nicht erreichbar ist, läuft er weiter wie gehabt, auf dem zuletzt übermittelten Limit.

was sagt denn der Log in der Zeit? 

> Ist das ein bekanntes Verhalten oder kann ich irgendwo optimieren, sodass die Kommunikation zwischen DTU und Wechselrichter aufrecht erhalten wird?

bei meinem HMS-2000 kommt das gelegentlich vor. allerdings löst sich das Problem bei mir nicht von alleine. Ich muss dann die "DTU" Einstellungen erneut speichern.

Hintergrund bei mir: Ich hatte massive Probleme auf der Standardfrequenz. Daher habe ich eine andere Frequenz eingestellt. Wenn die DTU neustartet, "vergisst" sie die geänderte Frequenz. in der Config steht sie zwar noch drin, aber sie wird scheinbar nicht verwendet. 
bis zu einer gewissen Last / Limit am HMS ist das auch kein Problem, aber ab einem höheren Limit klappt es dann nicht. bzw nur, nachdem ich die "DTU" Einstellungen neu speichere und der CMT mit der eigentlich eingestellten Frequenz arbeitet.

das sieht bei mir dann ungefähr so aus:

[image]

über ~150W gibt es viele Probleme und Aussetzer. drunter gar keine. 10:54 hab ich die "DTU" Einstellung erneut gespeichert, ab da lief es wieder sauber, auch bei höheren Lasten.

Ob das zu deinem Bild passt, kann ich nicht sagen. könnte aber ähnlich sein.

Keuler272, 2025-02-01:
Danke für die ausführliche Antwort. Interessantes Verhalten bei dir mit der Frequenz. Heute hatte ich auffällig viele "Empfang Fehler: nichts empfangen". Dabei habe ich an der DTU nichts verändert - vielleicht liegt es ja auch an der Sendefrequenz.

Wie komme ich denn an die log files?

Ein weiteres Phänomen hatte ich ebenfalls ein paar mal die letzten Tage. Aus einer "perfekten" Nulleinspeisung heraus stellt die DTU plötzlich einen Zielwert(?) von 100W ein. Auf eine Regelung zu 0W komme ich nur, wenn ich dem Wechselrichter kurzzeitig ein anders Limit (z.B. 400W) gebe. Hat jemand Ähnliches erfahren?

[image]

Keuler272, 2025-02-05:
Heute wieder das Verhalten: die DTU stellt plötzlich die im DPL eingestellte Grundlast ein (hier:200W) und regelt nicht mehr. Eine dynamische Regelung funktioniert erst wieder, wenn ich dem Wechselrichter kurzfristig manuell einen neuen Zielwert mittels der DTU gebe.

Der Log spricht folgendes. Kann mir hier jemand weiterhelfen?

18:29:41.682 > [DPL::loop] ******************* ENTER **********************
18:29:41.709 > [DPL::calcHouseholdConsumption] target consumption: 0 W, base load: 200 W
18:29:41.735 > [DPL::calcHouseholdConsumption] power meter value: 81.0 W, power meter valid: yes
18:29:41.767 > [DPL::calcHouseholdConsumption] inverter 116495106239 is behind power meter producing 199 W
18:29:41.792 > [DPL::updateInverterLimits] requested: 280 W, producing: 199 W using 1 solar-powered inverters, diff: 81 W, hysteresis: 10 W
18:29:41.820 > [DPL::updateInverterLimits] will cover 199 W using solar-powered inverters
18:29:41.860 > [DPL inverter 116495106239]:
18:29:41.900 >     solar-powered, producing 199 W
18:29:41.923 >     lower/current/upper limit: 10/225/1800 W, output capability: 1800 W
18:29:41.949 >     sending commands enabled, reachable, eligible
18:29:41.976 >     max reduction production/standby: 189/189 W, max increase: 0 W
18:29:42.004 >     target limit/output/state: -1 W (unchanged)/199 W/unchanged, 0 update timeouts
18:29:42.031 > [DPL::calcBatteryAllowance] power requested: 81 W
18:29:42.057 > [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (0 W): 0 W
18:29:42.082 > [DPL::updateInverterLimits] requested: 0 W, producing: 0 W using 0 battery-powered inverters, diff: 0 W, hysteresis: 10 W
18:29:42.107 > [DPL::loop] consumption: 280 W, target output: 280 W (limited to 1600 W), solar inverters output: 199 W, battery allowance: 0 W, battery inverters output: 0 W
18:29:42.148 > [PowerMeterHttpJson] New total: 81.00
18:29:42.188 > Fetch inverter: 116495106239
18:29:42.216 > TX RealTimeRunData 865.00 MHz --> 15 95 10 62 39 80 16 07 60 80 0B 00 67 A3 A0 05 00 00 00 00 00 00 00 00 15 E0 25 
18:29:42.256 > Interrupt received
18:29:42.283 > RX 865.00 MHz --> 95 95 10 62 39 80 16 07 60 01 00 01 01 DC 01 DC 00 70 00 6E 02 13 02 0B 00 01 BD | -63 dBm
18:29:42.324 > Interrupt received
18:29:42.356 > RX 865.00 MHz --> 95 95 10 62 39 80 16 07 60 02 26 94 00 01 2A 09 00 8B 00 8C 01 DC 01 DD 00 6E 40 | -63 dBm
18:29:42.399 > Interrupt received
18:29:42.426 > RX 865.00 MHz --> 95 95 10 62 39 80 16 07 60 03 00 6E 02 0A 02 0E 00 01 1B AB 00 01 1C 7A 00 8F 8A | -63 dBm
18:29:42.469 > Interrupt received
18:29:42.497 > RX 865.00 MHz --> 95 95 10 62 39 80 16 07 60 04 00 91 08 CD 13 86 07 CE 00 00 00 58 03 E8 00 79 7C | -63 dBm
18:29:42.545 > Interrupt received
18:29:42.576 > RX 865.00 MHz --> 95 95 10 62 39 80 16 07 60 85 00 07 7B C8 8B | -63 dBm
18:29:42.618 > RX Period End
18:29:42.657 > Success
18:29:42.713 > [DPL::loop] ******************* ENTER **********************
18:29:42.762 > [DPL::calcHouseholdConsumption] target consumption: 0 W, base load: 200 W
18:29:42.789 > [DPL::calcHouseholdConsumption] power meter value: 81.0 W, power meter valid: yes
18:29:42.828 > [DPL::calcHouseholdConsumption] inverter 116495106239 is behind power meter producing 199 W
18:29:42.855 > [DPL::updateInverterLimits] requested: 280 W, producing: 199 W using 1 solar-powered inverters, diff: 81 W, hysteresis: 10 W
18:29:42.905 > [DPL::updateInverterLimits] will cover 199 W using solar-powered inverters
18:29:42.953 > [DPL inverter 116495106239]:
18:29:42.993 >     solar-powered, producing 199 W
18:29:43.019 >     lower/current/upper limit: 10/225/1800 W, output capability: 1800 W
18:29:43.046 >     sending commands enabled, reachable, eligible
18:29:43.072 >     max reduction production/standby: 189/189 W, max increase: 0 W
18:29:43.098 >     target limit/output/state: -1 W (unchanged)/199 W/unchanged, 0 update timeouts
18:29:43.128 > [DPL::calcBatteryAllowance] power requested: 81 W
18:29:43.177 > [DPL::calcBatteryAllowance] limited by battery (0 W) and/or solar power (0 W): 0 W
18:29:43.205 > [DPL::updateInverterLimits] requested: 0 W, producing: 0 W using 0 battery-powered inverters, diff: 0 W, hysteresis: 10 W
18:29:43.230 > [DPL::loop] consumption: 280 W, target output: 280 W (limited to 1600 W), solar inverters output: 199 W, battery allowance: 0 W, battery inverters output: 0 W
18:29:43.257 > [PowerMeterHttpJson] New total: 82.00
18:29:43.289 > Fetch inverter: 116495106239
18:29:43.328 > TX RealTimeRunData 865.00 MHz --> 15 95 10 62 39 80 16 07 60 80 0B 00 67 A3 A0 06 00 00 00 00 00 00 00 00 E5 F4 C2 
18:29:43.368 > Interrupt received
18:29:43.407 > RX 865.00 MHz --> 95 95 10 62 39 80 16 07 60 01 00 01 01 DC 01 DC 00 70 00 6E 02 13 02 0B 00 01 BD | -63 dBm
18:29:43.452 > Interrupt received
18:29:43.478 > RX 865.00 MHz --> 95 95 10 62 39 80 16 07 60 02 26 94 00 01 2A 09 00 8B 00 8C 01 DC 01 DD 00 6E 40 | -64 dBm
18:29:43.529 > Interrupt received
18:29:43.558 > RX 865.00 MHz --> 95 95 10 62 39 80 16 07 60 03 00 6E 02 0A 02 0E 00 01 1B AB 00 01 1C 7A 00 8F 8A | -63 dBm
18:29:43.601 > Interrupt received
18:29:43.630 > RX 865.00 MHz --> 95 95 10 62 39 80 16 07 60 04 00 91 08 CD 13 86 07 CE 00 00 00 58 03 E8 00 79 7C | -64 dBm
18:29:43.739 > Interrupt received
18:29:43.768 > RX 865.00 MHz --> 95 95 10 62 39 80 16 07 60 85 00 07 7B C8 8B | -63 dBm
18:29:43.816 > RX Period End
18:29:43.861 > Success
18:29:43.900 > [PowerMeterHttpJson] New total: 82.00

schlimmchen, 2025-02-05:
Hier gehts weiter: #1606. Bitte die dort bereitgestellte/verlinkte Firmware einspielen und testen.

Keuler272, 2025-02-06:
Danke, hab aktualisiert und werde die nächsten Tage fleißig testen.