hoylabs/OpenDTU-OnBattery #1530: [DPL] with only the base load configured the limit is constantly changing

AndreasBoehm, 2025-01-14:
### What happened?

The limit is constantly changed for a system with a configured base load and no powermeter. I know that this is not a meaningful real life scenario but it shows that something is weird in the DPL.

With a configured base load of 800W or 1600W the limit is set to 120W on every second DPL loop.

### To Reproduce Bug

- setup DTU without battery and without powermeter
- enable DPL with a base load (tried 800W and 1600W, both gave the same result)

### Expected Behavior

- inverter limit matches the base load without changing to 120W every second DPL loop

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.01.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
Success
09:30:00.233 > Fetch inverter: 116494200313
09:30:00.276 > [DPL inverter 116494200313]: actual limit is 7.5 % (120 W respectively), effective 2071 ms after update started, requested were 7.5 %
09:30:00.315 > [DPL] waiting for sufficiently recent inverter data
09:30:00.717 > Success
09:30:00.755 > [DPL] targeting 0 W, base load is 1600 W, power meter reads 0.0 W (stale)
09:30:00.797 > [DPL] requesting 1600 W from 1 solar-powered inverter currently producing 72 W (diff 1528 W, hysteresis 10 W)
09:30:00.835 > [DPL] will cover 1600 W using 1 solar-powered inverter
09:30:00.882 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
09:30:00.917 > [DPL] requesting 0 W from 0 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 10 W)
09:30:00.955 > [DPL inverter 116494200313]:
09:30:00.997 >     solar-powered, producing 72 W
09:30:01.037 >     lower/current/upper limit: 10/120/1600 W, output capability: 1600 W
09:30:01.075 >     sending commands enabled, reachable, eligible
09:30:01.115 >     max reduction production/standby: 62/62 W, max increase: 1528 W
09:30:01.156 >     target limit/output/state: 1600 W (update)/1600 W/production, 0 update timeouts
09:30:01.195 >     MPPTs AC power: a: 19 W b: 19 W c: 18 W d: 18 W
09:30:01.235 > [DPL inverter 116494200313]: sending limit of 100.0 % (1600 W respectively), max output is 1600 W
09:30:01.277 > [DPL] waiting for a start/stop/restart/limit command to complete
09:30:02.757 > Success
09:30:02.795 > Fetch inverter: 116494200313
09:30:02.835 > [DPL inverter 116494200313]: actual limit is 100.0 % (1600 W respectively), effective 2038 ms after update started, requested were 100.0 %
09:30:02.875 > [DPL] waiting for sufficiently recent inverter data
09:30:03.317 > Success
09:30:03.355 > [DPL] targeting 0 W, base load is 1600 W, power meter reads 0.0 W (stale)
09:30:03.395 > [DPL] requesting 1600 W from 1 solar-powered inverter currently producing 72 W (diff 1528 W, hysteresis 10 W)
09:30:03.437 > [DPL] will cover 120 W using 1 solar-powered inverter
09:30:03.475 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 1480 W AC
09:30:03.515 > [DPL] requesting 0 W from 0 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 10 W)
09:30:03.555 > [DPL inverter 116494200313]:
09:30:03.595 >     solar-powered, producing 72 W
09:30:03.635 >     lower/current/upper limit: 10/1600/1600 W, output capability: 1600 W
09:30:03.677 >     sending commands enabled, reachable, eligible
09:30:03.715 >     max reduction production/standby: 62/62 W, max increase: 48 W
09:30:03.756 >     target limit/output/state: 120 W (update)/120 W/production, 0 update timeouts
09:30:03.795 >     MPPTs AC power: a: 18 W b: 19 W c: 18 W d: 18 W
09:30:03.835 > [DPL inverter 116494200313]: sending limit of 7.5 % (120 W respectively), max output is 1600 W
09:30:03.875 > [DPL] waiting for a start/stop/restart/limit command to complete
09:30:05.456 > Success
09:30:05.461 > Fetch inverter: 116494200313
[... 76 lines cut ...]
09:30:15.836 > [DPL inverter 116494200313]: actual limit is 7.5 % (120 W respectively), effective 2038 ms after update started, requested were 7.5 %
09:30:15.875 > [DPL] waiting for sufficiently recent inverter data
09:30:16.276 > Success
09:30:16.316 > [DPL] targeting 0 W, base load is 1600 W, power meter reads 0.0 W (stale)
09:30:16.357 > [DPL] requesting 1600 W from 1 solar-powered inverter currently producing 72 W (diff 1528 W, hysteresis 10 W)
09:30:16.395 > [DPL] will cover 1600 W using 1 solar-powered inverter
09:30:16.436 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
09:30:16.477 > [DPL] requesting 0 W from 0 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 10 W)
09:30:16.516 > [DPL inverter 116494200313]:
09:30:16.557 >     solar-powered, producing 72 W
09:30:16.598 >     lower/current/upper limit: 10/120/1600 W, output capability: 1600 W
09:30:16.635 >     sending commands enabled, reachable, eligible
09:30:16.675 >     max reduction production/standby: 62/62 W, max increase: 1528 W
09:30:16.717 >     target limit/output/state: 1600 W (update)/1600 W/production, 0 update timeouts
09:30:16.756 >     MPPTs AC power: a: 19 W b: 19 W c: 18 W d: 18 W
09:30:16.795 > [DPL inverter 116494200313]: sending limit of 100.0 % (1600 W respectively), max output is 1600 W
09:30:16.837 > [DPL] waiting for a start/stop/restart/limit command to complete
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-01-14:
I will retest this with a build from https://github.com/hoylabs/OpenDTU-OnBattery/pull/1529 later today.

schlimmchen, 2025-01-15:
> I know that this is not a meaningful real life scenario

Let me clarify that we actually do explicitly support this. It's for users with simple setups, maybe small batteries, who want to produce a fixed amount of AC power, often only during the night or when the battery is full.

closed 2025-01-16 by schlimmchen