hoylabs/OpenDTU-OnBattery #1531: [Request] Solar-Charger Values also possible via Mqtt.

Kroki0815, 2025-01-14:
### Describe the solution you'd like

I have all data for battery, charger & grid-measurement in my mqtt-broker. I can add the grid-power & the battery-values from mqtt, but the values for the solar-shargers can't be assigend via mqtt. I think it will help many more people to integrate their chargers in your project.

Kroki0815, 2025-01-14:
Sorry, hab grad gesehen, das es schon mehrfach Requested wurde.

closed 2025-01-14 by Kroki0815

AndreasBoehm, 2025-01-14:
Kommt bald. Danke das du den issue wieder zugemacht hast :)

AndreasBoehm, 2025-01-15:
Duplicate of https://github.com/hoylabs/OpenDTU-OnBattery/issues/457