hoylabs/OpenDTU-OnBattery #1535: Inverter Tagesertrag wird zurück gesetzt wenn der Inverter offline geht

Saarblick, 2025-01-14:
### What happened?

Ich habe derzeit zwei Inverter in OnBattery 2025.01.10
Beim HM600 wird der Tagesertrag erst Nachts nach 12 Uhr gelöscht.
Beim HMT2250 wird der Tagesertrag auf 0 gesetzt nach Sonnenuntergang zurück gesetzt nachdem der HMT2250 abgeschaltet hat.
Einstellungen für beide Umrichter sind in der DTU gleich.
Ich arbeite mit einem Fusion Board mit Can und POE shield.
Beide Umrichter sind nur Solarbetrieben ohne Batterie.

[image]
[image]
[image]
[image]

### To Reproduce Bug

Jeden Tag das Problem nach Sonnenuntergang

### Expected Behavior

??

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

v4.4.7-dirty 2025.01.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Anything else?

Erste mal schon aufgefallen als an dem HM600 im Herbst auch noch ein Akku angeklemmt war.
Deswegen wurde das Problem nicht gemeldet.
Der akku ist aber über Winter abgeklemmt und steht im Keller!
Issue war auch in der letzten Version von 2024 auch schon.

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Saarblick, 2025-01-14:
[image]

Saarblick, 2025-01-14:
Hier die DPL Einstellungen.
Vorsicht! Der HM600-Neu ist derzeit gar nicht angeklemmt/in Betrieb!
Zur Info, ich habe im Winter keinerlei direkte Sonneneinstrahlung auf den Kollektoren.
Vernachlässige deshalb die DPL  Einstellungen derzeit kpl.

[image]
[image]

Saarblick, 2025-01-18:
Ich beobachte das Verhalten jeden Tag mehrmals.
Ein einziges mal wurde in dieser Woche der Tagesertrag des HMT nicht bei Sonnenuntergang zurückgesetzt.
Könnte das Problem es mit dem wiederholten Versuch zu starten des WRs bei Sonnenuntergang oder dem Zeitabgleich zusammenhängen?

schlimmchen, 2025-01-19:
Und du hast die gleichen erweiterten Einstellungen für alle WR (insb. Tagesertragkorrektur)?

Es wird aber nicht so dunkel, dass der Inverter neu startet? Falls das passiert, und der WR kommt dann wieder "online", dann wird auch dessen gemeldeter Ertrag übernommen. Aber dafür ist die Tagesertragskorrektur da... Ist die eingeschaltet für alle Inverter?

Saarblick, 2025-01-20:
Ich hatte gestern gedacht es wäre mein Fehler, hatte tatsächlich was unterschiedlich eingestellt..
Heute die Ernüchterung hat sich nicht verändert:

spcqike, 2025-01-20:
Warum misst der eine WR 402V auf der Phase? Macht der das öfters?

Saarblick, 2025-01-20:
Der mit dem Problem ist ein dreiphasiger HMT2250.
Daher die 400V
Die 3 Phasen werden ja nicht richtig von der DTU ausgewiesen.
Ich schaue morgen bei Tageslicht wieder rein wieviel Volt angezeigt werden.

spcqike, 2025-01-20:
Ah, Macht Sinn.

AndreasBoehm, 2025-01-21:
Kann es sein das es sich um ein HMT spezifisches Problem handelt? Das müssten wir mal in der Hoymiles lib prüfen.

Saarblick, 2025-01-22:
Seltsam, gestern hatte der HMT sich nach sonnenuntergang irgendwann wieder genullt.
Heute hat er den Tagesertrag noch um 21.00 gehabt.

Saarblick, 2025-01-24:
Kann ich bei dem Thema noch irgendwas zur Problemsuche beitragen?
Gestern hat der HMT wieder den Tagesertrag behalten, heute ist er wieder (der von heute) auf 0 nach sonnenuntergang.

Saarblick, 2025-02-05:
Heute mal live zugeschaut als zurück gesetzt wurde..
Kann es sein das es mit dem Neustart zusammenhängt?
Einen Zeitabgleich macht der HMT ja im Gegensatz zu dem HM ja nie.
Bekomme ich irgendwie raus ob der Umrichter selbst überhaupt ne Zeit hat?

[image]

spcqike, 2025-02-05:
> Kann es sein das es mit dem Neustart zusammenhängt?

Natürlich. Startet der wechselrichter neu, setzt er sich zurück. Im „Normal Betrieb“: geht die Sonne unter, geht der WR aus. Kommt sie wieder, startet er wieder bei 0. 

Die „Tagesertragkorrektur“ der openDTU sollte das erkennen und berücksichtigen. das tut sie auch. Aber auch nur so lange die openDTU nicht ebenfalls Neustartet.

closed 2025-03-06 by github-actions[bot]