hoylabs/OpenDTU-OnBattery #1544: [Request] Huawei AC Charger: Set offline limits automatically

schlimmchen, 2025-01-16:
### Is your feature request related to a problem? Please describe.

See discussion in #1378.

### Describe the solution you'd like

Clean up the Huawei modal in the live view: Settings limits there will always update the "online" limit values. The offline limits are hardcoded to 48V (lowest permissible value, as far as I understand) and 0A, i.e., to shut off the charger in case the CAN connection breaks.

### Describe alternatives you've considered

If users actually want to set "no shutdown" values for the "offline" case, they should have the option to specify which offline settings to transmit to the charger. Only after user complain...

schlimmchen, 2025-01-18:
For the record: Setting an offline limit of 48V and 0A does work as expected (output becomes 0V and delivers 0A), even though the slot-detect pins still allow the PSU to run, with two caveats:
1. It takes quite some time for the PSU the reduce the current. Not sure why that is, yet. In my tests is takes ~30s from 2A to zero. Not sure how long it takes from 50A to zero...
2. The fan will blow at full (?) power, at least when the output approaches 0V/0A.

Also, the PSU will not resume the last online limit values, once it is "online" again (data is polled and the yellow LED in the middle stops flashing).

Andrix82, 2025-01-24:
Fully support the safety feature to disable the R4850 offline charging per default.

Could save literally someones house who is not aware/caring about that and using bad connections.

But pls. definitely keep the option to use the openDTUonBattery as possible offline configurator.

I already needed that for other applications/tests and will further need it.

Thanks!

schlimmchen, 2025-01-24:
Are you okay with settings in the AC charger web UI settings page that allow to specify the desired offline voltage and max current? OpenDTU-OnBattery would then take care to apply these settings automatically. However, the option to set the offline limits from withing the live view's modal will go away.

PR #1806 (2025-05-21): Feature: Improved support for R4850 and support for R48xx in general
This changeset improves upon the integration for R4850 power supplies by implementing new features/functions like

* reading board properties
* setting fan control modes
* shutdown over CAN
* processing ACKs from the PSU
* auto-setting offline limits (user-configurable)

and adds general support for (hopefully) all Huawei R48xx power supplies.

[image]

Closes #1544 (set (user-configurable) offline limits automatically)
Closes #1440 (support for R48xx in general, here: R4875)
Closes #435 (there is now a MQTT topic that allows [... cut]
closed 2025-05-21 by schlimmchen