hoylabs/OpenDTU-OnBattery #1551: DPL no longer works for me

synack666, 2025-01-17:
### What happened?

DPL no longer works for me since these changes:
Two things have changed yesterday, firstly I have updated to the latest version (DTU) and my WR (HMS 1600 4T) has been updated to 2.0.4 with 'Power Distribution Logic'. Before that it worked as desired.

[image]

It is always at 624 watts.

[image]

[image]

At the moment there is actually nothing to regulate because I hardly get over 100W yield but in the past it was always regulated according to consumption.

### To Reproduce Bug

Use the SW version with a WR that has PDL active

### Expected Behavior

the control should be analogous to the consumption (hysteresis included)

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

	2025.01.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Manos1966, 2025-01-17:
> my WR (HMS 1600 4T) has been updated to 2.0.4 with 'Power Distribution Logic'.

This was probably not a good idea?

The difficult part is that you changed TWO variables (DTU-Firmware and WR-Firmware) and now you wonder what happened.
Did you post a bug request to Hoymiles as well? (just kidding 😉 ).

You need to post the log in a text version, pictures do not help much.
Below you will find a description on how to store the log at PASTEBIN and post the link here:
[create-a-logfile-datei-with-console and PASTEBIN](https://github.com/hoylabs/OpenDTU-OnBattery/wiki/How-to-Search-for-Solutions%2C-Ask-Questions%2C-Report-Problems#41-create-a-logfile-datei-with-console)

Immediate step would be to re-flash your DTU with the previous Firmware (that you know, it was working), as a quick fix to ensure that it is not the WR-Firmware that caused the problem.

Another quick fix would be to check your DPL settings and make sure that the WR minimum limit is abve 60W (I think I saw some values of 22 - 25W at your log but it is difficult reading pictures).

Webranger2, 2025-01-17:
I have a HM-1500 and since updating to 2025.01.10 I have the same problem that DPL is no longer regulated.

If I go back to the previous version 2024.08.18 it works as before.

schlimmchen, 2025-01-17:
@synack666 You set your hysteresis to 100W (which occurs unreasonable to me -- why do you set the hysteresis so high?) but the inverter assumes it can only produce 48W more than it already does. So no new limit will ever be sent.

@AndreasBoehm fixed a regression which will be part of the next release. It should also fix your issue. Still, consider reducing your hysteresis setting.

synack666, 2025-01-17:
I'm going there on Monday, I'm on a short vacation right now.
The 100 watt hysteresis is so that the WR is not written to so often, I once read somewhere that the eprom has a limited service life. Sunday should be sunny and I'll record with Home Assistant what the DTU is doing, maybe everything is ok after all. It was different before but that doesn't mean it's wrong. Otherwise I'll fall back on the version for the DTU and see.

Manos1966, 2025-01-18:
> so that the WR is not written to so often, I once read somewhere that the eprom has a limited service life.

This does not apply in the case of OpenDTU-OnBattery.
The eprom has a limited service life on "permanent write" operations. For example when you give your WR to command to permanently change its limit (even after restart).
Instead, wWhen you give your WR the command to change Limit until the next restart, this is not written in the EPROM.

OpenDTU-OnBattery does NOT write permanent changes every time it changes the limit.

Home Assistant reports erroneusly that the change was PERMANENT.
It is not Home Assistant's fault, rather that the HOYMILES Firmware does not know anything else than reporting "permanent changes".

In summary: Do not worry about it!

synack666, 2025-01-20:
Morning :-)

Log:
https://pastebin.com/1wgSV1mm

[image]

Today it looks as if more or less power is being requested from the inverter, so it is being regulated, but the display is at 624 watts.
If I interpret the log correctly.

synack666, 2025-01-20:
@Manos1966 

Oh, ok, thanks!
The display in Home Assistant was actually the reason I was thinking about it.

synack666, 2025-01-20:
I took a closer look at it today, according to the log it is regulated (requests xxx from inverter), but the display does not change.

AndreasBoehm, 2025-01-23:
Solange die Panels nicht genug Leistung liefern gibt es für den DPL keinen Grund das Limit zu ändern.

```
08:56:24.413 > [DPL] requesting 1600 W from 1 solar-powered inverter currently producing 8 W (diff 1592 W, hysteresis 50 W)
08:56:24.461 > [DPL] will cover 8 W using 1 solar-powered inverter
08:56:24.506 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 1592 W AC
08:56:24.552 > [DPL] requesting 0 W from 0 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 50 W)
08:56:24.598 > [DPL inverter 1234567890]:
08:56:24.647 >     solar-powered, producing 8 W
08:56:24.691 >     lower/current/upper limit: 60/624/1600 W, output capability: 1600 W
08:56:24.737 >     sending commands enabled, reachable, eligible
08:56:24.782 >     max reduction production/standby: 0/0 W, max increase: 48 W
08:56:24.827 >     target limit/output/state: -1 W (unchanged)/8 W/unchanged, 0 update timeouts
08:56:24.886 >     MPPTs AC power: a: 4 W b: 2 W c: 1 W d: 2 W
```
Es wird versucht 1600 W durch deinen Solarbetriebenen WR abzudecken, der produziert aber nur 8 W bei einem Limit von 624 W -> der DPL ändert das limit nicht, weil es nichts bringt. 

Es funktioniert also erstmal alles wie es soll.

Ein Problem besteht nur wenn das Limit nicht hoch genug ist um den bedarf zu decken obwohl die Sonne voll auf die Panels scheint.
Oder wenn das Limit und somit die Produktion höher als der bedarf ist und somit viel zu viel eingespeist wird.

synack666, 2025-01-23:
DonJohnLong: Hab es geändert, dachte das ist nur wichtig wenn eine Batterie verbaut ist.

Ändert aber nichts.

AndreasBoehm: OK, das war vorher anders, da hat DPL das immer angepasst.
Mangels Sonne kann ich nicht sagen wie es ist wenn ausreichend Leistung von dem WR kommt.

Manos1966, 2025-01-23:
> AndreasBoehm: OK, das war vorher anders, da hat DPL das immer angepasst.

@AndreasBoehm das war richtig so vorher, oder? Seit welcher Firmware hat es sich geändert?

AndreasBoehm, 2025-01-23:
Kann sein das es vorher anders war, aber ich versteh nicht warum allen so wichtig ist das sich das Limit ändert, auch wenn es keinen unterschied macht 😂

AndreasBoehm, 2025-01-23:
Ich denke das der DPL sich seit https://github.com/hoylabs/OpenDTU-OnBattery/releases/tag/2024.11.20 anders verhält.

Harisjems, 2025-01-23:
Mir ist auch aufgefallen, dass sich der DPL-Limit (zumindest visuell) bei zu wenig Sonne seit dem 20.11.2024 nicht ändert.  Z. B. hält der DPL den WR bei 80 % oder weniger, es kommen aber nur 100 W bei 2 kWp raus, weil das Wetter schlecht ist.  Dann habe ich manuell den WR auf 100 % gesetzt und DPL deaktiviert, aber das ändert nichts an der AC-seitig abgegebenen Leistung. Wo nichts ist, kann nichts rauskommen. Es ist halt nur ein psychologischer Effekt, dass 100 % angezeigt werden müssen.

Der DPL ist davon unbeeindruckt; ich denke, er berechnet das. Manchmal ist Nichts tun am sinnvollsten.

synack666, 2025-01-24:
Moin!
OK, damit kann ich gut leben.
Bisher hat es sich halt ständig geändert, wenn das direkt nach dem Update nicht mehr so ist denke ich halt das da was nicht stimmt.
Leider werde ich noch warten müssen bis mal ausreichend Sonne da ist um zu sehen ob wirklich geregelt wird.
Heute dann das nächste Update.

RicoBach68, 2025-01-24:
Habe gestern das neue 2024.01.23 Update eingespielt. Leider regelt der DLP immer nur eine kurze Zeit auf die Anforderung des Stromzählers. Dann verharrt er bei einem beliebigen Wert. Durch einen Neustart der OpenDTUoB kann ich den DLP für eine kurze Zeit wieder zum regeln der korrekten Werte bewegen.

2x HM-1500 (gespeist mit Zendure HUB-2000)

Vielleicht kann jemand in meinem Log den Fehler erkennen???

11:08:51.000 > [DPL] targeting 0 W, base load is 50 W, power meter reads 470.8 W (valid)
11:08:51.009 > [DPL] inverter 116183124923 is behind power meter producing 53 W
11:08:51.016 > [DPL] inverter 116183121705 is behind power meter producing 38 W
11:08:51.022 > [DPL] requesting 562 W from 2 solar-powered inverters currently producing 91 W (diff 471 W, hysteresis 10 W)
11:08:51.028 > [DPL] will cover 91 W using 2 solar-powered inverters
11:08:51.035 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 471 W AC
11:08:51.041 > [DPL] requesting 0 W from 0 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 10 W)
11:08:51.047 > [DPL inverter 116183124923]:
11:08:51.060 >     solar-powered, producing 53 W
11:08:51.067 >     lower/current/upper limit: 25/60/1500 W, output capability: 1500 W
11:08:51.072 >     sending commands enabled, reachable, eligible
11:08:51.077 >     max reduction production/standby: 28/28 W, max increase: 0 W
11:08:51.082 >     target limit/output/state: -1 W (unchanged)/53 W/unchanged, 0 update timeouts
11:08:51.087 >     MPPTs AC power: a: 27 W b: 27 W
11:08:51.095 > [DPL inverter 116183121705]:
11:08:51.102 >     solar-powered, producing 38 W
11:08:51.108 >     lower/current/upper limit: 25/45/1500 W, output capability: 1500 W
11:08:51.113 >     sending commands enabled, reachable, eligible
11:08:51.119 >     max reduction production/standby: 13/13 W, max increase: 0 W
11:08:51.125 >     target limit/output/state: -1 W (unchanged)/38 W/unchanged, 0 update timeouts
11:08:51.134 >     MPPTs AC power: a: 20 W b: 19 W
11:08:51.140 > Fetch inverter: 116183124923
11:08:51.146 > Queue size - NRF: 2 CMT: 0
11:08:51.153 > [PowerMeterHttpJson] New total: 474.68
11:08:51.159 > Last missing
11:08:51.169 > Request retransmit: 3
11:08:51.253 > Last missing
11:08:51.259 > Request retransmit: 4
11:08:51.368 > Middle missing
11:08:51.377 > Request retransmit: 1
11:08:51.469 > Success
11:08:52.022 > Last missing
11:08:52.030 > Request retransmit: 3
11:08:52.127 > [DPL] targeting 0 W, base load is 50 W, power meter reads 474.7 W (valid)
11:08:52.137 > [DPL] inverter 116183124923 is behind power meter producing 53 W
11:08:52.142 > [DPL] inverter 116183121705 is behind power meter producing 34 W
11:08:52.147 > [DPL] requesting 562 W from 2 solar-powered inverters currently producing 87 W (diff 475 W, hysteresis 10 W)
11:08:52.153 > [DPL] will cover 87 W using 2 solar-powered inverters
11:08:52.161 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 475 W AC
11:08:52.164 > [DPL] requesting 0 W from 0 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 10 W)
11:08:52.170 > [PowerMeterHttpJson] New total: 476.70
11:08:52.177 > [DPL inverter 116183124923]:
11:08:52.185 >     solar-powered, producing 53 W
11:08:52.191 >     lower/current/upper limit: 25/60/1500 W, output capability: 1500 W
11:08:52.196 >     sending commands enabled, reachable, eligible
11:08:52.202 >     max reduction production/standby: 28/28 W, max increase: 0 W
11:08:52.207 >     target limit/output/state: -1 W (unchanged)/53 W/unchanged, 0 update timeouts
11:08:52.212 >     MPPTs AC power: a: 27 W b: 27 W
11:08:52.219 > [DPL inverter 116183121705]:
11:08:52.225 >     solar-powered, producing 34 W
11:08:52.231 >     lower/current/upper limit: 25/45/1500 W, output capability: 1500 W
11:08:52.238 >     sending commands enabled, reachable, eligible
11:08:52.244 >     max reduction production/standby: 9/9 W, max increase: 0 W
11:08:52.251 >     target limit/output/state: -1 W (unchanged)/34 W/unchanged, 0 update timeouts
11:08:52.256 >     MPPTs AC power: a: 18 W b: 17 W
11:08:52.263 > Last missing
11:08:52.268 > Request retransmit: 3
11:08:52.273 > Fetch inverter: 116183121705
11:08:52.280 > Queue size - NRF: 3 CMT: 0
11:08:52.286 > Last missing
11:08:52.292 > Request retransmit: 3
11:08:52.376 > Last missing
11:08:52.383 > Request retransmit: 3
11:08:52.483 > Last missing
[... 1675 lines cut ...]
11:09:47.865 > [DPL] will cover 87 W using 2 solar-powered inverters
11:09:47.866 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 480 W AC
11:09:47.875 > [DPL] requesting 0 W from 0 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 10 W)
11:09:47.885 > [DPL inverter 116183124923]:
11:09:47.886 >     solar-powered, producing 53 W
11:09:47.892 >     lower/current/upper limit: 25/60/1500 W, output capability: 1500 W
11:09:47.900 >     sending commands enabled, reachable, eligible
11:09:47.901 >     max reduction production/standby: 28/28 W, max increase: 0 W
11:09:47.907 >     target limit/output/state: -1 W (unchanged)/53 W/unchanged, 0 update timeouts
11:09:47.914 >     MPPTs AC power: a: 26 W b: 27 W
11:09:47.920 > [DPL inverter 116183121705]:
11:09:47.926 >     solar-powered, producing 34 W
11:09:47.933 >     lower/current/upper limit: 25/45/1500 W, output capability: 1500 W
11:09:47.933 >     sending commands enabled, reachable, eligible
11:09:47.939 >     max reduction production/standby: 9/9 W, max increase: 0 W
11:09:47.945 >     target limit/output/state: -1 W (unchanged)/34 W/unchanged, 0 update timeouts
11:09:47.952 >     MPPTs AC power: a: 18 W b: 17 W
11:09:47.958 > Middle missing
11:09:47.964 > Request retransmit: 3
11:09:47.969 > Fetch inverter: 116183124923
11:09:47.978 > Queue size - NRF: 2 CMT: 0
11:09:47.984 > Success
11:09:48.447 > [PowerMeterHttpJson] New total: 477.53
11:09:48.489 > Middle missing
11:09:48.504 > Request retransmit: 2

AndreasBoehm, 2025-01-24:
@RicoBach68 
Das Problem mit Plug&Play Akkus ist das diese sehr ungleichmäßig oder stark verzögert Leistung abgeben. Beides führt dazu das der DPL annimmt das es keinen Sinn hat das Limit zu erhöhen, da nicht mit mehr Leistung zu rechnen ist.

Diese "prüfungen" finden nur im Solarbetrieb statt (Wechselrichter wird von Solarmodulen gespeist) und wenn dem wirklich so ist, machen diese checks auch Sinn.

Du hast zwei Optionen um das Problem zu umgehen:
a) WR auf Batteriebetrieb umstellen (häckchen bei 'Wechselrichter wird von Solarmodulen gespeist' rausnehmen)
b) "Überskalierung" aktivieren (häckchen bei 'Verschattetet/Ungenutzte Eingänge ausgleichen' setzen) und dann mit Wert in 'Schwellenwert für Überskalierung' rumprobieren ob du es damit besser ans laufen bekommst

RicoBach68, 2025-01-24:
Danke Andreas für deine schnelle Antwort. Ich werde beide Varianten ausprobieren.
Muss ich bei den Start und Stop Schwellwerten und dem Lastkorrekturfaktor etwas beachten? Mein HUB-2000 liefert zwischen 46V und 49V. Muss ein Wechselrichter für Spannungsmessungen ausgewählt werden? Das Zendure System regelt sich ja komplett selbst.
Es ist trotzdem komisch. Mit alten OpenDTUoB Versionen und nur einem Wechselrichter lief das alles völlig problemlos. Der HUB-2000 kann binnen einer Sekunde 1200W liefern. Da war noch nie etwas unregelmäßig oder verzögert.

Liebe Grüße und vielen Dank.

Rico.

AndreasBoehm, 2025-01-24:
@RicoBach68 Hier gibts eine ganz gute Anleitung, passend zu deinem Setup: https://github.com/hoylabs/OpenDTU-OnBattery/pull/1473#issuecomment-2604680142

KarlHeinzz1, 2025-01-27:
Hallo,

ich habe exakt das gleiche Problem. Bei mir hängen 4x BKW Speicher an je einem Eingang vom HMS2000.
Der Wr bleibt beim unteren Limit hängen (52W) wenn ich als "Solarmodule gespeist" verwende und macht davon auch nur die halbe Leistung von dem eingestellten Limit.
Stelle ich es auf Batterie um, schaltet sich der WR ab (Fernsteuerung) sobald einer der Speicher nicht mehr liefert, vermutlich wegen der dann unterschrittenen Spannung, je nachdem auf welchen Speicher er schaut. Es liegen dann noch <1V an. Es ist dann auch "Unterspannung" im Speicher.
Besser wäre es, man könnte die Spannungsüberwachung aus machen und Opendtuonbattery macht dann einfach weiter.

Am besten wäre es, das Limit würde einfach dumm gesetzt, egal was der WR grade meint liefern zu können und man kann 4 Speicher wie Solarmodule verwenden. Ist ja im Prinzip nichts anderes für den WR. Mehr Einstellungen braucht es dann nicht.

Josef1966, 2025-01-27:
> [@RicoBach68](https://github.com/RicoBach68) Hier gibts eine ganz gute Anleitung, passend zu deinem Setup: [#1473 (comment)](https://github.com/hoylabs/OpenDTU-OnBattery/pull/1473#issuecomment-2604680142)

@AndreasBoehm Ich würde - sofern von euch gewünscht - die Anleitung gerne im Wiki hinterlegen . Ich weiß nur nicht, wie man das (korrekt) macht - gibt es hier vielleicht irgendeine Beschreibung? Ich würde dann den anhängenden pdf-File dort hinterlegen und zukünftig pflegen/updaten.

[System-Daten_Zendure-SolarFlow_Hoymiles-HMS1600_OpenDTU-DPL_01.2-2025.pdf](https://github.com/user-attachments/files/18558870/System-Daten_Zendure-SolarFlow_Hoymiles-HMS1600_OpenDTU-DPL_01.2-2025.pdf)

Josef1966, 2025-01-28:
@Manos1966 .... vielen Dank für die Vorarbeit!

1. Die Überschrift finde ich ok, das kann man so lassen.
2.  Hinsichtlich Platzierung hätte ich einen Vorschlag, da es sich ja nicht "nur" um ein Hardware-Thema handelt.
Kann man nicht einen neuen Ober-Bereich "System-Konfigurationen" aufmachen (z.B. zwischen "Basics" und "Advanced", in dem man dann alle bisher vorgeschlagenen Aufbauten und Beispiele zusammenfasst? Also neben der "Zendure SolarFlow with OpenDTU‐OnBattery" auch die Konfigurationen aus dem bisherigen Bereich "Builds & Examples" und die sonst noch unter "Pages" verstreuten Beiträge, z.B. https://github.com/hoylabs/OpenDTU-OnBattery/wiki/HM%E2%80%901500,-4x-385W,-150-35-MPPT,--2x-25,6V-(51,2V)-LiFePo und https://github.com/hoylabs/OpenDTU-OnBattery/wiki/HM%E2%80%90400,-2x410Wp,-150-35-MPPT,-Pylontech-2,4kWh-48V---HM%E2%80%90350,-1x410Wp hineinnimmt.
3. Ich würde dann erst mal mit dem bisherigen Inhalt aus dem pdf (Text + Bilder) starten und dann im Nachgang ggf. selber noch weitere Erfahrungen und Informationen ergänzen.

Was haltet ihr davon?

KarlHeinzz1, 2025-01-28:
Also an sich muss man schon sagen ein super System was hier gebaut wurde.
Das es bei den vielen Varianten mal harkt ist doch ganz normal. 

Man könnte noch überlegen eine Fake Wert als Batteriespannung wenn es hier keine Abhilfe geben soll. Leider hab ich bis zum WE keinen Zugriff auf das Gerät.
Also Batteryinterface auf MQTT stellen und dann Festwert einlesen. Vielleicht würde der Tasmota Lesekopf gehen und dort z.b. die Netzspannung. Oder man versucht einen Wert von opendtu gleich wieder einzulesen, z.b. die Temperatur.

Josef1966, 2025-01-28:
@RicoBach68 

Nummer 2 habe ich bei meinem System auch schon gehabt. Ursache unbekannt.

KarlHeinzz1, 2025-01-30:
Gestern verhielt sich das System noch mal anders merkwürdig. Nach ein paar Sonnenstunden hatten alle Speicher ca. 20% geladen und wären bereit gewesen einzuspeisen. Aber statt das die Reglung wieder anläuft, blieb es bis 22:30 beim ausgeschalteten WR. Um 22:30 ging dann der WR an, hat mit 100% entladen bis alle Speicher leer waren. Dabei war um die Zeit keine größere Last mehr aktiv.
Also den Tagesertrag einfach ins Netz, nicht so toll.

Es wäre von der Logik eigentlich richtig, die Speicher wie Solarmodule zu behandeln, denn alles andere steckt ja in den Speichern (Spannungsüberwachung, DOD, Wlan usw.) Kein Grund irgendwas zu regeln außer der WR Leistung.
Leider geht das in der aktuellen Version aber nicht, da der WR dann bei der unteren Grenze fest sitzt. 
Daher hier bitte diese erwähnte eingeschränkte Reglung optional machen oder ganz raus nehmen. 
Mich würde interessieren welchen Vorteil das generell bringen soll? Er schadet doch nicht das Limit zu erhöhen, ob es nun Sinn hat oder nicht. 
Das Problem betrifft auch nicht nur Zendure, das tritt mit jedem BKW Speicher auf.

RicoBach68, 2025-01-30:
@AndreasBoehm @schlimmchen 

Ich denke, die Anpassung von OpenDTUoB an Plug&Play Speichersysteme wär eine lohnenswerte Sache. Immer mehr Zendure (und andere) Betreiber von Plug&Play Systemen stellen fest, dass die Steuerung durch OpenDTUoB wesentlich schneller und genauer arbeiten kann als die von den Herstellern angebotenen. Ich glaube, dass in Zukunft mehr User mit Plug&Play Systemen OpenDTUoB nutzen werden als Bastler von selbst gebauten Systemen. Dadurch könnte OpenDTUoB einen wesentlich höheren Anwenderkreis erschließen und viel mehr Menschen nützlich sein.
Im wesentlichen sind fast alle Vorraussetzungen dafür schon geschaffen, das wichtigste wär aber, dass das System ohne ständiges Eingreifen von selbst dauerhaft zuverlässig arbeitet. Dazu brächte es nur noch einen "Easy Mode" für Plug&Play Systeme, der sich auf eine reine Steuerung beschränkt, ohne Spannungsmessung und vielen anderen für uns unnötigen Parametern. Ich weiss natürlich, dass all die wunderbaren Möglichkeiten für Bastler ein Paradies sind aber ich denke, man sollte auch die Plug&Play Nutzer im Blick behalten.

Liebe Grüße
Rico.

synack666, 2025-01-31:
Tach auch!
Heute mal Sonne und was sehe ich, es wird geregelt.
Sorry für meine fehlerhafte Einschätzung!

Soll ich dicht machen?

closed 2025-01-31 by synack666