hoylabs/OpenDTU-OnBattery #1556: Nighttime Discharging trickles the battery

schlimmchen, 2025-01-18:
### What happened?

My system periodically starts the inverter in the evening to trickle the last bit of juice out of the battery, even though the stop threshold was reached previously.

[image]

### To Reproduce Bug

Enable "Use battery at night even if only partially charged" and set the load correction to a value that causes the battery voltage to bounce back higher than the stop threshold once discharging stops.

### Expected Behavior

If "nighttime discharging" was used to start a discharge cycle, "nighttime discharging" must not be used again until.... a new day began, I guess.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU-OnBattery?

60fc9b39

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

ali-baba-de, 2025-01-19:
interesting finding, I had that when using the "Use Voltage Threshols Only " setting and that was one of the reasons to move back to SOC based limits. Did not notice it since then:

[image]

but when looking a bit more in depth I see now that at exactly at midnight the trickling starts:

[image]

the only thing that comes into my mind wrt mignight is the automatic inverter restart??? 

PS: there is a typo in the "Use Voltage Threshols Only " label :)

schlimmchen, 2025-01-19:
Yeah, wen using SoC, the SoC will not jump up, even though the voltage does, when discharging ends.

What the hell is going on at midnight in your graph? That is weird... I agree that the inverter restart probably has something to do with that.

> PS: there is a typo in the "Use Voltage Threshols Only " label :)

Yes. Fixed in b65db96d.

spcqike, 2025-01-20:
i don't know if its related to something here, but there must be some general trigger at midnight

as you know, i only have a solar powered inverter, but openDTU reports some stats and even a power reading at midnight to HA

[image]

[image]

even though i don't want to send or receive data at night

[image]

PR #1558 (2025-01-21): Fix nighttime discharging starting multiple times during the night
Make sure we only start a (nighttime) discharge cycle on a partially charged battery once during the night. This should be easy as the `_nighttimeDischarging` flag is already only reset at sunrise, so we "simply" have to NOT start another discharge cycle if the flag is already true.

Closes #1556.
closed 2025-01-21 by schlimmchen