hoylabs/OpenDTU-OnBattery #1561: Keeping WebConsole "open" causes reboot

schlimmchen, 2025-01-20:
### What happened?

We know, and in a discussion with Aviv on discord this was shown once more, that the ESP32 performs unexpected reboots when the web console is kept "running" for a longer periode of time.

### To Reproduce Bug

Open a browser and navigate to the web console. Keep it running for long (how long is unclear). Possibly this happens faster if more than one tab is open with the web console. Observe that the ESP32 reboots unexpectedly.

### Expected Behavior

Keeping a browser connected to the web console should not crash the ESP32.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.01.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Sorbat, 2025-01-21:
Also ich habs gerade nochmal getestet, bei mir ist es nicht nach längerer Zeit. Sobald ich die Webkonsole öffne, wird innerhalb von 15-20 Sekunden ein Reboot durchgeführt, welcher sich dann alle 25-30 Sekunden wiederholt. 
Ich habe dabei einen Tab mit der Webkonsole geöffnet, nicht mehrere.

schlimmchen, 2025-01-21:
Hast du die Möglichkeit, per USB einen Backtrace mitzuschneiden, wenn das bei dir so leicht reproduzierbar ist? Ist es bei mir "leider" nicht... Das wäre super! Muss dann bitte wissen, welche Version genau du einsetzt, auch welches PIO Environment.

Sorbat, 2025-01-21:
Ich kann es versuchen. Aber was ist ein Backtrace?

Ich vermute, ich soll über die USB Konsole ein Log erstellen, während die Webkonsole läuft?

schlimmchen, 2025-01-21:
Genau. Während die Büchse unerwartet neu startet. Im seriellen Log steht dann was für mich drin (der Backtrace). Das Wort steht da auch. Sieht z.B. so aus: 

```
abort() was called at PC 0x401ec56f on core 0

Backtrace: 0x400840ed:0x3ffef420 0x4008fb61:0x3ffef440 0x40095f31:0x3ffef460 0x401ec56f:0x3ffef4e0 0x401ec5b6:0x3ffef500 0x401ec8fb:0x3ffe0

ELF file SHA256: 38eee8b326b2eef7
```

Bitte mind. zwanzig Zeilen vorher und nachher mit kopieren, zwecks des Kontexts. Danke für deine Hilfe!

schlimmchen, 2025-01-22:
@Sorbat Welche Batterieschnittstelle verwendest du? Ist es eine Pytes oder Pylontech oder eine andere per CAN angeschlossene Batterie?

@AndreasBoehm Aviv hat gerade für uns herausgefunden, dass das Problem verschwindet, wenn er die Batterieschnittstelle zu seiner Pytes abschaltet. Vorher habe ich anhand seiner Backtraces stark vermutet, dass es sich um ein out-of-memory Problem handelt. Das konnte er zusätzlich untermauern: Wenn er auf den Heap schaut in der Web UI unter Info -> System, dann steigt die Heap Usage (mehr oder weniger) kontinuierlich an, bis zu bei >90% ist und der ESP neu starten muss.

Ich versuche das Problem jetzt nochmal zu reproduzieren unter der Annahme, dass es nicht am Battery Provider liegt, sondern daran, dass ein Task, der nicht die main loop() ist, Ausgaben erzeugt.

AndreasBoehm, 2025-01-22:
Weißt du wie lange es bei Aviv dauert bevor der heap voll läuft? Dann versuche ich das mal nachzustellen.

schlimmchen, 2025-01-22:
@gitisgreat2023 Du hast diesem Issue ein 👍 spendiert. Bist du von dem Problem betroffen in dem Sinne, als dass du es reproduzieren kannst? Hast du eine Batterie per CAN angebunden?

AndreasBoehm, 2025-01-22:
Nach 5 minuten mit verbose logging für die folgenden komponenten hat sich nichts getan, läuft immer noch.
- MQTT
- DTU
- Solarladeregler Ve.Direct
- Stromzähler MQTT
- DPL
- Batterie Pytes CAN

[image]

AndreasBoehm, 2025-01-22:
Wenn ich 5 Tabs der Konsole aufmache läuft der Heap innerhalb von 10 Sekunden voll. Dann gibts nen reboot.

schlimmchen, 2025-01-22:
Danke fürs Testen! Schade, dass wir scheinbar woanders weitersuchen müssen. Ist das ein Release oder ein development Build?

AndreasBoehm, 2025-01-22:
Build ist von einem PR `hoylabs/OpenDTU-OnBattery/pr1558-202501192043` #1558

AndreasBoehm, 2025-01-22:
Wenn die DTU und das Macbook per LAN im Netzwerk hängen brauch ich 12 Tabs um einen reboot zu provozieren.

Macbook ist dann auch komplett ausgelastet.
[image]

AndreasBoehm, 2025-01-22:
Sobald der Client (Browser) nicht mehr nachkommt gibts nen reboot. 
Das es jetzt 12 Tabs gebraucht hat und vorher nur 5 halte ich für einen Zufall, vielleicht durch den reboot bedingt den es gebraucht hat um den PoE HAT zu montieren.

Sorbat, 2025-01-22:
[image]

Das sind meine Werte unter Netzwerk.

Sorbat, 2025-01-22:
Ich habs gerade nochmal am Mini M1 mit Safari getestet, ein Tab mit der Konsole, neustart nach ca. 1 min. 

Könnte es vielleicht an meinem ESP liegen? Also an der Hardware?

schlimmchen, 2025-01-22:
> Könnte es vielleicht an meinem ESP liegen? Also an der Hardware?

Meh... Hast du nicht ein OpenDTU Fusion? Solange das genügend Spannung/Strom hat, suche ich da keinen Fehler. Ausnahme: Interferenzen. Es ist ja noch ungeklärt, warum der Reboot bei dir plötzlich nicht mehr auftritt, wenn du das Battery Interface ausschaltest. Das ist ja hochgradig merkwürdig...

Sorbat, 2025-01-22:
Ja das Fusion. habe seit gestern aber wieder ein zweites hier und könnte es mal tauschen. 
Das Hutschienennetzteil hat 2.4A. Spannung kann ich nochmal messen, aber normalerweise stelle ich die immer auf 5,01V ein. 

> Es ist ja noch ungeklärt, warum der Reboot bei dir plötzlich nicht mehr auftritt, wenn du das Battery Interface ausschaltest. Das ist ja hochgradig merkwürdig...
ich dachte, vielleicht sind es einfach zu viele daten und wenn der Esp nen knacks hat, verabschiedet er sich 

Ich weiß nicht, ob das relevant sein könnte. Aber ich hab nicht die Pytes v5, sondern die 48100. Vielleicht ist da was anders?

Sorbat, 2025-01-23:
Mir war eben aufgefallen, durch Andreas Kommentar oben, dass Batterie auch noch ein Verbose log hat 🙈

Das hab ich jetzt mal deaktiviert, seitdem läuft die Konsole ohne reboot seit 20min

Heap ist von 55 auf 79 gestiegen in der Zeit 

Dann Batterie verbose wieder aktiviert, reboot nach 30 Sekunden

schlimmchen, 2025-02-02:
@Sorbat Du bist ja aktuell (bedauerlicher Weise) auf dem Oktober-Release unterwegs. Tritt denn das Problem da auch auf? Das zu wissen wäre immerhin eine kleine Erkenntnis.

Sorbat, 2025-02-02:
Wieso bedauerlicher Weise? 

Ja, das tritt auch bei [2024.10.22](https://github.com/hoylabs/OpenDTU-OnBattery/releases/tag/2024.10.22) auf.

Bei nur aktiviertem DPL Verbose dauert es ca 4:30min bis zum Reboot, bei nur Batterie Verbose ca. 65sek, und wenn beides aktiviert ist ca. 50 sek.

schlimmchen, 2025-02-02:
> Wieso bedauerlicher Weise?

Naja, weil das heißt dass wir für manche Nutzer einen Rückschritt gemacht haben und keinen Fortschritt... Das ist natürlich schade. Keine Kritik an dir. Wenn die alte Version für dich insgesamt besser funktioniert, dann nutz die gerne.

dk7td, 2025-02-21:
Hi!

Habe gerade mal - wie in der anderen Discussion besprochen - den seriellen Log aktiviert und nach ein paar Minuten offener Konsole kam das hier. Hoffe das hilft:

abort() was called at PC 0x4211c4cb on core 1

Backtrace: 0x40378096:0x3fcebb40 0x4037d4c1:0x3fcebb60 0x403842d1:0x3fcebb80 0x4211c4cb:0x3fcebc00 0x4211c512:0x3fcebc20 0x4211c6ed:0x3fcebc40 0x4211c7a8:0x3fcebc60 0x42023889:0x3fcebc80 0x42023e9b:0x3fcebcb0 0x4208d4df:0x3fcebd00 0x4207da36:0x3fcebda0 0x4207d522:0x3fcebdc0 0x4207c311:0x3fcebe00 0x4207c51a:0x3fcebe30 0x4207d778:0x3fcebea0 0x4207b103:0x3fcebf00 0x4201982e:0x3fcebf30 0x4215d651:0x3fcebf50 0x4205f792:0x3fcebf70 0x4205f90d:0x3fcebf90 0x4205f926:0x3fcebfb0 0x42091ccd:0x3fcebfd0

ELF file SHA256: b9e75a3b8a4bc5b1

E (8711) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...

Und nach wenigen Minuten wieder:

abort() was called at PC 0x4211c4cb on core 1

Backtrace: 0x40378096:0x3fcebb50 0x4037d4c1:0x3fcebb70 0x403842d1:0x3fcebb90 0x4211c4cb:0x3fcebc10 0x4211c512:0x3fcebc30 0x4211c6ed:0x3fcebc50 0x4211c7a8:0x3fcebc70 0x420235f1:0x3fcebc90 0x42023f0d:0x3fcebcb0 0x4208d4df:0x3fcebd00 0x42035655:0x3fcebda0 0x42034bbe:0x3fcebe60 0x4215e121:0x3fcebf50 0x4205f792:0x3fcebf70 0x4205f90d:0x3fcebf90 0x4205f926:0x3fcebfb0 0x42091ccd:0x3fcebfd0

ELF file SHA256: b9e75a3b8a4bc5b1

E (16049) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...

dk7td, 2025-02-22:
Moin!

Gerade gabs wieder einen Neustart:

abort() was called at PC 0x4211c4cb on core 1

Backtrace: 0x40378096:0x3fcebb50 0x4037d4c1:0x3fcebb70 0x403842d1:0x3fcebb90 0x4211c4cb:0x3fcebc10 0x4211c512:0x3fcebc30 0x4211c857:0x3fcebc50 0x4211c7b6:0x3fcebc70 0x420235f1:0x3fcebc90 0x42023f0d:0x3fcebcb0 0x4208d4df:0x3fcebd00 0x42035655:0x3fcebda0 0x42034bbe:0x3fcebe60 0x4215e121:0x3fcebf50 0x4205f792:0x3fcebf70 0x4205f90d:0x3fcebf90 0x4205f926:0x3fcebfb0 0x42091ccd:0x3fcebfd0

ELF file SHA256: b9e75a3b8a4bc5b1

E (8895) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x4216271a
SPIWP:0xee
mode:DOUT, clock div:2
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
SHA-256 comparison failed:
Calculated: ca4adcc736b98868c479048883edac1969cfd9dde0951723205cf1625628f181
Expected: 27accf466d6a3ffbdd2fe4a9dcdf2a85d67a20eb4b1f2dced8b0c8a88771aee2
Attempting to boot anyway...
entry 0x403c98d0
E (525) esp_core_dump_flash: No core dump partition found!
E (525) esp_core_dump_flash: No core dump partition found!

schlimmchen, 2025-02-24:
[backtrace1.txt](https://github.com/user-attachments/files/18944109/backtrace1.txt)
[backtrace2.txt](https://github.com/user-attachments/files/18944108/backtrace2.txt)
[backtrace3.txt](https://github.com/user-attachments/files/18944110/backtrace3.txt)

Alle drei deuten auf ein out-of-memory hin.

Bitte nutze die Webkonsole nicht mehr, bis wir einen Workaround implementiert haben.

Falls du bisher die Webkonsole regelmäßig genutzt hast, also nicht nur um Backtraces für dieses Issue zu provozieren, dann lass das bitte vorerst. Dann lösen sich deine Probleme in #1557 ggf. von selbst.

Außerdem bestehe ich darauf, dass du den Flash deines ESP32 Moduls einmal vollständig löschst, die Firmware .factory.bin neu einspielst, und deine Einstellungen und dein pin mapping aus einem Backup wiederherstellst, bevor sich Leute @spcqike @Snusme83 weiterhin mit deinem Problem beschäftigen.

```
SHA-256 comparison failed:
Calculated: ca4adcc736b98868c479048883edac1969cfd9dde0951723205cf1625628f181
Expected: 27accf466d6a3ffbdd2fe4a9dcdf2a85d67a20eb4b1f2dced8b0c8a88771aee2
Attempting to boot anyway...
```

Da ist offensichtlich irgendetwas nicht in Ordnung, und das kann alles sein und jede Auswirkung haben. Daher: Bitte einmal den Flash zurücksetzen und das ESP32-Modul neu initialisieren. Z.B. mit esptool.py oder dem [Flash Download Tool](https://www.espressif.com/en/support/download/other-tools).

dk7td, 2025-02-24:
Danke für die Rückmeldung!
Gibt es eine Beschreibung, wie ich das Flash vollständig löschen kann? Geht das nur mit dem esptool.py oder dem Flash Download Tool?
Ich habe hier die OpenDTU Fusion im Einsatz.

dk7td, 2025-02-24:
> Falls du bisher die Webkonsole regelmäßig genutzt hast, also nicht nur um Backtraces für dieses Issue zu provozieren, dann lass das bitte vorerst. Dann lösen sich deine Probleme in [#1557](https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1557) ggf. von selbst.

Die hatte ich genutzt, um den Fehler mit der fixen Einspeisung einzugrenzen - allerdings war sie beim Fallback auf die Grundeinspeisung jeweils nicht aktiv gewesen, also keine entsprechenden Probleme dadurch.
Normalerweise nutze ich die Webkonsole nicht.

Danke für den Hinweis!

schlimmchen, 2025-02-24:
>  Geht das nur mit dem esptool.py oder dem Flash Download Tool?

Mit den webbasierten Tools geht es, soweit ich das überblicke, jedenfalls nicht.

esptool.py erase

oder den Erase Knopf in diesem Windows-Dings.

dk7td, 2025-02-24:
@schlimmchen der comparison failed kommt aber leider weiterhin ...

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x4216305a
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
SHA-256 comparison failed:
Calculated: 1bd442e459ee1f562eca9028c37a6f7cf01acd691640506110d285f4e298d736
Expected: 27accf466d6a3ffbdd2fe4a9dcdf2a85d67a20eb4b1f2dced8b0c8a88771aee2
Attempting to boot anyway...
entry 0x403c98d0
E (526) esp_core_dump_flash: No core dump partition found!
E (526) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
Initialize FS... done
Reading configuration... Check for default DTU serial... done
done
Reading language pack... done
Reading PinMapping... found valid mapping done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
[SyslogLogger] Disable
[SyslogLogger] Syslog not enabled
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Display... done
Initialize LEDs... done
Initialize Hoymiles interface... NRF: Connection successful
CMT: Connection successful
  Setting country mode... 
  Setting CMT target frequency... 
  Setting radio PA level... 
CMT TX power set to 0 dBm
  Setting DTU serial... 
  Setting poll interval... 
  Setting verbosity... 
  Adding inverter: 114184847551 - Garage-Ersatz done
  Adding inverter: 114190550241 - Garage-neu done
done
Initialize Huawei AC charger interface...
Switch to WiFi mode
Setting Hostname... done
Configuring WiFi STA using new credentials... done
Configuring WiFi STA DHCP IP... done
[SyslogLogger] Disable
[SyslogLogger] Syslog not enabled
[DPL] automatic inverter restart disabled
[DPL] reloading DPL configuration
[DPL] targeting -30 W, base load is 150 W, power meter reads 0.0 W (stale)
[DPL] requesting 150 W from 1 solar-powered inverter currently producing 0 W (diff 150 W, hysteresis 10 W)
[DPL] will cover 0 W using 1 solar-powered inverter
[DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 150 W AC
[DPL inverter 114190550241]:
    solar-powered, standing by at 0 W
    lower/current/upper limit: 50/0/800 W, output capability: 0 W
    sending commands enabled, reachable, disqualified (max output unknown)
    max reduction production/standby: 0/0 W, max increase: 0 W
    target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
    MPPTs AC power: a: 0 W b: 0 W

dk7td, 2025-02-24:
Ich habe den ERASE-Button im Flash Download Tool verwendet und auf FINISH gewartet. Danach hat die OpenDTU nach einem Reset nicht mehr reagiert. Sollte also gelöscht gewesen sein.

schlimmchen, 2025-02-24:
Hm..... also das ist definitiv nicht normal.

Hier ist mein Fusion-Development-Board:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x421692fa
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (416) esp_core_dump_flash: No core dump partition found!
E (416) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
```

Kann leider nicht abschließend bewerten, wie problematisch das ist. Sieht mir jedenfalls nach einem Fehler im Flash aus, was mich beunruhigt und mich an der Zuverlässigkeit deines Moduls zweifeln lässt.

Der Erase-Knopf sollte es eigentlich tun. Kannst du esptool.py noch ausprobieren? Da weiß ich, dass `esptool.py erase_flash` tatsächlich alles aufräumt...

dk7td, 2025-02-24:
Danke für den Hinweis. Probiere ich morgen oder übermorgen mal aus. Installiert sollte es auf meinem Mac mini mit Debian eigentlich sein.

dk7td, 2025-02-24:
Hat mich jetzt doch nicht ruhig ins Bett gehen lassen ... ich mußte das noch durchziehen ;)
Kaum macht mans richtig, klappt es auch!

dk7td@macmini:~$ sudo esptool.py erase_flash
esptool.py v4.7.0
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 80:65:99:c4:87:a0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 6.5s
Hard resetting via RTS pin...

Und jetzt sieht das ganze auch besser aus:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x420a0f1b
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (411) esp_core_dump_flash: No core dump partition found!
E (411) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU

Mal sehen, ob meine Probleme weiterhin bestehen.
Danke für den Tip!

PR #1748 (2025-03-23): Fix webconsole heap issue
I am done trying to figure out what goes wrong in some people's setups. Aviv/Sorbat and I are chatting about this from time to time for weeks now, but I am out of ideas.

However, what I can provide, is this changeset, which solves the worst part of the issue: The heap will not overrun any more and using the webconsole should therefore no longer cause unexpected reboots.

New approach: We move log message lines to the websocket clients individually. If any client's queue is full after adding the message line, we do the following:
1. Replace the last message with a war [... cut]
closed 2025-03-23 by schlimmchen