hoylabs/OpenDTU-OnBattery #1564: [Request] Show all 3-phases for HMT inverters in live view

AndreasBoehm, 2025-01-21:
### Is your feature request related to a problem? Please describe.

As written by @Saarblick: The 3 phases are not shown separately in the live view.
> ... ein dreiphasiger HMT2250. Daher die 400V Die 3 Phasen werden ja nicht richtig von der DTU ausgewiesen... (https://github.com/hoylabs/OpenDTU-OnBattery/issues/1535#issuecomment-2603186005)

[image]

### Describe the solution you'd like

Show all phases individually.

kthemall, 2025-03-08:
i appreciate that feature too. thx tom

AndreasBoehm, 2026-08-18:
This feature might land soon: https://github.com/tbnobody/OpenDTU/pull/3188