hoylabs/OpenDTU-OnBattery #1587: Home Assistent auto-discovery broken for multiple batteries

schlimmchen, 2025-01-26:
As the MQTT Home Assistent handler for the battery uses a hard-coded "serial", users cannot use two OpenDTU-OnBattery to talk to multiple batteries.

I think that integration of battery information in home automation and/or MQTT is a stand-alone feature of OpenDTU-OnBattery, so that should work properly when combining multiple "agents" (OpenDTU-OnBattery instances) to connect to multiple batteries.

> Hallo, seit diesem WE nutze ich eine 2. Open DTU Parallel in Verbindung mit HA. 
Diesen Schritt bin ich gegangen, da ich seit je ein24V mit AGM und ab sofort ein 48V System mit Pylontech 5K betreibe. Vorerst soll das auch so bleiben, auch wenn es Energetisch Unsinn ist, das 24V System jedoch ein vorhandenes USV System für Server speist. und solange die Batterien noch Ok sind, möchte ich Sie nicht entsorgen, bzw gegen Converter Tauschen. Das ist insofern alles OK, beide DTU verichten Ihren Dienst, bis Home Assistant bzw. Mqtt ins Spiel kommt, ab da wirds komisch....der Topic "dtu_battery_0001" ist ja in beiden Fällen der gleiche. So steuert der Victron SmartShunt als auch der Pylontech seine Daten bei ...das führt zu Datensalat im HA. Gibt es eine Möglichkeit die Pfade entsprechend anzupassen ? Leider erlaubt es momentan ja jeweils nur ein Batteriesystem seitens der openDTU.

_Originally posted by @Mickel4711 in https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1586_

schlimmchen, 2025-01-26:
Still borken in #1451, but for no good reason: There, every battery provider already has a "HASS integration" instance, which is specialized, i.e., derived from the base class that does all the sensor publishing. It shouldn't be too hard to transport an actual serial into that class instance. Something to consider is merely that the hass integration implementation must not publish sensors until the serial is known. That should also be no big deal.

AndreasBoehm, 2025-01-29:
What if we include the unique dtu id as described in https://github.com/hoylabs/OpenDTU-OnBattery/issues/963 in every topic to avoid collisions? 

Or as you suggested use a different strategy for things that can be uniquely identified and "moved" between different DTUs without causing interruptions/duplicates on mqtt receivers. (Batteries, ac chargers, solar chargers)

schlimmchen, 2025-02-02:
> What if we include the unique dtu id as described in https://github.com/hoylabs/OpenDTU-OnBattery/issues/963 in every topic to avoid collisions?

That's a very good idea. Unless this is a breaking change (HA history), which we need to figure out first. If the entities are still the same after implementing such a change, we should do both (use a DTU unique ID in the topics to ensure uniqueness, and use the device's serial number to be able to actually identify what one is looking at).

Manos1966, 2026-01-20:
Hi @spcqike 
Es sind meine Dorf-Nachbarn. Wir haben 8-10 OpenDTU-OnBattery, davon 5 mit Batterie.
Da die Daten ueber die Cloud kommen, bin ich nicht begeistert, so viele offene Ports zu lassen um sie getrennt zu empfangen.

Vier MQTT Container helfen mir nicht glaube ich. Oder wird HA die Batterie-Daten aus vier MQTT Container eindeutig unterscheiden koennen? (Hmmm... es waere eine Interessante Idee)

> die Firmware selber kompilieren und das topic jeweils leicht verändern

Hmmm... wollte ich immer vermeiden (wegen Firmware Updates spaeter) **ABER** das klingt verlockend. 
Ich habe nie OpenDTU-OnBAttery kompiliert, guter Anlass (einen Tod muss man sterben).

Hast du vielleicht einen Link wo ein paar Anweisungen stehen wie/was man kompiliert?
Und wo das Topic dass ich aendern koennte zu finden ist?

Es gab ein Thread im Forum wo jemand genau das gemacht hat, ich habe es damals nicht notiert (und bezahle heute die Strafe dafuer) 🤦‍♂️

spcqike, 2026-02-20:
<details><summary>OT 😉 </summary>
<p>

Im "Write" Fenster hier auf Github, sieht man auch nicht mehr, im Preview sollte es passen

https://github.com/hoylabs/OpenDTU-OnBattery/blob/0dc7b6c5fae6b144dabf691e657d26686ca7fcc2/src/battery/pylontech/Stats.cpp#L45-L71

write:

[image]

preview:

[image]

</p>
</details>

closed 2026-10-11 by AndreasBoehm

Manos1966, 2026-10-11:
DANKE! 🥇