hoylabs/OpenDTU-OnBattery #1588: [Bug] Data age Huawei R4850G2 offset by 3 seconds

gitisgreat2023, 2025-01-27:
### What happened?

The data age of the Huawei charger doesn't reset to zero but get's back to 3 at minimum. In the past it worked like a charm.
@schlimmchen it's a pain to flash it manually, I use 2025.01.10, do you remember fixing something since then? There is appeared for the first time.

[image]

### To Reproduce Bug

Look at the data age of the Huawei.

### Expected Behavior

It counts correctly.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.01.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-01-27:
> it's a pain to flash it manually

See [the docs](https://opendtu-onbattery.net/hardware/esp32_devkit/#flash-memory), where I added more info on how to replace those 38pin ESP32 development boards with ones offering 8 MB of flash (literally minutes ago :wink:).

> do you remember fixing something since then?

Yes, see 370fe1d8e30cf3c1c57d3e64f3cd6ea77e233ac8, which is part of 2025.01.23, which is why I am going to close this issue as completed.

gitisgreat2023, 2025-01-27:
Perfect, thanks for mentioning the 8MB version!
Software wise, do I need to change something, or can I flash it and upload the config and pin layout and it should work?

schlimmchen, 2025-01-27:
> can I flash it and upload the config and pin layout and it should work?

I did not test it, but I expect no issues when doing so. The board's pin-out should be compatible, especially as the Espressif docs only show one overview. As far as I understand it, the baseboard is the same for all ESP32-DevKitC, merely the modules are different.

Actually, I ordered a couple of modules with 8MB flash and intend to solder them onto those immitations to see what happens 😉 Should be a viable upgrade path, except that it requires proper soldering tools and skills.