hoylabs/OpenDTU-OnBattery #1595: [Bug] DPL starts discharging before Voltage Start Threshold has been met

gitisgreat2023, 2025-01-30:
### What happened?

Today DPL started discharging though the start threshold wasn't met (by far) yet. I had no other way to stop it then by increasing the stop voltage above the current voltage and then again put the old value back again.

I only recently switched from SoC start stop to voltage based stop. With SoC I didn't have an issue almost a year long, it never started discharging before the SoC start value was reached. 
Did anybody observe this behavior as well?
I had the Start Threshold at 55.5V (yeah, wanted to fully charge once) and stop at 51.2V. The value was at that moment 51.46V. No way somehow 4Vs more were reported by the inverter.

### To Reproduce Bug

Don't know yet, it just happened out of nowhere.

### Expected Behavior

Discharging starts only when the voltage threshold has been reached.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

205.01.23

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Anything else?

I'm curious whether anyone else also observed this issue, probably then we'll have a chance chasing the cause down.

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

gitisgreat2023, 2025-01-31:
Waaiitt... it's a feature not a bug...

I have "Use battery at night even if only partially charged" ... so probably it's a feature, not a bug, right?

Probably it's a good idea to have a "I" with the remark: "Discharging starts though Start Threshold has not been reached, but stops at the Stop Threshold".

I had interpreted as "use the battery at night as well".

schlimmchen, 2025-02-01:
Or read the [docs](https://opendtu-onbattery.net/firmware/configuration/dpl/#use-battery-at-night-even-if-only-partially-charged)?

Yeah... So we'll keep this open and add a tooltip as you suggested, with a similar text. I will adjust the issue title accordingly.

closed 2025-02-07 by AndreasBoehm