hoylabs/OpenDTU-OnBattery #1597: DPL / Full Solar Pass Throug is not working when Victron is in Volltage/Current limit

embed-3d, 2025-02-01:
### What happened?

Today arround noon my battery was nearly fully charged and my Victron was in float mode. Once my Victron  MPPT Tracker got in the State "Voltage or Current limit", The power of the MPPT dropped down to ~20W and my Inverter power was also set to this Limit.
Eventhoug there was enough power on the panels available my Inverter produced during the sunny time nearly nothing

[image]

### To Reproduce Bug

Get Your Victron MPPT in the MPPT state: "Voltage or Current limit" with the following settings:

[image]

Then wait till the Victron MPPT regulates down. The Inverter will follow

### Expected Behavior

When full Battery pass through is enabled and the MPPT state is "Voltage or Current limit", the invert power should be increased until the max setting of the inverter is reached or the state "MPP Tracker active" is reached

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.11.20

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Anything else?

I just realized there was a new release. But this issue is not covered there.

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-02-01:
This is a [documented](https://opendtu-onbattery.net/firmware/solar_passthrough/#full-solar-passthrough_1) limitation:

[image]

If your inverters are unable to consume as much power as the charge controllers produce, there is currently no way around this issue. Otherwise make sure that your line losses setting is not too large such that the inverters have a chance to consume all of the charge controllers output power. Also, start full solar-passthrough earlier, as to avoid the absorption state altogether.

And then there is work in progress to mitigate this issue in #1167.

I assume that a user's setup must be such that the inverters are already at the allowed maximum power limit if this situation occurs. Otherwise, something is very off. Am I right?

I can image preventing to reduce the inverters' output power if the charge controller switches to absorption. That should be doable. However, picture me this: What if you have multiple charge controllers and only some switch to absorption. What should be the logic? Do all need to switch to absorption to prevent lowering their output? Only one? What about the MQTT solar charger provider? Does it need a new topic/setting for this?

It would be much preferred if we could base this decision on the state of the battery, I guess. 

~~How about we prevent lowering the inverter output power while in full solar-passthrough in general? That would mean we keep increasing the power, but never decrease it, until the full solar-passthrough stop threshold is reached (voltage limit value or SoC dropped by one percent point). What happens when clouds appear and the solar charger output actually drops?~~ Nah, this seems to be a bad idea.

embed-3d, 2025-02-01:
> It would be much preferred if we could base this decision on the state of the battery, I guess.

I totally agree on this. I think the best solution overall would be to tell the solar chargers the actual battery voltage and maybe also the temperature. So If you have multiple they should behave all the same and different line losses on multiple chargers shouldn't matter.

When I understand the docs correctly, we would only need to send 0x2002 & 0x2003 to the solar chargers. 

[image]

I already had a detailed look on this a few months ago. Hacking it into the VE direct frame handler library would be possible but is not my prefered solution. The big problem of the current implementation is the "static" queue. Inserting additional commands/requests makes it harder to maintain and more complex than required.
Rewriting this library is maybe better but without additional hardware for testing and integration I'm not able to do this.

> What if you have multiple charge controllers and only some switch to absorption. What should be the logic?
Telling the charger controllers the current battery voltage should solve this issue. I guess they will switch all on the same time to absorption. But then we need to Be able to control these settings.  

> Do all need to switch to absorption to prevent lowering their output? Only one?
To handle this case I think we need for each charger a kind of controller. And allow the inverter to draw more. With kind of a pid controller

> What about the MQTT solar charger provider? Does it need a new topic/setting for this?
Currently no Idea but maybe yes.

> And then there is work in progress to mitigate this issue in https://github.com/hoylabs/OpenDTU-OnBattery/pull/1167.
Looks nice, I had a look through the pdf's but I'm not 100% sure If this will solve the issue

I think the main issue is not the absorption/float mode but the mppt state **"Voltage or Current limit"**. **In this state the panels can produce more power than the charger controller delivers to the battery & inverter**, so we can draw more power with the inverter(s). But in this case we need to regulate the inverter output based on this state and maybe also based on the battery state

schlimmchen, 2025-02-01:
> I totally agree on this.

Then why do you argue that we need to talk to the charge controllers? My main concern with this is the assumption that all users have Victron SmartSolar charge controllers. That kind of vendor-lock is an issue.

> So If you have multiple they should behave all the same and different line losses on multiple chargers shouldn't matter.

**should**

> When I understand the docs correctly, we would only need to send 0x2002 & 0x2003 to the solar chargers.

Many users don't even have the TX line to the charge controller wired up, so... we won't fix the issue for those users?

Don't get me wrong, I understand where you are coming from and your suggestions would solve the issue for some users, but only for those few...

How about this: While we are in full solar-passthrough, we don't decrease the inverters' limits until we see the battery voltage drop (by what amount?).
* If the solar output becomes less than the inverters currently use, we will start discharging the battery. This swing from charging (or even idling) to discharging should be noticable.
* If there is surplus solar power even though we are in full solar-passthrough, the inverters keep running at full power, and the solar charger(s) will eventually reduce their power output to match the inverter power.

@SW-Niko I am sure you have an opinion on this :wink:

SW-Niko, 2025-02-01:
😆 Yes, I have.

But before I start discussing the pros and cons I will explain the options.

@embed-3d 
I assume your inverters can produce more power as the solar charger can deliver, right?
In this combination you have 3 options.

1) You can configure the "(Full) Solar-Passthough Verluste" with a value above 5% (or more):
Your system will charging the battery up to absorption mode but will not use the available solar power once the absorption mode was reached. The situation you have now.

2) You can configure the "(Full) Solar-Passthough Verluste" with a value below 2% (or less) and configure the "Full Solar-Passthough Start Schwellenwert" identical to the absorption voltage of the solar charger :
Your system will deliver all available solar power once the solar charger absorption mode is reached.

3) You can configure the "(Full) Solar-Passthough Verluste" with a value below about 2% and reduce the Inverter limit to a value below the solar charger power.
Your system will load the battery and use all available solar power in bulk and absorption mode.

Try it out and give feedback of the result.

SW-Niko, 2025-02-02:
Hello @embed-3d ,

I have forgotten the 4. option.
You can tweak the "(Full) Solar-Passthough Verluste" to fit perfekt to your system and keep the "Full Solar-Passthough Start Schwellenwert" below the Absorption voltage. 
In that case you transfer all available solar power into the inverter and the battery will neither charged or discharged.

>I think the best solution overall would be to tell the solar chargers the actual battery voltage and maybe also the temperature. So If you have multiple they should behave all the same and different line losses on multiple chargers shouldn't matter.

What kind of charger do you have? Can you activate and use the Victron VE.Smart Network (Bluetooth)?

@schlimmchen 
>What if you have multiple charge controllers and only some switch to absorption. What should be the logic? Do all need to switch to absorption to prevent lowering their output? Only one? What about the MQTT solar charger provider? Does it need a new topic/setting for this?
Good point. I can check that out. Disable the VE.Smart Network and let's see what happens.
I assume this not a problem. In worst case we do not use all available solar power as long as the charger are in different states.

embed-3d, 2025-02-03:
@schlimmchen 
got your point with the vendor locking and users not having the tx line connected.

I think we should go both routes, the voltage handling you suggested for other solar chargers that can't communicate and those who don't have the solar charger connected and a enhanced communication variant for users with available communication with smart solar chargers.

I will start to rework the ve direct library in the next weeks when I have time.

@SW-Niko 
This is my setup:
2kWp Solar panels, connected to one SmartSolar MPPT 250|60 in serial. (Absorption Voltage 52V (might will change it to 52.4V), Float Voltage 51V)
The battery output goes to a pylontech US3000C and to a HMS-2000-4T
I have also a Shelly pro3em to archive a zero feed-in.

So without any losses the Charger and the Inverter have the same power, but I'm currently limited to the 800W in Germany.

My plan was to configure It like this:
My System should not discharge the battery and feed all into to the grid to the point of zero feed-in. If there is more power available this should go in the battery. When the battery is "full" and enough solar power is available It should deliver the full 800W to the grid. When the battery is full it should discharge the battery during the night.
When I understand it right this will only be possible in the upcoming surplus mode?

I think I messed up my configuration since I finished my build end of November and I had not enough sun to adopt it properly...

SW-Niko, 2025-02-03:
Hello @embed-3d,

>When I understand it right this will only be possible in the upcoming surplus mode?

Yes, the surplus mode can do that. But the "(Full) Solar-Passthough Mode" can do that as well.
You must go a little bit below the full load (below the absorption voltage) and tune the "(Full) Solar-Passthough Verluste" to fit perfekt to your system.
For more information you can read the pdf-File in #1167.

embed-3d, 2025-05-21:
@SW-Niko @schlimmchen I created a pull request to fix that issue.