hoylabs/OpenDTU-OnBattery #1598: scheinbar unmotivierte Einspeisung

Sorbat, 2025-02-01:
### What happened?

Hallo, 
mir ist ein merkwürdiges Verhalten aufgefallen, welcher ich mir nicht erklären kann. 

Aufgrund der Problemtik in #1577 bin ich aktuell auf der 2024.10.22. 

Es werden unregelmäßig, für mich nicht nachvollziehbar, größere Mengen Strom eingespeist. Ich habe hier einmal 2 Screenshots des Graphen der Tibber App. 

Ein Log wäre sicherlich sehr hilfreich, allerdings war das Problem heute nachmittag wärend der Rund 90 min. Logausgabe leider nicht nocheinmal aufgetragen. 

Habt Ihr vielleicht eine Idee, wie ich das Problem eingrenzen kann?

[image]

[image]

### To Reproduce Bug

keine Ahnung

### Expected Behavior

Bezogen auf dieses Problem, dass nicht soviel eingespeist wird 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.10.22

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-02-01:
Uptime deiner DTU ist wie lange?

Du hast mich extra im discord gefragt, ob du ein Issue aufmachen solltest. Dabei habe ich nicht bedacht, dass du auf dem Oktober-Release unterwegs bist. Daher bin ich (hoffentlich verständlicher Weise) nicht sonderlich motiviert, einen Fehler zu suchen. Denn der Support für mehrere Inverter hat ja den ganzen DPL umgekrempelt, und das kam im November-Release. Daher können wir zwar gerne abwarten, ob jemand eine Eingebung hat und wir dahinterkommen, aber es würde mich sehr überraschen, wenn dieser Fehler im aktuellen Release auch noch besteht.

Log wäre natürlich wichtig. Danke, dass du dich schon drum bemüht hast. Vielleicht hast du Glück und du kannst es mal einfangen.

Sorbat, 2025-02-01:
Der Hintergrund zu deiner Frage nach der Uptime ist, dass du Reboots vermutetest und das Problem in der Folge auftritt, oder?

Ich denke auch nicht unbedingt, dass es ein Bug ist. Vermutlich bin ich hier dann auch nicht richtig mit einem Issue. 

Ich hatte schon mal an den Stromzähler gedacht. Als Grundlast ist aktuell 250w eingestellt. Würde von den Werten her halbwegs passen.

schlimmchen, 2025-02-01:
> Der Hintergrund zu deiner Frage nach der Uptime ist, dass du Reboots vermutest und das Problem in der Folge auftritt, oder?

Ja, genau, denn wenn die DTU neustarten muss, kann sie nicht nachjustieren. Wenn du dann auch noch einen Verbraucher abschaltest, speist dein Inverter natürlich fleißig ein.

> Ich hatte schon mal an den Stromzähler gedacht. Als Grundlast ist aktuell 250w eingestellt.

Dann hättest du eine Last von weniger als 10W? Das erscheint mir nicht realistisch?

> Ich denke auch nicht unbedingt, dass es ein Bug ist.

😲 Aber du hast doch "I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation." angehakt?! 😜 

> Ich beobachte das die nächsten Tage mal.

Alles klar.

Sorbat, 2025-02-01:
> Dann hättest du eine Last von weniger als 10W? Das erscheint mir nicht realistisch?

Ich hab ja noch den anderen (ungeregelten) Inverter, der heute bei strahlendem Sonnenschein recht gut eingespeist hatte. Der hatte vermutlich die Grundlast schon gedeckt

> 😲 Aber du hast doch "I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation." angehakt?! 😜

Ich konnte meine Frage nicht absenden ohne das anzuhaken 😜

Sorbat, 2025-02-02:
Die Uptime war wie erwartet, also keine Reboots. 

Es liegt wohl tasächlich am Stromzähler. Vielleicht habe ich da mit 1sek Timeout auch nicht den richtigen Wert gewählt.

Ich habe heute mehrfach die Verbindung zum Stromzähler unterbrochen und der über den DPL geregelet Inverter wurde dann auf den Wert der Grundlast +/- (hier 500w) eingestellt, ohne Berücksichtung auf den weiteren Inverter, der die Grundlast schon fast alleine deckte. 

Ich habe nun die Grundlast auf 30w eingestellt. Damit tritt das Problem nicht mehr auf. 

Vielen Dank. 
[image]

schlimmchen, 2025-02-02:
Alles klar, danke für die Rückmeldung!